MarlinFirmware/Marlin #19544: XPT2046 TFT has touchscreen feature inverted

Whiskaya, 2020-09-29:
### Bug Description

Touchscreen (only TOUCH) is inverted, so when touching bottom left, it activates top right.

Hardware:
FLSUN Q5
MKS Robin Nano 1.2
Stock TFT display

### My Configurations

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/5295727/Config.zip)

### Steps to Reproduce

1. Load firmware.
2. Touch the screen.

**Expected behavior:**

Touch recognition in the touched place.

**Actual behavior:**

Touch is recognized inverted in X and Y (when touching bottom left, it recognizes top right).

### Additional Information

1.- Used:
     #define TFT_320x240
     //#define TFT_DRIVER AUTO (doesn't make a difference for this issue)

2.- Already tried changing these values, from positives to negatives and all combinations possible.
     #define XPT2046_X_CALIBRATION -12316
     #define XPT2046_Y_CALIBRATION 8981
     #define XPT2046_X_OFFSET        -43
     #define XPT2046_Y_OFFSET        257

3.- I'm not that good at coding, so fiddled around a bit with touch.init(), but then left it as it was.

4.- #define GRAPHICAL_TFT_ROTATE_180 doesn't work with my settings (most probably I'm overlooking something. If so, I'm sorry!)

rhapsodyv, 2020-09-29:
I think is already fixed in the #19192, as the new code support any orientation.

Whiskaya, 2020-09-29:
> I think is already fixed in the #19192, as the new code support any orientation.

That refers to the orientation of the whole screen -- this issue is only about the touchscreen functionality (unless I missed it).

Tried fresh Marlin from SEP 28, and the drivers and rotation lines don't seem to work.

Looked around for other definitions (as there are many) but didn't find any redundancies so far.

rhapsodyv, 2020-09-29:
Did you test #19192?

I made the touch follow the screen orientation.

jmz52, 2020-09-29:
Use M995 to calibrate touch screen.
Then update configuration files with your calibration data (reported to COM port).
Also you can save calibration data to EEPROM if it is enables.

Whiskaya, 2020-10-01:
> Use M995 to calibrate touch screen.
> Then update configuration files with your calibration data (reported to COM port).
> Also you can save calibration data to EEPROM if it is enabled.

This actually did it!

Thank you so much!

closed 2020-10-01 by Whiskaya