MarlinFirmware/Marlin #19547: Error when building 2.0.7 for SKR mini E3 V2.0

Portzal, 2020-09-29:
### Bug Description
SKR Mini E3 V2.0, Marlin 2.0.7
To build without errors with TMC2209,s selected for X,Y,Z and E0 I had to comment out the following in trinamic.ccp. without the change, enabling any more than 1 driver resulted in an error.
Any thoughts?
#define TMC_HWSERIAL_CONFLICT_MSG(A) STRINGIFY(A) "_SLAVE_ADDRESS conflicts with another driver using the same " STRINGIFY(A) “_HARDWARE_SERIAL” 

#define SA_NO_TMC_HW_C(A) static_assert(1 >= count_tmc_hw_serial_matches(TMC_HW_DETAIL_ARGS(A), 0, COUNT(sanity_tmc_hw_details)), TMC_HWSERIAL_CONFLICT_MSG(A)); 

//SA_NO_TMC_HW_C(X);SA_NO_TMC_HW_C(X2);

//SA_NO_TMC_HW_C(Y);SA_NO_TMC_HW_C(Y2);

//SA_NO_TMC_HW_C(Z);SA_NO_TMC_HW_C(Z2);SA_NO_TMC_HW_C(Z3);SA_NO_TMC_HW_C(Z4);

//SA_NO_TMC_HW_C(E0);SA_NO_TMC_HW_C(E1);SA_NO_TMC_HW_C(E2);SA_NO_TMC_HW_C(E3);SA_NO_TMC_HW_C(E4);SA_NO_TMC_HW_C(E5);SA_NO_TMC_HW_C(E6);SA_NO_TMC_HW_C(E7);

### My Configurations

[Configuration.h.zip](https://github.com/MarlinFirmware/Marlin/files/5296996/Configuration.h.zip)
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/5297000/Configuration_adv.zip)

ellensp, 2020-09-29:
Not a error, you have a configuration problems.

You have TMC2209's all of which share one serial port (this is how this board was designed) so you need to set corect unique slave addresses to each stepper driver. 

Ie you need You need the following in Confiuration_adv.h

  #define  X_SLAVE_ADDRESS 0
  #define  Y_SLAVE_ADDRESS 2
  #define  Z_SLAVE_ADDRESS 1
  #define E0_SLAVE_ADDRESS 3

closed 2020-09-29 by ellensp

Portzal, 2020-09-29:
Thanks for clearing that up for me. I have never come across this issue before. Made the changes and it built without errors.

ellensp, 2020-09-29:
Many have this issue and M122 says everything is fine, but what is really happening is all TMC2209's setting for X,Y,Z and E get applied to slave 0 only. M122 also only gets all data from slave 0, The other TMCs run with power on defaults.  
This this test was added to catch this issue as many don't know about this setting.

Portzal, 2020-09-29:
@ellensp sorry to bother you, but is there a reason behind the order they are assigned? As why they are not just 0,1,2,3 as opposed to 0,2,1,3 Thanks again for being so helpful.

sjasonsmith, 2020-09-29:
@Portzal that is just how they are wired on the board. I suspect that was a mistake when designing the board, and it wasn’t worth fixing.