MarlinFirmware/Marlin #19549: Filament Change Temperature Issue[BUG] (short description)

benebrady, 2020-09-29:
### Bug Description

 When the filament change is triggered by the filament runout sensor, the temperature is set to 170 degrees no matter what the print temperature has been set at. 

### My Configurations

**Required:** Please in
[Ben E. Brady - Filament Change.zip](https://github.com/MarlinFirmware/Marlin/files/5297539/Ben.E.Brady.-.Filament.Change.zip)
clude a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. Set print temperature to 200-240
2. Trigger filament change via runout switch
3. 

**Expected behavior:** [What you expect to happen]
I expect the temperature to remain at the desired print temperature as it has done in Marlin 2.0.5.4
**Actual behavior:** [What actually happens]
Temperature is automatically set to 170 c and when the filament change is completed, you have to wait for the print temperature to be increased to the original temperature. The reduction of the temperature causes an issue with filaments that need to be printed at higher temperatures such as PETG and nylons.
#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

sjasonsmith, 2020-11-13:
It looks like 170 is your setting for `EXTRUDE_MINTEMP`. Something in the pause code must be setting the temperature to that during the tool change.

rhapsodyv, 2020-11-30:
Can you better describe step-by-step what are you doing?
This config is used to tell marlin how many seconds it will wait with heaters turned on, while changing filament: `PAUSE_PARK_NOZZLE_TIMEOUT`. 
If you takes more than `PAUSE_PARK_NOZZLE_TIMEOUT` seconds to change the filament, marlin will start cold down the heaters for safety.
Do you try increase it?
The cold down happens in less than `PAUSE_PARK_NOZZLE_TIMEOUT` seconds? or after?

benebrady, 2020-11-30:
Before the print starts and the hot end is up to printing temperature, If the filament change is triggered, the temperature is automatically set to the minimum extrusion temperature before the filament load takes place. Once the filament load and purge completes, then the print temperature is restored to the hot end, but it takes some amount of time before the print temperature is reached again.Previous versions of Marlin maintained the print temperature throughout the filament change process.Sent from my Galaxy

rhapsodyv, 2020-11-30:
Sorry, I didn’t understand you. Why would you or marlin trigger a filament runout before the printing?
And if the nozzle already have the temperature, you can just load the filament, I don’t see a change event here here.

Can you provide the step by step of what you are doing, trying to write all details? This is important. What are you using? Screen? Menu? Gcode? Octoprint? Are you preheating? Etc...

benebrady, 2020-11-30:
If there is a filament break or if there is no filament in the filament runout sensor before the print starts printing.Sent from my Galaxy

rhapsodyv, 2020-11-30:
Filament runout uses the current target temperature to reload the filament. But if it's triggered before a print, Marlin don't even know what is the target temperature for the hotend, that is the reason it uses 170 (min temp).

So, filament runout will never work if it happen before Marin run the gcode command that set the target temperature for the hot end (`M104`). And we can do nothing, as, well, we don't know which is target temperature.

The only way to be able to use filament runout sensor for this situation, is disabling runout by default (`#define FIL_RUNOUT_ENABLED_DEFAULT false`), and enabling it manually with `M412 S1` **AFTER** set the target temperature for the hotend (`M104`), never before.

I'm closing this, as it is not a bug, but a behaviour that we can't handle at all. And we have no drawback, as the primary usage for filament runout feature is to avoid a print disruption. If it trigger before a printing start, the user can simply load the filament and restart the printing, with no single damage.

Thanks!

closed 2020-11-30 by rhapsodyv