MarlinFirmware/Marlin #19550: Uploading from ardunio IDE exceeds command character limit

mikeandsuch, 2020-09-29:
### Bug Description
Im not sure if this actually a bug or im doing something wrong.

During the compiling step of uploading i get an error stating that:
**The filename or extension is too long.
Error compiling for board Sanguino.**

### My Configurations

[config's.zip](https://github.com/MarlinFirmware/Marlin/files/5298590/config.s.zip)

### Steps to Reproduce

Just standard uploading from ardunio IDE procedure following this tutorial -> https://marlinfw.org/docs/basics/install_arduino.html

**Expected behavior:** [What you expect to happen]
For to successfully upload the firmware to my ender 3 pro

**Actual behavior:** [What actually happens]
Error message from above.

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ellensp, 2020-09-29:
Welcome to Windows. Windows 10 has a hard coded 32k character limit on exec/fork. The Arduino IDE is badly written and the final link command exceeds this limit on large programs like Marlin
Use vscode with platfomio. Its faster, it knows how to avoid this windows limitation, and produces smaller code. (requires on old controllers with only 128KB)

closed 2020-09-29 by ellensp

mikeandsuch, 2020-09-29:
I edited the report to encompass all three of the issues i've encountered instead of posting three separate reports

ellensp, 2020-09-29:
re avrdude: stk500_recv(): programmer is not responding

You selected the wrong bootloader type, or the board has no bootloader.
Easiest fix is to install the optiboot bootloader, (is default with modern arduino ide) then you know what bootloader it has. Its also only 2k vs 4k of the older bootloader.  

No idea on sublime.

ellensp, 2020-09-29:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

mikeandsuch, 2020-09-29:
Oh ok thanks