MarlinFirmware/Marlin #19551: [BUG] MBL "invalid mesh" error got after M420 S1 command

digant73, 2020-09-29:
### Bug Description
I enabled MBL and babystep on last official Marlin 2.0.6.1.
I executed the MBL process on a 5x5 grid.
When the process is completed I try to enable the mesh using the command M420 S1 but I got the error "invalid mesh".

Below the sequence I followed for the MBL process:
 1) G29 S1: start MBL process and move to point 1
 2) M290 xx: move up / down the nozzle to the desired height
 3) G29 S2: store the offset for the point (eventually changed at step 2) and move to next point
 4) repeat step 2) and 3) until the end of the process
 5) send M500 to store on eeprom
 6) M420 S1

**Expected behavior:** [What you expect to happen]
The mesh is successfully loaded with the command M420 S1

**Actual behavior:** [What actually happens]
The M420 S1 provide the error "invalid mesh".
I also got the same error using the command G29 S0 to display the mesh state

#### Additional Information
Tested on official Marlin FW 2.0.6.1 with MBL and babystep enabled

boelle, 2020-09-29:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

digant73, 2020-09-29:
sorry for the missing files.
Attached the two configuration files (the only ones I changed).
I will also try to use the bugfix branch and verify if the problem is still present

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5300494/Marlin.zip)

closed 2020-10-05 by digant73