MarlinFirmware/Marlin #19552: [BUG] IDEX Sensorless homing with TMC2209, again

ogland, 2020-09-29:
### Bug Description

When using sensorless homing with IDEX on a GTR 1.0 and TMC2209 X1 won't home even after the workaround described here:
https://github.com/MarlinFirmware/Marlin/issues/17532

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/5299589/Configuration_adv.zip)

### Steps to Reproduce

1. Start printer
2. Flash firmware with fixed in link above
3. [and so on...]

**Expected behavior:** Successful homing sequence for X1 and X2

**Actual behavior:** X2 homes properly, then immediately after homing fails.

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

sjasonsmith, 2020-09-29:
I want to set some realistic expectations here. Both IDEX and sensorless homing are uncommon and prone to have issues. It is already hard to find contributors to debug and fix IDEX issues. If your issue also requires sensorless homing there are very few people able to help.

I believe @InsanityAutomation has previously stated that sensorless really isn’t accurate to  use reliably with IDEX. Is there a particular reason you are using these combined, or are you able to use mechanical switches?

ogland, 2020-09-29:
@sjasonsmith I figured I could tune it to be accurate enough since the coil position update to the sensorless homing routine. I can of course add mechanical switches fairly easily, but they're not super accurate either.

sjasonsmith, 2020-09-29:
@ogland very few contributors seem to work with IDEX, and I don’t know if any of them are using sensorless. Do feel capable to help debug the code to fix this?

ogland, 2020-09-29:
@sjasonsmith I'm unfortunately not much of a developer, so I hope someone will pick up on this reported bug.

sjasonsmith, 2020-09-29:
Does your machine have switches? If so, could you test using switches instead of sensorless, to determine if the problem is all IDEX homing or only when using sensorless?

cbaugher, 2020-09-30:
I've been recently diving into the waters of IDEX printer setup, and I would concur that sensorless homing is not a great idea.  For the Y axis it would be fine, and maybe Z if you also have probing.  But the positioning of the X axes is critical if you intend to have repeatable nozzle alignment.  Because there are 2 axes the potential positioning range error is doubled.

C|

ogland, 2020-09-30:
@cbaugher I'm not sure how your personal opinions are relevant to this bug report.

This is an experiment and I'm pretty sure it's more exact than some seem to think. It's certainly exact enough for duplication mode or soluble supports, which is what I'll be using it for.

cbaugher, 2020-10-01:
> @cbaugher I'm not sure how your personal opinions are relevant to this bug report.
> 
> This is an experiment and I'm pretty sure it's more exact than some seem to think. It's certainly exact enough for duplication mode or soluble supports, which is what I'll be using it for.

@ogland 
My opinion is not really that relevant.  Mostly I was just jumping on the "Are you really sure you want to do this?" bandwagon.  It seems the answer is yes, so... ok.

I did look through your config files to compare them to mine (which so far are working for the most part*).  I see you have probing turned on; which shouldn't matter but it might be something to turn off to see if it makes a difference (since it is kinda sorta related to homing).  But one other thing that sticks out is these three defines in config.h:

```
// Manually set the home position. Leave these undefined for automatic settings.
// For DELTA this is the top-center of the Cartesian print volume.
#define MANUAL_X_HOME_POS  -50
#define MANUAL_Y_HOME_POS -4
#define MANUAL_Z_HOME_POS  0
```

For me these are commented out, and my homing works as it should*.  The reason this sticks out to me is because an IDEX printer has X and X2 axes, and you say that your X2 homes correctly but X doesn't.  The 3 defines above are for X, Y, and Z which are the axes that aren't homing for you.  Of course it's entirely possible this has nothing to do with it, but it seems strangely coincidental.

C|

* I do actually have one strange issue that occurs on my printer (Hictop D3 Hero IDEX).  Homing works correctly, but the first g1 move after homing sometimes does this thing where it moves the X and Y axes together at the slow homing feedrate for a bit, before it executes the G1 move.  It's totally bizarre and I even submitted a bug report for it.  It's been closed due to inactivity because there isn't anyone to work on it (and I've been too busy).  Fortunately it's not a show stopper, but only a weird annoyance when you start a print.

ogland, 2020-10-01:
@cbaugher is there a known conflict with setting home positions and using sensorless homing for X1&X2? Because I can home XYZ fine if I don't use Dual X Carriage.

The X2 home position is set under Dual X Carriage settings.

The primary reason why I don't think it's good to comment home positions out is because I then can't define where 0 is properly right?

ogland, 2020-10-01:
@sjasonsmith I have ordered some optical endstops to get it up and running, so that will most likely work anyway.

cbaugher, 2020-10-02:
> @cbaugher is there a known conflict with setting home positions and using sensorless homing for X1&X2? Because I can home XYZ fine if I don't use Dual X Carriage.
> 

There isn't a conflict that I know of.  But when you add in IDEX it changes a lot of things in a lot of places.  So it's entirely possible it only matters when IDEX is turned on.

> The primary reason why I don't think it's good to comment home positions out is because I then can't define where 0 is properly > right?

Whether or not you need it for your particular setup I don't know.  For both of my printers (IDEX and traditional) the default homing settings have been fine (meaning no manual settings).  And to be completely honest I'm not even sure how those manual settings are supposed to work.  On my IDEX for example: after the Y axis is homed, the nozzle is off the edge of the bed.  So I set Y_MIN_POS to be -8, so that the 0 (minimum printable) position is on the bed.  If it needed a different offset I could just change it there.  Obviously the manual home positions have some purpose, I just don't know that is.

But anyway, what I was really suggesting is to comment those lines out just to see if it makes any difference.  If it does, the issue (bug) is probably related to that somehow.  If not, then the problem is likely somewhere else.

When you try homing; what happens after X2 homes?  Does X1 move at all or does it just sit there?  Do Y and Z home?  Are there any screen messages?

C|

InsanityAutomation, 2020-10-02:
Manual home position is primarily used nowadays for multiple tools where the second tool needs to be able go lower the Z_MIN position, or first tool homing point. If T1 is higher than T0 then Z_MIN needs to allow going below 0 while the manual home still forces the homed point at 0 after applying the probe offset. 

Another case is for situations where the home position is off the bed with a known offset, such as the physical switch used on the Lulzbot Taz 6 machines.

ogland, 2020-10-02:
@InsanityAutomation yeah exactly. In this case the extruders can both go outside the build plate at opposite ends, so they both can reach the entire build plate.

@cbaugher I'll try using offsets in a different way and see if that works.

The homing sequence fails immediately after x2 successfully homes, which it does before x1.

cbaugher, 2020-10-02:
> Manual home position is primarily used nowadays for multiple tools where the second tool needs to be able go lower the Z_MIN position, or first tool homing point. If T1 is higher than T0 then Z_MIN [...]

Ahh, ok.  So it's for situations where the home position is not at the end of travel for that axis?

> @InsanityAutomation yeah exactly. In this case the extruders can both go outside the build plate at opposite ends, so they both can reach the entire build plate.
> 
> @cbaugher I'll try using offsets in a different way and see if that works.

If I understand what IA is saying, I think you want to use the max and min positions instead of the manual home positions.  With sensorless homing the home position WILL be at the end of travel.

On my machine X1_MIN_POS is -48 (48mm off the left of the bed) and X1_MAX_POS is the bed size.  X2_MIN_POS is 0 and X2_MAX_POS is 347.6 (47.6mm off the right of the bed).

> 
> The homing sequence fails immediately after x2 successfully homes, which it does before x1.

So after X2 homes, everything stops and it just sits there?

C|

closed 2020-11-09 by github-actions[bot]