MarlinFirmware/Marlin #19553: [BUG] Ender 3 V2 POWER_LOSS_RECOVERY function not working

dennieb, 2020-09-29:
### Bug Description

Changed the POWER_LOSS_RECOVERY function from false to true, but when I turn off the power by flicking the switch after printing a few layers, then turn it on again, it does not have the option to resume the print.

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/5300343/configs.zip)

### Steps to Reproduce

1. Start print like normal, then turn machine off manually after a few layers
2. Turn machine back on
3. No message or option to resume print

**Expected behavior:** Print needs to resume where it left off

**Actual behavior:** No option to continue the print before the power got cut

#### Additional Information

dennieb, 2020-09-29:
Quick update: I managed to get this feature on my printer working by flashing the .rar from the creality website named 
 Ender-3 V2-V1.0.2 Source code.rar uploaded on 28 Sep, 2020. This uses an older version of Marlin. But atleast you can use the power loss recovery feature on your Ender 3 V2 if someone reads this and needs a quick fix.

sjasonsmith, 2020-10-10:
@dennieb you may need to use M413 to enable the feature prior to your print.
https://marlinfw.org/docs/gcode/M413.html

boelle, 2020-10-13:
@dennieb did you try M413?

closed 2020-11-20 by github-actions[bot]

Kozonak, 2021-01-16:
Having the same issue, using the configs above and adding M413 S1 in Cura results in the resume print menu appearing but then nothing happens. It will heat up and just sit there doing nothing. 

Going back to pre 2.0 marlin with the same settings and removing M413 from Cura results in the printer recovering ok.