MarlinFirmware/Marlin #19554: [BUG] (short description)

fred2088, 2020-09-29:
only change this from stock firmware 2.0.7

#define mainboard BOARD_ESPRESSIF_ESP32
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

------------------------------ got this error -----------------------
sketch\src\HAL\AVR\u8g_com_HAL_AVR_sw_spi.cpp: In function 'void u8g_com_arduino_init_shift_out(uint8_t, uint8_t)':
sketch\src\HAL\AVR\u8g_com_HAL_AVR_sw_spi.cpp:71:15: error: cannot convert 'volatile uint32_t* {aka volatile unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment
   u8g_outData = portOutputRegister(digitalPinToPort(dataPin));
               ^
sketch\src\HAL\AVR\u8g_com_HAL_AVR_sw_spi.cpp:72:16: error: cannot convert 'volatile uint32_t* {aka volatile unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment
   u8g_outClock = portOutputRegister(digitalPinToPort(clockPin));
                ^

is it the U8g2lib need to replace U8glib ?

Regards

boelle, 2020-09-29:
try with vs code + platform.io extension

that will load the correct versions down and also update when needed

also use [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) 2.0.7 is just a frozen copy of that

and for good messure attach configs as zip file

ellensp, 2020-09-30:
BOARD_ESPRESSIF_ESP32 is s generic esp32 board. it has no pins defined for where to plug a REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
You need to define the pins you are using.
IE you need to add defines for LCD_PINS_ENABLE and LCD_PINS_RS

closed 2020-09-30 by ellensp

ellensp, 2020-09-30:
This is the defines in it entirty https://github.com/MarlinFirmware/Marlin/pull/11020