MarlinFirmware/Marlin #19555: [BUG] (short description)

hugo4715, 2020-09-29:
### Bug Description

When trying to use the G35 Assisted Tramming command, Marlin is sending an invalid string to the serial console, which means i can't follow the process correctly. 
It says "Attempted to write invalid text to console, which could be due to an invalid baudrate" but there is only a few string causing this issue so I don't think the issue is with my baudrate.

### My Configurations

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5300512/Marlin.zip)

### Steps to Reproduce

1. Send a G35
2. Wait for all points to be probed
3. The error shows up in the serial monitor

#### Additional Information
I'm using Proterface to control an Anet A8 with a probe.
You can find the error log here:
https://gist.github.com/hugo4715/a261fdbff2ac0801f03d0d05d1208339

ellensp, 2020-09-30:
Please update the title from "[BUG] (short description)" to something more descriptive.

sjasonsmith, 2020-09-30:
I don't have a machine handy to test this on, but I think some of the `SERIAL_ECHO` macros need to use the `_P` variants. Many people don't see this because it doesn't have any impact on 32-bit boards.

@hugo4715 can you try making the changes below in your Marlin source? This is a 2.0.7 file. I know you are using 2.0.6, but I hope these simple edits can be easily made to your file in the same way. I'm not sure how much this changed between the releases.

For each change you are simply adding `_P` to the macro name.
https://github.com/sjasonsmith/Marlin/commit/8b7399908edc358ebe8edc1454694b55f34d2679

hugo4715, 2020-09-30:
I did this but it didn't change, the issue is still present.

thisiskeithb, 2020-10-02:
This may have been fixed in https://github.com/MarlinFirmware/Marlin/pull/19577 since Pronterface didn’t like certain words being sent back.

Can you apply that PR to your build / try the latest `bugfix-2.0.x` and see if Pronterface no longer complains?

ellensp, 2020-10-07:
The issue was fixed in pronterface Sep 6th https://github.com/kliment/Printrun/issues/1103
Perhaps you need to update your Pronterface.

boelle, 2020-10-07:
finding the precompiled version can be a bit tricky

most link goes to http://kliment.kapsi.fi/printrun/ but that has not been updated in 3 years, https://github.com/kliment/Printrun/releases is the way to get it

sjasonsmith, 2020-10-10:
@hugo4715 can you try the change included in this issue, and report back whether that helped?
#19473

hugo4715, 2020-10-11:
[image]
I am still getting an error with the latest version of Pronterface, i'll try updating marlin as @thisiskeithb suggested

sjasonsmith, 2020-10-26:
Duplicate of #19473

closed 2020-10-26 by sjasonsmith