MarlinFirmware/Marlin #19556: Reprap Discount Full Graphics Smart Controller Display is Garbled #12003[BUG]

ktraglin, 2020-09-29:
Reprap Discount Full Graphics Smart Controller Display is Garbled #12003
Previously closed.  Athough I can NOT reproduce this error in Marlin version 2.0.5, this problem appears to have re-emerged in 2.0.6.

Using the default configuration (except added to configuration.h):
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

The timing changes from the previously closed bug/issue yeild no results with this new version (2.0.6), e.g.:
(added to configuration.h)
#define CPU_ST7920_DELAY_1 DELAY_NS(250)
#define CPU_ST7920_DELAY_2 DELAY_NS(250)
#define CPU_ST7920_DELAY_3 DELAY_NS(250)

Also tried these timing settings:
#define CPU_ST7920_DELAY_1 DELAY_NS(0)
#define CPU_ST7920_DELAY_2 DELAY_NS(250)
#define CPU_ST7920_DELAY_3 DELAY_NS(0)

And these:
#define CPU_ST7920_DELAY_1 DELAY_NS(0)
#define CPU_ST7920_DELAY_2 DELAY_NS(400)
#define CPU_ST7920_DELAY_3 DELAY_NS(0)

I really would like to take advantage of the newer 2.0.6 firmware, but for the moment, I'm forced to go back to Marlin version 2.0.5.  Any suggestions will be welcomed. : )

boelle, 2020-09-29:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) its newer than 2.0.5 and 2.0.6 and 2.0.7

ktraglin, 2020-09-29:
I'm using the default files (except for the 4 lines of changes/additions I mentioned).

sjasonsmith, 2020-09-29:
> I'm using the default files (except for the 4 lines of changes/additions I mentioned).

It doesn't matter. You need to attach them to request help. From this issue nobody can tell even what motherboard you are using, which is inseparable from these LCD timings.

ktraglin, 2020-09-29:
> > I'm using the default files (except for the 4 lines of changes/additions I mentioned).
> 
> It doesn't matter. You need to attach them to request help. From this issue nobody can tell even what motherboard you are using, which is inseparable from these LCD timings.

Understood... attached, here.
[Uploading configuration-files.tgz…]()

sjasonsmith, 2020-09-29:
@ktraglin that download link appears to just be a link back to the issue?

ktraglin, 2020-09-29:
> @ktraglin that download link appears to just be a link back to the issue?

I see the same thing.  I'll try again using "zip" instead of "tar-gzip"... stupid me.
[Uploading configuration-files.zip…]()

ktraglin, 2020-09-29:
Still only see link to issue.   Once more, here.
[configuration-files.zip](https://github.com/MarlinFirmware/Marlin/files/5301804/configuration-files.zip)

ktraglin, 2020-09-29:
> Configurations, please
> Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
> requested in the Issue template) and drop them into your next reply.
> We'll check them over and see if [...]

I wasn't even aware of a version 2.0.7, but now I've tried with Marlin-bugfix-2.0.x.  Same results.

sjasonsmith, 2020-09-29:
I was going to test it, but I apparently damaged by Mega2560 board. Hopefully somebody else will volunteer to try it.

ellensp, 2020-09-29:
Firstly did you notice the yellow warnings?
"CPU_ST7920_DELAY_2" Redefined
"CPU_ST7920_DELAY_2" Redefined

as 
  #define CPU_ST7920_DELAY_1 DELAY_NS(0)
  #define CPU_ST7920_DELAY_2 DELAY_NS(0)
  #define CPU_ST7920_DELAY_3 DELAY_NS(63)
are already defined in Marlin/Marlin/src/lcd/dogm/ultralcd_st7920_u8glib_rrd_AVR.h

But regardless of this. Your configs compiles and the display works as expected on my system.

ellensp, 2020-09-29:
to actually change these values you need to set 
#define BOARD_ST7920_DELAY_1 DELAY_NS(250)
#define BOARD_ST7920_DELAY_2 DELAY_NS(250)
#define BOARD_ST7920_DELAY_3 DELAY_NS(250)
not CPU_ST7920_DELAY

ktraglin, 2020-09-30:
> to actually change these values you need to set
> #define BOARD_ST7920_DELAY_1 DELAY_NS(250)
> #define BOARD_ST7920_DELAY_2 DELAY_NS(250)
> #define BOARD_ST7920_DELAY_3 DELAY_NS(250)
> not CPU_ST7920_DELAY

1.  I do recall seeing the yellow warning messages you mentioned.
2.  Your suggestion works (#define BOARD..., instead of #define CPU...
3.  THANK YOU, THANK YOU, THANK YOU!!!  Now I can play around with Marlin 2.0.6. : )

closed 2020-09-30 by thisiskeithb

ktraglin, 2020-09-30:
@thisiskeithb Why was this issue closed?  Has the code been updated to address this issue?

thisiskeithb, 2020-09-30:
> Why was this issue closed? Has the code been updated to address this issue? Is there a newer release users can download?

Your last comment made it sounds like it worked. Did it not?

> THANK YOU, THANK YOU, THANK YOU!!! Now I can play around with Marlin 2.0.6. : )

Also: 

> Is there a newer release users can download?

Yes. Please download [`2.0.7`](https://github.com/MarlinFirmware/Marlin/archive/2.0.x.zip)/[`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

sjasonsmith, 2020-09-30:
Also @ellensp said it worked fine on their system with the default values. Most likely there is a noise or cable length issue in @ktraglin making slower speeds necessary.

ktraglin, 2020-09-30:
> Your last comment made it sounds like it worked. Did it not?
>
>
> Also:
>
>
> Yes. Please download [...]

Yes, what @ellensp proposed did fix the issue, but I had to add their code to my configuration.h file.  My question was about other users being able to have this fix without the need to add code on their own.

ktraglin, 2020-09-30:
> Also @ellensp said it worked fine on their system with the default values. Most likely there is a noise or cable length issue in @ktraglin making slower speeds necessary.

I do not experience the issue with Marlin version 2.0.5, using the exact same printer.  So, I have reservations about a noise or cable length issue being the culprit.

ellensp, 2020-09-30:
issue is specific to you..   either a slow clone screen or long cables...
On my RRD GLCD defaults works fine on the standard RAMPS.

ktraglin, 2020-09-30:
> issue is specific to you.. either a slow clone screen or long cables...
> On my RRD GLCD defaults works fine on the standard RAMPS.

I don't know if a RRD GLCD is the same as the REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER.  I'm also skeptical because plenty of people have not upgraded yet.  It could take months for more people to upgrade, and those who are still using 2.0.5 (like me, until just this week) may not see/experience the issue until they are decide to upgrade.

ellensp, 2020-09-30:
RRD = RepRap_Discount
GLCD = FULL_GRAPHIC_SMART_CONTROLLER

ktraglin, 2020-09-30:
> RRD = RepRap_Discount
> GLCD = FULL_GRAPHIC_SMART_CONTROLLER

I interpreted the RRD part OK, but the GLCD part kinda puzzled me.  Thanks, again.

ktraglin, 2020-09-30:
> Your last comment made it sounds like it worked. Did it not?
>
>
> Also:
>
>
> Yes. Please download [...]

During the past hour, I've had opportunity to download 2.0.7, and applied bugfix-2.0.x.  I've uncommented:
//#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
to reflect:
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

I'm still having the same issue, but again, using 2.0.5, it works fine.  This issue appears to be unresolved without adding the timing parameter lines of code.  This appears to be an issue for 2.0.6, and 2.0.7, although I only tried the bugfix update for 2.0.7.

ktraglin, 2020-09-30:
@thisiskeithb I have attached the config files for 2.0.7 with the bugfix applied, here.
[Configuration-files.zip](https://github.com/MarlinFirmware/Marlin/files/5308330/Configuration-files.zip)