MarlinFirmware/Marlin #19557: [BUG] (new comment to #16355 since random people cannot comment)

tigert, 2020-09-29:
This is just a comment to #16355 which was closed due to inactivity, so I had to file a new one.

In any case, I have a SKR Mini E3 v1.2 with TMC2209. Steppers were loud on Marlin 2.0.7. even with HYBRID_THRESHOLD disabled.

But when I set StealthChop to OFF in the Advanced Settings -> TMC Drivers menu, it became very quiet, like with the firmware that shipped with the board. Sounds almost like the setting functioned backwards.

sjasonsmith, 2020-09-29:
Old issues are locked so that new issues will be opened. Often people attach comments to old issues, when really it is an entirely new problem. If you want to add to a bug that has previously been locked, you need to create a new report and fill out the entire issue template. It is fine to refer to the old issue, but the new issue needs to be complete on its own.

My first recommendation to you is that you re-initialize your EEPROM to be sure any stored settings are not impacting behavior. You may have already been forced to do this when you upgraded your firmware. If not, you should manually do it again.

We typically recommend that you seek assistance in one of our public forums prior to filing a bug report, to help identify whether there is really a bug, or if there is a configuration issue. For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org/)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-09-29 by sjasonsmith

tigert, 2020-09-30:
Ok, will check eeprom, and will file a new issue if needed. 

Thanks for the instructions, very useful for future useful bug reports.