MarlinFirmware/Marlin #19562: [BUG] SD card support for teensy 4.1 does not compile in Arduino IDE with Marlin 2.0.7 from 2.x.x

nanotuxi, 2020-09-30:
### Bug Description

Compiling Marlin 2.0.7 from the default branch for teensy 4.1 with  `MOTHERBOARD BOARD_TEENSY41` and `#define SDSUPPORT` enabled does not work in Arduino IDE 1.8.13/Teensyduino 1.53
**error is**
Compiler says `In file included from /tmp/arduino_build_871308/sketch/src/HAL/STM32F1/onboard_sd.cpp:19:0:
/tmp/arduino_build_871308/sketch/src/HAL/STM32F1/SPI.h:28:37: fatal error: libmaple/libmaple_types.h: No such file or directory
compilation terminated.
Fehler beim Kompilieren für das Board Teensy 4.1.`

### My Configurations
Everything else but the two changes in the config mentioned above are left to default.

### Steps to Reproduce
1. Download fresh sources from `https://github.com/MarlinFirmware/Marlin.git` 2.x.x branch
2. set `#define MOTHERBOARD BOARD_RAMPS_14_EFB` to `#define MOTHERBOARD BOARD_TEENSY41 `
3. change `//#define SDSUPPORT` to `#define SDSUPPORT`
4. compile the sources in Arduino IDE as shown above.

**Expected behavior:** 
Compilation should  not show any errors and sd card support for the teensy 4.1 onboard sd-card reader should be enabled.

**Actual behavior:** 
Compilation fails with file not found error mentioned above.

#### Additional Information
Compilation with `//#define SDSUPPORT` (sd card support disabled) works fine and the board can be connected without any error with serial connection. In Arduino IDE keybord layout for teensy board is set to `German`.
The board is not connected to anything else (bare).
Some examples from the SDfat library from  `https://github.com/greiman/SdFat-beta.git` compiled fine and worked as expected.

boelle, 2020-09-30:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands.

boelle, 2020-09-30:
> Arduino IDE 1.8.13/Teensyduino 1.53

please use vs code + platform.io extension

boelle, 2020-09-30:
> fatal error: libmaple/libmaple_types.h: No such file or directory

there is a small clue

nanotuxi, 2020-09-30:
`
Marlin 2.0.7

@thinkyhead thinkyhead released this yesterday

` , so there are no updates in the bugfix branch.
Tried  with platform.io and compilation is fine now. Connection over serial is fine.
But sending a `M20` says `No media`. Do I have to configure the PINS for the onboard sd card somewhere else to use it?

sjasonsmith, 2020-09-30:
Does M20 work if you first send M21?

Teensy 4.1 was only added a few days ago. I expect there are some issues, but SD seems like core functionality that hopefully works.

ellensp, 2020-09-30:
In Marlin/Marlin/src/HAL/STM32F1/onboard_sd.cpp  add `#ifdef __STM32F1__` after the end of comment `*/` (line 12)
And close it with adding a `#endif`to the very end of the file.

ellensp, 2020-09-30:
Issue is arduino ide trying to compile STM32F1 code. Platformio is smart enough to not compile it, but arduino ide you have to explicitly tell it just skip this when not compiling for a STM32F1.

This compiles, let 'us' know if sd card actually works.

nanotuxi, 2020-09-30:
@ellensp
Compilation works now with Arduino IDE. But for some unknown reason the board doesn't work anymore as it should. There is something wrong with the serial connection/driver on my linux machine. Trying to connect to the board results sometimes in a kernel oops and I have to hard reset the whole machine. Have to investigate this later, so I can't tell you now, if the sd card works. Will give you a feedback when I found the reason for that.
Thank you.

nanotuxi, 2020-10-01:
Compilation works. But the result is not usable. Connecting the tennsy4.1 to e.g. RepetierHost via usb serial results in a not responding system. Disconnecting the cable sometimes releases RepetierHost again, sometimes leads to a kernel oops on my computer. The system needs  a hard- or softreset (magic keys) . 
Tried another Marlin fork which does not show this behaviour and works fine. So there is some faulty code/configuration somewhere in the official branch. 
Tested with both Marlin 2.0.7 and bugfix-branch. 
Tested with both Arduino IDE 1.8.13/Teensyduino1.53 and updated platformio.
[nt_marlin_teeny4_1.zip](https://github.com/MarlinFirmware/Marlin/files/5310901/nt_marlin_teeny4_1.zip)

nanotuxi, 2020-10-01:
@ellensp
Thanks for the solution

closed 2020-10-01 by nanotuxi