MarlinFirmware/Marlin #19563: MMU2 Serial port not working (DUE+RAMPS)

SidDrP, 2020-09-30:
Using Latest Bugfix2.0. 
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5304969/Marlin.zip)
I am using Due + Ramps(1.4) setup. Tested the setup with bare minimum mmu2 settings. Pin 19/18 are freed from use by shifting end stops. I have tried every combination but nothing's working. 
M43 shows both Pin 19/18 are unused.

//#define MMU2_SERIAL_PORT 1
  #define MMU2_SERIAL Serial1  
Also adding Pin config here.
[pins_RAMPS.zip](https://github.com/MarlinFirmware/Marlin/files/5305042/pins_RAMPS.zip)

Thank you.

sjasonsmith, 2020-09-30:
Is this a new configuration, or did it previously work with an older Marlin?

sjasonsmith, 2020-09-30:
I thought RAMPS 1.4 could not be used with DUE?

SidDrP, 2020-09-30:
Bug-fix is latest.
Earlier on Ramps + mega2560 this works on Z_min Z_max pin  as UART1 (19/18)
 #define INTERNAL_SERIAL_PORT 1                    //FOR 18/19 pin  
  #define MMU2_SERIAL internalSerial

I am trying to shift stuff on due (have spare board) just to save time hardware mod if i use different board.

1.4 can be used with little mod. i have level converts on servo and end stop pins (my own diy)

SidDrP, 2020-10-01:
OK after doing some trial and error stuff I managed to get mmu connected @ serial 1 (19/18 pin) on Ramps(1.4)+DUE.
Settings are as follows and should be applicable on 32bit platforms with identical HAL configuration. 

Configuration.h
#define SERIAL_PORT_2 1

Configuration_adv.h
#define MMU2_SERIAL customizedSerial2         (for me this only worked with latest bugfix)

This setting in beginning definitely don't work. One can check it in vscode by looking into pinsDebug.h one can see that pins 19/18 don't get highlighted.

Note: One have to fiddle with baud and tx/rx buffer to get reliable communication if using custom setup (may be Prusa one too) i.e diy mmu. Btw my mmu runs on Arduino uno + cnc shield. Posting pic of first print. No further progress as i am still lacking 3 MK8 gears. :(
Filament ABS
.2mm layer
Ender3 2560+ramps1.4 setup.

[image]

I hope my input helps. I would be thankful if someone crosschecks this info so I am passing correct info here.
Thanks.

closed 2020-10-01 by SidDrP