MarlinFirmware/Marlin #19564: [BUG] Anycubic i3 Mega S display shows negativ hotend temperature

fiechr, 2020-09-30:
### Bug Description

While the temperature in the terminal is showing up fine with 210°C for the hotend, the Anycubic display shows this:

[image]

Heatbed temperature shows up correctly.

Last known working configuration was last stable release 2.0.6.1.

### My Configuration

This is my current configuration for the i3 Mega S, based on 2.0.7:
https://github.com/fiechr/Marlin/tree/b32a496503c6d39843382eda5409f5062339f3c3/Marlin

Main PCB and display PCB of the printer:

[image]

wernju1, 2020-10-01:
I have the same problem at mine. Doesn't matter if compiled with arduino or platformio.
Further Marlin Versions displays correct values.

boelle, 2020-10-01:
> Further Marlin Versions displays correct values.

2.0.7 is the latest so there are no further than that, what version do display correct?

wernju1, 2020-10-01:
I meant that all versions before 2.0.7 were ok. 2.0.7. is the only version with this issue.

fiechr, 2020-10-01:
For me, the last stable release 2.0.6.1 (same configuration) showed the correct values, too. Before that I used Marlin 1.1.x.

boelle, 2020-10-01:
> 2.0.7. is the only version with this issue.

what about [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) ?

sjasonsmith, 2020-10-01:
The 2’s complement of 210 is -46. That doesn’t quite match what you reported, but there might just be a timing gap between reading your two numbers.

My guess is that somewhere the unsigned temperature is being forced into an int8_t, resulting in this behavior. Even if that were a uint8_t, that still seems too small for temperature, which can easily exceed 255.

wernju1, 2020-10-01:
It works. The problem is solved

fiechr, 2020-10-01:
Happy to confirm, that it works for me too.

Thank you for the fast fix and your work for the Anycubic printers! :-)

stef-ladefense, 2020-10-01:
it works very well by replacing "ui8tostr3rj" by "ui16tostr3rj"
it will support heads that support over 255 ° C.

closed 2020-10-02 by thisiskeithb