MarlinFirmware/Marlin #19566: [BUG] (short description)

Loreagle, 2020-09-30:
Hi everyone, I have a problem on my MKS Robin Nano V1.2: 

I'm using a PINDA V2 probe and I enabled temp compensation, the problem I have is that I can't make the board to read the temp value of the probe, the wiring is 100% correct because when I connect the probe instead of bed temp I can see the values.  The problem is that I have the same identical readings on BED and TH1 (where the pinda is connected) like if are the same pin!!!!

I had to add the #define TEMP_BED_PIN to my pins file since is not natively present.

I did like this:

//
// Temperature Sensors
//
#define TEMP_0_PIN                          PC1   // TH1
#define TEMP_1_PIN                          PC2   // TH2
#define TEMP_BED_PIN                     PC0   // TB1

#if HOTENDS == 1
  #if TEMP_SENSOR_PROBE
    #define TEMP_PROBE_PIN            TEMP_1_PIN
  #elif TEMP_SENSOR_CHAMBER
    #define TEMP_CHAMBER_PIN          TEMP_1_PIN
  #endif
#endif

My configuration: https://github.com/Loreagle/Marlin-2.0.7

Thank you!

Loreagle, 2020-10-01:
It looks like there is a problem with ADC and probe looks at the same reading as the bed. 

[image]

ellensp, 2020-10-01:
Seems like PROBE adc is missing from STM32F1 HAL
Give this a try. [HAL.zip](https://github.com/MarlinFirmware/Marlin/files/5312039/HAL.zip)
Extract HAL.cpp and put in Marlin/src/HAL/STM32F1

Loreagle, 2020-10-01:
It doesn't work, it compiles but when flashed I instantly have BED TEMP ERROR at startup with red screen :frowning:  BTW Thank you a lot for your work, at least we discovered where the problem is, now we only need to fix it! :smile:

> Seems like PROBE adc is missing from STM32F1 HAL
> Give this a try. [HAL.zip](https://github.com/MarlinFirmware/Marlin/files/5312039/HAL.zip)
> Extract HAL.cpp and put in Marlin/src/HAL/STM32F1
> Based on bugfix HAL

ellensp, 2020-10-02:
Ok, I missed one place this change was needed. 
[HAL.zip](https://github.com/MarlinFirmware/Marlin/files/5315683/HAL.zip)
Try this new version.

Loreagle, 2020-10-02:
[image]
It seems to work, thank you!!!!! <3

The temperatures seems to float a little but I don't know if it's normal.

ellensp, 2020-10-02:
Created a PR to fix this for all users of this HAL

closed 2020-10-02 by thinkyhead

Loreagle, 2020-10-06:
I confirm that this works very well and finally I have Probe temp readings. Unfortunately this fix broke my neopixels, before neopixel were working quite well with library 1.2.4, now, after only this fix in HAL, doesn't work anymore as expected. It could be a CPU load problem or something similar?

thisiskeithb, 2020-10-06:
> Unfortunately this fix broke my neopixels

Neopixels aren’t supported on `STM32F1` and we have a sanity check for this:

https://github.com/MarlinFirmware/Marlin/blob/e8177735a00d1d7bb20b439544d3324fe01cfe7e/Marlin/src/HAL/STM32F1/inc/SanityCheck.h#L52

Loreagle, 2020-10-06:
I know but using Adagruit 1.2.4 they works well and was working before so I asked.

thisiskeithb, 2020-10-06:
That’s precisely why they aren’t supported. They are really finicky on `STM32F1`-based boards.

Loreagle, 2020-10-06:
Solved! 

Using this library: https://github.com/ccccmagicboy/Adafruit_NeoPixel#meeb_3dp_use it works!!! 

Maybe is possible to add it to official marlin only for robin nano board using lib_deps?