MarlinFirmware/Marlin #19567: [BUG] M108 doesn't work with M701 purge

dermigoe, 2020-10-01:
### Bug Description

After the M701 command loaded the filament my printer want to purge or continue the print.  The message is "paused for user".
I can send the M108 command but there is no response. The printer send continuous "paused for user".
The only way to continue get the printer back to work is to use the LCD.

### My Configurations

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/5309515/Config.zip)

### Steps to Reproduce

1. Send M701
2. Wait until filament is loaded an you see "paused for user" in the Terminal. The LCD show the purge screen
3. Send M108

**Expected behavior:** The printer is able to continue. The message "paused for user" is away. The LCD show the normal screen.

**Actual behavior:** Nothing happen. In the terminal the message "paused for user" is continuing. The LCD shows the purge screen

#### Additional Information

I use a SKR V1.3 Board (LPC1768).
I enabled the EMERGENCY_PARSER.
The M108 command works when M109 or M190 is active. It also works with the M701 an the hotend temperatur i to low but over the cold extrusion temperature. Then it starts to feed the filament.
It seems that i can use M108 when i see the "processing" message in the terminal but not when i see the "paused for user" message

boelle, 2020-10-01:
Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands.

dermigoe, 2020-10-01:
I used the bugfix-2.0.x branch. Downloaded and compiled with the commits from Sep 30, 2020.

rhapsodyv, 2020-10-18:
This is not a bug, but a feature request. 

The code explicitly don't use the `wait_for_user` mechanism that `M108` uses. Seems, M108 is intended to be used when the user just need a "click", but without any option.

M701 is not only waiting for the user, it's asking for the user to respond between two options: Purge More or Continue. `M108` was not designed for that.

@thinkyhead do you see any trouble in making `M108` acting "Resume Print", when marlin is wanting for `PauseMenuResponse` response?

rhapsodyv, 2020-10-19:
@dermigoe 

Did you try `M876 S0` (purge more) or `M876 S1` (continue)?

dermigoe, 2020-10-25:
No i didn't try the M876 command. I will try it in about one and a half week when i am back at my printer. Thank you for this answer.

thinkyhead, 2020-10-29:
`M876` is a fairly new G-code that's meant to be used with a host (i.e., OctoPrint) that has support for "action" prompts.

Even if your connected host doesn't provide an interface for it, when you have `HOST_PROMPT_SUPPORT` enabled in the firmware you can send `M876` (as described) from the command-line to supply a response.

We'll think about whether it makes sense to extend `M108` for situations like this when there is no `HOST_PROMPT_SUPPORT` enabled.

closed 2020-10-29 by thinkyhead