MarlinFirmware/Marlin #19570: Compile Error when using a REPRAP_DISCOUNT_SMART_CONTROLLER Display (2004LCD) on FYSETC F6 1.3 (solution included)

arminth, 2020-10-01:
### Bug Description

See above. Message: Missing Pins-Definitions

### My Configurations
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5311226/Marlin.zip)

### Steps to Reproduce
Enable REPRAP_DISCOUNT_SMART_COTROLLER on FYSETC F6 board and compile

**Expected behavior:** compile, no error

**Actual behavior:** compile error, missing PIN definitions

#### Additional Information
I solved the issue by changing Marlin/src/pins/ramps/pins_FYSETC_F6_13.h
from 
```Line 264  #elif HAS_MARLINUI_U8GLIB```
to 
```Line 264  #elif EITHER(HAS_MARLINUI_U8GLIB, HAS_MARLINUI_HD44780)```

sjasonsmith, 2020-10-10:
@thinkyhead you renamed a bunch of constants related to this recently. This used to use `HAS_GRAPHICAL_LCD`. I am not quite sure whether the proposed change in this PR matches your expectation.

I don't see `HAS_MARLINUI_HD44780` used in other pins files for the purpose of these pin assignments. It seems like there must be either a more generic constant that covers both, or perhaps these displays are failing to build on a lot of board right now...

ellensp, 2020-10-10:
HAS_CHARACTER_LCD => HAS_MARLINUI_HD44780 according to https://github.com/MarlinFirmware/Marlin/pull/19533
Is in release 2.0.7 onwards

sjasonsmith, 2020-10-10:
> HAS_CHARACTER_LCD => HAS_MARLINUI_HD44780 according to #19533
> Is in release 2.0.7 onwards

Display configuration is very inconsistent across boards. I think in many cases these pins are simply defined in an `else` case. Maybe they should be the `else` case if `HAS_WIRED_LCD` is defined?

closed 2020-10-17 by arminth