MarlinFirmware/Marlin #19575: mmu2 compilation error

SidDrP, 2020-10-01:
Config:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5314193/Marlin.zip)

While compiling with //#define PRUSA_MMU2_S_MODE following compilation error occurs. It was not there earlier.

Marlin\src\feature\mmu2\mmu2.cpp:346:16: error: 'mmu2s_triggered' was not declared in this scope
           if (!mmu2s_triggered) {
                ^~~~~~~~~~~~~~~
Marlin\src\feature\mmu2\mmu2.cpp:346:16: note: suggested alternative: 'adc_trigger_t'
           if (!mmu2s_triggered) {
                ^~~~~~~~~~~~~~~
                adc_trigger_t
*** [.pio\build\DUE\src\src\feature\mmu2\mmu2.cpp.o] Error 1

ellensp, 2020-10-02:
in mmu2.cpp 

find
```
        if (ENABLED(PRUSA_MMU2_S_MODE) && last_cmd == MMU_CMD_C0) {
          if (!mmu2s_triggered) {
            can_reset = false;
            // MMU ok received but filament sensor not triggered, retrying...
            DEBUG_ECHOLNPGM("MMU => 'ok' (filament not present in gears)");
            DEBUG_ECHOLNPGM("MMU <= 'C0' (keep trying)");
            MMU2_COMMAND("C0");
          }
        }
```
and replace with
```
        #if ENABLED(PRUSA_MMU2_S_MODE)
          if (last_cmd == MMU_CMD_C0) {
            if (!mmu2s_triggered) {
              can_reset = false;
              // MMU ok received but filament sensor not triggered, retrying...
              DEBUG_ECHOLNPGM("MMU => 'ok' (filament not present in gears)");
              DEBUG_ECHOLNPGM("MMU <= 'C0' (keep trying)");
              MMU2_COMMAND("C0");
            }
          }
        #endif 
```
This compile but I have no way to test it.

SidDrP, 2020-10-02:
Thank you. Just tried it and it compiled. Unfortunately DUE-MMU serial communication again started running circles around me. I will test it on Mega2560 setup once it's done with printing. Will report back if issue any arises. 👍

closed 2020-10-02 by thisiskeithb