MarlinFirmware/Marlin #19576: Failure to compile 2.0.7 for AnyCubic Predator

rq3, 2020-10-01:
Marlin 2.0.7 will not compile for the Trigorilla Pro board used on the AnyCubic Predator delta machine.
It throws the error:

#error "FSMC_UPSCALE is now GRAPHICAL_TFT_UPSCALE. Please update your Configuration.h."

This is actually found in configuration_adv.h, and after fixing the issue, the compile is successful, with the following warnings:

warning: 'adc_result' initialized and declared 'extern'
 extern volatile unsigned int adc_result = 0;
 
warning: implicit declaration of function 'systick_attach_callback' [-Wimplicit-function-declaration]
  systick_attach_callback(&xPortSysTickHandler);

warning: 'prvSetupTimerInterrupt' defined but not used [-Wunused-function]
 void prvSetupTimerInterrupt( void )

These warnings do not prevent uploading the compiled firmware.bin to the machine, but the display is then inverted, and the touch buttons mis-aligned and/or of the wrong color.

There are also safety issues with the maximum bed and hot end temperatures, which can lead to overheating the stock hardware.
The configuration.h in the github repository "config\examples\delta\anycubic\predator" also has an incorrect delta height, which prohibits the machine from successfully calibrating.

The attached files address and fix all of these issues.
[Trigorilla_Pro.zip](https://github.com/MarlinFirmware/Marlin/files/5314820/Trigorilla_Pro.zip)

boelle, 2020-10-01:
have you tried vscode+platform.io extension? (i think you did)

as for the fix, please submit it as a PR, then the maintainers can quickly have a look and add it

ellensp, 2020-10-02:
"FSMC_UPSCALE is now GRAPHICAL_TFT_UPSCALE. Please update your Configuration.h." is the user using an old configuration file on latest marlin. It is not an error. It is a deliberate notification that you need to update this keyword.
The rest are warnings in the framework, outside the scope of marlin.

thisiskeithb, 2020-10-02:
@ellensp: The issue is with the config in the Marlin Configurations repo. They don't have the latest settings: https://github.com/MarlinFirmware/Configurations/blob/51efea51915f05e8855d044ae3408bc5d3e2bc10/config/examples/delta/Anycubic/Predator/Configuration_adv.h#L1543

rq3, 2020-10-03:
@thisiskeithb: Exactly. The example configuration files in the current repositories, for both bugfix and 2.0.7, do not compile under PlatformIO in VSCode. As I mentioned, the error that is thrown suggests the fix is in configuration.h. It is not, it is in configuration_adv.h

Even after the fix is made and the compile is successful, and the resulting firmware.bin is installed on the board, there are errors that prevent the machine from calibrating, displaying information correctly, or may cause the machine to catch fire.

The zip file I provided fixed all this, albeit perhaps not in the most satisfactory manner. But, the dismissive response that, in effect the error is known but left as an exercise for the end user, or that the repos couldn't possibly be out of date, is disheartening. In future, I'll think twice about reporting issues or attempting to provide fixes for them.

thisiskeithb, 2020-10-03:
> there are errors that prevent the machine from calibrating, displaying information correctly, or may cause the machine to catch fire.

Which errors? Maybe @TheNitek has some ideas since they originally submitted the Predator configs in https://github.com/MarlinFirmware/Configurations/pull/157

TheNitek, 2020-10-03:
The UI code kept changing faster then I could fix the config, so gave up on it, sorry.

rq3, 2020-10-03:
@nitek: No problem, I sympathize. My fixes are certainly sub-optimal, 
but they do work and may provide a launching point for
someone who actually knows how to code (not me!).

sjasonsmith, 2020-10-11:
I posted a PR to fix the compilation issue. You've provided a whole long list of other concerns. Since you actually have the printer it would be best for you to actually contribute changes back to the example yourself, since you are able to test the changes.

[Marlin Example Configuration Contribution Instructions](https://github.com/MarlinFirmware/Configurations/blob/import-2.0.x/.github/contributing.md)

sjasonsmith, 2020-10-26:
I closed the other PR as it got out of date. Here is another one that re-configures the TFT for the Predator:
https://github.com/MarlinFirmware/Configurations/pull/288

I don't actually have this printer, so it would be great if one of you could try it out.

twagler, 2020-11-15:
If the intent is for the config to work on unmodified stock Predators, you'll need
`#define TFT_ROTATION TFT_ROTATE_180`

sjasonsmith, 2020-11-15:
@twagler I don't have access to the printer, I had to make some educated guesses in order to fix broken compilation. If you are willing, please go ahead and post a pull-request to the `import-2.0.x` branch of the Configurations repo to update the configuration to work properly.

sjasonsmith, 2020-11-15:
@rq3, are you able to verify that (1) the example can now build properly and (2) the suggested change from @twagler is correct?

twagler, 2020-11-15:
Sorry, there's another problem with Configuration_adv.h.  Line 1060 should change from:
`#if IS_ULTIPANEL`
to
`#if ENABLED(IS_ULTIPANEL)`

thisiskeithb, 2020-11-15:
`#if IS_ULTIPANEL` is correct for the latest Marlin.

https://github.com/MarlinFirmware/Marlin/blob/cab83ba840192ec09ba527372055cfa8810ecaf4/Marlin/Configuration_adv.h#L1055-L1062

twagler, 2020-11-15:
@thisiskeithb, my apologies, I was based against 2.0.7.2.  For commits, should I base against bugfix-2.0.x or 2.0.x?

thisiskeithb, 2020-11-15:
PRs should go against the `bugfix-2.0.x` branch for this repo & [`import-2.0.x` in the Configurations repo](https://github.com/MarlinFirmware/Configurations).

twagler, 2020-11-15:
Thanks.  I submitted a PR.  For some reason I'm not able to get it to show up when with the hashtag.

boelle, 2020-11-15:
its here: https://github.com/MarlinFirmware/Configurations/pull/328

sjasonsmith, 2020-11-16:
@twagler thanks for taking care of that.
I know that issue hasn't merged yet, but since it is a minor fix and it is already pending in that PR, I'm going to go ahead and close out this bug report.

closed 2020-11-16 by sjasonsmith