MarlinFirmware/Marlin #19578: [BUG] 2.0.7 does not compile - error: 'class MarlinUI' has no member named 'preheat_hotend_temp'

Veritas83, 2020-10-01:
### Bug Description

does not compile - error: 'class MarlinUI' has no member named 'preheat_hotend_temp'

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5315210/Marlin.zip)

### Steps to Reproduce

1. Download Marlin-2.0.x.zip
2. Extract
3. Configure
4. Compile

**Expected behavior:** Compiles successfully.

**Actual behavior:** Does not compile successfully.

<pre>
sketch\src\module\configuration_store.cpp: In static member function 'static void MarlinSettings::reset()':
sketch\src\module\configuration_store.cpp:2544:8: error: 'class MarlinUI' has no member named 'preheat_hotend_temp'
     ui.preheat_hotend_temp[0] = PREHEAT_1_TEMP_HOTEND;
        ^~~~~~~~~~~~~~~~~~~
sketch\src\module\configuration_store.cpp:2545:8: error: 'class MarlinUI' has no member named 'preheat_hotend_temp'
     ui.preheat_hotend_temp[1] = PREHEAT_2_TEMP_HOTEND;
        ^~~~~~~~~~~~~~~~~~~
sketch\src\module\configuration_store.cpp:2546:8: error: 'class MarlinUI' has no member named 'preheat_bed_temp'
     ui.preheat_bed_temp[0] = PREHEAT_1_TEMP_BED;
        ^~~~~~~~~~~~~~~~
sketch\src\module\configuration_store.cpp:2547:8: error: 'class MarlinUI' has no member named 'preheat_bed_temp'
     ui.preheat_bed_temp[1] = PREHEAT_2_TEMP_BED;
        ^~~~~~~~~~~~~~~~
sketch\src\module\configuration_store.cpp:2548:8: error: 'class MarlinUI' has no member named 'preheat_fan_speed'
     ui.preheat_fan_speed[0] = PREHEAT_1_FAN_SPEED;
        ^~~~~~~~~~~~~~~~~
sketch\src\module\configuration_store.cpp:2549:8: error: 'class MarlinUI' has no member named 'preheat_fan_speed'
     ui.preheat_fan_speed[1] = PREHEAT_2_FAN_SPEED;
        ^~~~~~~~~~~~~~~~~
In file included from c:\users\vtstech\appdata\local\temp\arduino_build_297232\sketch\src\core\boards.h:24:0,
                 from c:\users\vtstech\appdata\local\temp\arduino_build_297232\sketch\src\inc\MarlinConfigPre.h:35,
                 from c:\users\vtstech\appdata\local\temp\arduino_build_297232\sketch\src\inc\marlinconfig.h:28,
                 from sketch\src\module\configuration_store.h:24,
                 from sketch\src\module\configuration_store.cpp:47:
sketch\src\module\configuration_store.cpp: In static member function 'static void MarlinSettings::report(bool)':
sketch\src\module\configuration_store.cpp:3076:28: error: 'class MarlinUI' has no member named 'preheat_hotend_temp'
       LOOP_L_N(i, COUNT(ui.preheat_hotend_temp)) {
                            ^
c:\users\vtstech\appdata\local\temp\arduino_build_297232\sketch\src\core\macros.h:253:58: note: in definition of macro 'LOOP_S_L_N'
 #define LOOP_S_L_N(VAR, S, N) for (uint8_t VAR=(S); VAR<(N); VAR++)
                                                          ^
sketch\src\module\configuration_store.cpp:3076:7: note: in expansion of macro 'LOOP_L_N'
       LOOP_L_N(i, COUNT(ui.preheat_hotend_temp)) {
       ^~~~~~~~
sketch\src\module\configuration_store.cpp:3076:19: note: in expansion of macro 'COUNT'
       LOOP_L_N(i, COUNT(ui.preheat_hotend_temp)) {
                   ^~~~~
sketch\src\module\configuration_store.cpp:3076:28: error: 'class MarlinUI' has no member named 'preheat_hotend_temp'
       LOOP_L_N(i, COUNT(ui.preheat_hotend_temp)) {
                            ^
c:\users\vtstech\appdata\local\temp\arduino_build_297232\sketch\src\core\macros.h:253:58: note: in definition of macro 'LOOP_S_L_N'
 #define LOOP_S_L_N(VAR, S, N) for (uint8_t VAR=(S); VAR<(N); VAR++)
                                                          ^
sketch\src\module\configuration_store.cpp:3076:7: note: in expansion of macro 'LOOP_L_N'
       LOOP_L_N(i, COUNT(ui.preheat_hotend_temp)) {
       ^~~~~~~~
sketch\src\module\configuration_store.cpp:3076:19: note: in expansion of macro 'COUNT'
       LOOP_L_N(i, COUNT(ui.preheat_hotend_temp)) {
                   ^~~~~
[... 60 lines cut ...]
c:\users\vtstech\appdata\local\temp\arduino_build_297232\sketch\src\core\serial.h:175:45: note: in expansion of macro '_SELP_4'
 #define _SELP_6(a,b,V...)  do{ _SEP_2(a,b); _SELP_4(V); }while(0)
                                             ^~~~~~~
c:\users\vtstech\appdata\local\temp\arduino_build_297232\sketch\src\core\serial.h:177:45: note: in expansion of macro '_SELP_6'
 #define _SELP_8(a,b,V...)  do{ _SEP_2(a,b); _SELP_6(V); }while(0)
                                             ^~~~~~~
c:\users\vtstech\appdata\local\temp\arduino_build_297232\sketch\src\core\serial.h:168:28: note: in expansion of macro '_SELP_8'
 #define __SELP_N(N,V...)   _SELP_##N(V)
                            ^~~~~~
c:\users\vtstech\appdata\local\temp\arduino_build_297232\sketch\src\core\serial.h:169:28: note: in expansion of macro '__SELP_N'
 #define _SELP_N(N,V...)    __SELP_N(N,V)
                            ^~~~~~~~
c:\users\vtstech\appdata\local\temp\arduino_build_297232\sketch\src\core\serial.h:201:33: note: in expansion of macro '_SELP_N'
 #define SERIAL_ECHOLNPAIR(V...) _SELP_N(NUM_ARGS(V),V)
                                 ^~~~~~~
sketch\src\module\configuration_store.cpp:3078:9: note: in expansion of macro 'SERIAL_ECHOLNPAIR'
         SERIAL_ECHOLNPAIR(
         ^~~~~~~~~~~~~~~~~
exit status 1
Error compiling for board Arduino Mega or Mega 2560.
</pre>
#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

mikemerryguy, 2020-10-01:
I have a success, be sure to download current 2.0.7 and update VSS code and Platformoio and all plugins and add ons

```

Linking .pio\build\mega2560\firmware.elf
Building .pio\build\mega2560\firmware.hex
Checking size .pio\build\mega2560\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [====      ]  35.0% (used 2871 bytes from 8192 bytes)
Flash: [====      ]  36.3% (used 92208 bytes from 253952 bytes)
======================================= [SUCCESS] Took 35.90 seconds =======================================
Environment    Status    Duration
-------------  --------  ------------
mega2560       SUCCESS   00:00:35.896
======================================= 1 succeeded in 00:00:35.896 ======================================= 

Terminal will be reused by tasks, press any key to close it.
```

mikemerryguy, 2020-10-01:
[Marlin-2.0.x - Copy.zip](https://github.com/MarlinFirmware/Marlin/files/5315267/Marlin-2.0.x.-.Copy.zip)

your configs my marlin download from today

Veritas83, 2020-10-02:
It was working previously with just Arduino IDE ... thou now not even that is working. So it must be something about my environment. I will try Visual Studio Code w\ Platformio

thisiskeithb, 2020-10-02:
Your configs build fine in VSCode with PlatformIO.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-10-02 by thisiskeithb

mikemerryguy, 2020-10-02:
@Veritas83 
I uploaded a zip with compiled hex if you can't get it to work. 
It is not modified at all from what you uploaded for configs!

Veritas83, 2020-10-02:
It works, Even in Arduino now. I didn't have Platform.IO installed. Even having it installed in VS makes it work/compile in Arduino IDE

Just got my new board all jumpered and wired up. FW successfully flashed. About to test out the motors and thermistors now

Veritas83, 2020-10-02:
Well. it's back. Not sure what it is about 2.0.7 but it doesn't like to compile in Arduino for me. Or VS either now. I was using my older 2.0.5.3 build to test compiling (which was also failing before...)
--
Compiling .pio\build\mega2560\src\src\module\planner.cpp.o
Marlin\src\module\configuration_store.cpp: In static member function 'static void MarlinSettings::reset()':
Marlin\src\module\configuration_store.cpp:2607:8: error: 'class MarlinUI' has no member named 'preheat_hotend_temp'
     ui.preheat_hotend_temp[0] = PREHEAT_1_TEMP_HOTEND;
        ^
Marlin\src\module\configuration_store.cpp:2608:8: error: 'class MarlinUI' has no member named 'preheat_hotend_temp'
     ui.preheat_hotend_temp[1] = PREHEAT_2_TEMP_HOTEND;
        ^
Marlin\src\module\configuration_store.cpp:2609:8: error: 'class MarlinUI' has no member named 'preheat_bed_temp'
     ui.preheat_bed_temp[0] = PREHEAT_1_TEMP_BED;
        ^
Marlin\src\module\configuration_store.cpp:2610:8: error: 'class MarlinUI' has no member named 'preheat_bed_temp'
     ui.preheat_bed_temp[1] = PREHEAT_2_TEMP_BED;
        ^
Marlin\src\module\configuration_store.cpp:2611:8: error: 'class MarlinUI' has no member named 'preheat_fan_speed'
     ui.preheat_fan_speed[0] = PREHEAT_1_FAN_SPEED;
        ^
compilation terminated due to -fmax-errors=5.
*** [.pio\build\mega2560\src\src\module\configuration_store.cpp.o] Error 1
===================================================== [FAILED] Took 41.21 seconds =====================================================
Environment    Status    Duration
-------------  --------  ------------
mega2560       FAILED    00:00:41.208
================================================ 1 failed, 0 succeeded in 00:00:41.208 ================================================The terminal process "C:\Users\VTSTech\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

ellensp, 2020-10-02:
configuration_store.cpp does not exist in current marlins It was replaced with settings.cpp
You have old code. Perhaps you extracted marlin over and old version?

Veritas83, 2020-10-02:
It does appear that was the issue. Older source files that no longer exist in current.

I'm able to compile both 1.1.x and 2.0.x now