MarlinFirmware/Marlin #19579: #define Z_AFTER_HOMING  10  appears to be ignored

xeddog, 2020-10-02:


ellensp, 2020-10-02:
Didn't follow template, no config files, nothing to go on. Closing.

xeddog, 2020-10-02:
Dang it!  I don't know what happened, but I had everything in it.  Configs, descriptions, even a video.  I'll try again tomorrow.

qwewer0, 2020-10-02:
Copied from Discord, not everything, but better than nothing:

### Bug Description
When homing, with or without safe homing, the Z axis completes the double "probe", but the nozzle stays ON the bed.

### My Configurations
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5315555/Marlin.zip)

### Steps to Reproduce
Just home the machine

Expected behavior: [What you expect to happen]
Expect the printer to home, and then raise Z_AFTER_HOMING distance

Actual behavior: [What actually happens]
Printer homes but does not...

xeddog, 2020-10-02:
qwewer0 - Thank you for copying what you could get.  

That is the gist of it.   In addition -

My printer is a Monoprice Mini V1 with a lot of mods, including a 32-bit controller and TMC2209 drivers.  There is no probe, and no levelling of any kind defined in Marlin.  Not even LCD levelling.  Since the bed is only 120mm square, manual levelling is about as easy as it gets.  This behavior has been around with several versions of Marlin 2.0, but lately it's 2.0.6.1, 2.0.7, and bugfix-2.0.x updated a few days ago.  

Also, re-reading the bug description I said "with or without safe homing", but I have had safe_homing defined for a long time.  I don't think normal homing does the multi-probe, but I'm not sure about that.  In any event, it still says on the bed and does not raise Z.

I also have two other printers that DO have probes and levelling, and those do raise the Z axis after homing.

The attached configs are the last of maybe 5 attempts to get this working.

ellensp, 2020-10-03:
looking at the code, the function used to move z up is not defined unless you have a probe.. so yes indeed it does seem to be ignored when there is no probe...    

I moved the function out of the #ifdef block.  It compiles but I am not in a position to test on a actual machine atm.
  
Carefully give this this new probe.h a try  [probe.zip](https://github.com/MarlinFirmware/Marlin/files/5321214/probe.zip)
Unzip and put in Marlin/Marlin/src/module/probe.h

xeddog, 2020-10-03:
YAY.  That worked, Thank you.  I'll try it out on one of my other printers "just to make sure".

xeddog, 2020-10-03:
I just tried this on my CR-10S which DOES have a probe and levelling, and now it does not raise Z after homing.  I will go through my Marlin config to make sure I didn't miss anything, but I think this is the case.  Just traded one for the other.

xeddog, 2020-10-03:
Nevermind my previous post.  I am pretty sure it works.  I got a fresh copy of Marlin 2.0.7 (I used bugfix before) and configured it for my CR-10S using settings from my "old" version as a guide.  I will just make sure on my Ender 5 Plus too.

ellensp, 2020-10-04:
I created a PR so it will be fixed for all, moving forward.

xeddog, 2020-10-04:
I'll close this as soon as I double confirm with my Ender 5 Plus (SKR Pro controller/TMC2209's and EZABL probe if that matters) tomorrow.

xeddog, 2020-10-04:
ellensp - I hovered over your id and is says you are want boards to work with.  I don't suppose there is a place to send one that could be used by all developers, or at least pooled somehow????

ellensp, 2020-10-04:
Unfortunately no. We all work independently mostly from home (but some from their work) and we are planet wide. 
Remember most of us are volunteers, working in our own time.

xeddog, 2020-10-04:
That's what I expected but I was hoping anyway.  Anyway, I have an SKR Pro V1.1 laying around not doing anything.  It works EXCEPT the mosfets for fan0 and fan1 blew, but fan2 still works.   I am just keeping it as a backup, but now I think if I needed it I might just buy something anyway.

ellensp, 2020-10-04:
Well If you want to send it. Contact me Directly we can work out the details.    My E-mail is currently on my github profile  (but wont be for long!)  
I don't have access to this type of controller board at all.

xeddog, 2020-10-04:
Update:   Tested on my E5+ which also has probe and levelling defined.  I had a little scare when the first time I tried it, it didn't lift.  A quick check of the config revealed that it wasn't enabled on this machine, so enabling and re-flashing and all is good.  Closing.

closed 2020-10-04 by xeddog