MarlinFirmware/Marlin #19585: [BUG] Baby-stepping screen not accepting user change

appliedmike, 2020-10-02:


ellensp, 2020-10-02:
Hasn't followed template, no config files, nothing to go on. Closing.

sjasonsmith, 2020-10-02:
That is strange. It had content in the Discord bot notification. @appliedmike did you actually come back here and delete all the content?

qwewer0, 2020-10-02:
This isn't the first one. https://github.com/MarlinFirmware/Marlin/issues/19579

Wouldn't deleting the content leave an edit mark?

ellensp, 2020-10-02:
@sjasonsmith  that is intersting. Discord does have some info...  and issue doesn't say edited..  broken github??

Re posting here.

> 
> - Tried latest this morning 02/oct
> 
> ### Bug Description
> Since update to latest version 2.0.7 baby stepping screen is displayed but not change when pressing + or -
> I have downloaded the bugfix ( 02 oct )
> 
> ### My Configurations
> [Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5317210/Marlin.zip) 
>

appliedmike, 2020-10-02:
Hi All,@sjasonsmith, No I tried to follow the requested format and did not edit once submitted Sent from Mail for Windows 10 From: Jason SmithSent: Friday, 2 October 2020 11:14 PMTo: MarlinFirmware/MarlinCc: Michael Christie; MentionSubject: Re: [MarlinFirmware/Marlin] [BUG] Baby-stepping screen not accepting user change (#19585) That is strange. It had content in the Discord bot notification. @appliedmike did you actually come back here and delete all the content?

	
        
		Virus-free. www.avast.com

appliedmike, 2020-10-02:
Hey All,
@sjasonsmith, no I hadn't edited anything once I had posted. Being in the southern hemisphere I just waited for a glimpse of brilliant to my issue once everyone woke up :-) 
I followed the template as fully as possible

appliedmike, 2020-10-10:
Working as at 2.0.7.1 :-)

closed 2020-10-10 by appliedmike