MarlinFirmware/Marlin #19590: [BUG] SRAM_EEPROM_EMULATION does not save

chrstrvs, 2020-10-02:
### Bug Description

When choosing SRAM_EEPROM_EMULATION in 2.0.7 settings does not save. When powering off the board and powering it on again I get the message "EEPROM Version Error. Initialize EEPROM?" even though I have chosen Reset the last time the question came up.
The board I use is BTT SKR Pro 1.2 and when I used 2.0.5.3 I used FLASH_EEPROM_EMULATION, but when trying to compile 2.0.7
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5319444/Marlin.zip)
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5319448/Marlin.zip)

 with FLASH_EEPROM_EMULATION I get the error message saying "#warning "FLASH_EEPROM_EMULATION may cause long delays when writing and should not be used while printing." [-Wcpp]".

### My Configurations

Are attached.

### Steps to Reproduce

1. Compile 2.0.7 with SRAM_EEPROM_EMULATION enabled.
2. Boot board.
3. Save settings.
4. Power off and then power on board.

**Expected behavior:**
I expect my settings to be saved so they can be loaded the next time I boot the board.

**Actual behavior:**
Settings does not save and I have to manually change M851 Z for example.

#### Additional Information

I have none.

thisiskeithb, 2020-10-02:
Here's the warning & error message: 
https://github.com/MarlinFirmware/Marlin/blob/e8177735a00d1d7bb20b439544d3324fe01cfe7e/Marlin/src/HAL/STM32/inc/SanityCheck.h#L43-L46

You can/should still use `FLASH_EEPROM_EMULATION`, but you should disable `PRINTCOUNTER`.

---
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-10-02 by thisiskeithb

chrstrvs, 2020-10-02:
Got it, thanks!

sjasonsmith, 2020-10-02:
If SRAM emulation works at all, it is probably volatile and requires your board to have an SRAM backup battery.

chrstrvs, 2020-10-03:
> 
> 
> If SRAM emulation works at all, it is probably volatile and requires your board to have an SRAM backup battery.

You learn something new everyday. Thanks!