MarlinFirmware/Marlin #19592: [BUG] #19468 breaks teensy2/AT90USB architectures

fiveangle, 2020-10-02:
#19468 broken teensy2 architectures:

Using Printrboard RevF configs: https://github.com/fiveangle/Marlin/tree/bisect5/Marlin

Commit 90bc1993b6811623dd9d0de00251e0c63415cb49 results in architecture not detected correctly:

```
Processing at90usb1286_dfu (platform: teensy; extends: env:at90usb1286_cdc; board: at90usb1286; board_build.f_cpu: 16000000L; src_filter: +<src/*> -<src/config> -<src/HAL> +<src/HAL/shared>, -<src/lcd/HD44780> -<src/lcd/dwin> -<src/lcd/dogm> -<src/lcd/tft>, -<src/lcd/menu>, -<src/lcd/menu/game/game.cpp> -<src/lcd/menu/game/brickout.cpp> -<src/lcd/menu/game/invaders.cpp>, -<src/lcd/menu/game/maze.cpp> -<src/lcd/menu/game/snake.cpp>, -<src/lcd/menu/menu_backlash.cpp>, -<src/lcd/menu/menu_bed_corners.cpp>, -<src/lcd/menu/menu_bed_leveling.cpp>, -<src/lcd/menu/menu_cancelobject.cpp>, -<src/lcd/menu/menu_delta_calibrate.cpp>, -<src/lcd/menu/menu_filament.cpp>, -<src/lcd/menu/menu_info.cpp>, -<src/lcd/menu/menu_job_recovery.cpp>, -<src/lcd/menu/menu_led.cpp>, -<src/lcd/menu/menu_media.cpp>, -<src/lcd/menu/menu_mmu2.cpp>, -<src/lcd/menu/menu_password.cpp>, -<src/lcd/menu/menu_power_monitor.cpp>, -<src/lcd/menu/menu_spindle_laser.cpp>, -<src/lcd/menu/menu_temperature.cpp>, -<src/lcd/menu/menu_tmc.cpp>, -<src/lcd/menu/menu_touch_screen.cpp>, -<src/lcd/menu/menu_ubl.cpp>, -<src/lcd/extui/lib/mks_ui>, -<src/lcd/extui/lib/dgus> -<src/lcd/extui/dgus_lcd.cpp>, -<src/lcd/extui/example.cpp>, -<src/lcd/extui/malyan_lcd.cpp>, -<src/lcd/extui/lib/ftdi_eve_touch_ui>, -<src/lcd/extui/lib/anycubic_i3mega>, -<src/lcd/extui/anycubic_tft.cpp>, -<src/lcd/lcdprint.cpp>, -<src/sd/usb_flashdrive>, -<src/feature/backlash.cpp>, -<src/feature/baricuda.cpp> -<src/gcode/feature/baricuda>, -<src/feature/bedlevel/abl> -<src/gcode/bedlevel/abl>, -<src/feature/bedlevel/mbl> -<src/gcode/bedlevel/mbl>, -<src/feature/bedlevel/ubl> -<src/gcode/bedlevel/ubl>, -<src/feature/binary_stream.cpp> -<src/libs/heatshrink>, -<src/feature/bltouch.cpp>, -<src/feature/cancel_object.cpp> -<src/gcode/feature/cancel>, -<src/feature/caselight> -<src/gcode/feature/caselight>, -<src/feature/closedloop.cpp>, -<src/feature/controllerfan.cpp> -<src/gcode/feature/controllerfan>, -<src/feature/dac> -<src/feature/digipot>, -<src/feature/direct_stepping.cpp> -<src/gcode/motion/G6.cpp>, -<src/feature/e_parser.cpp>, -<src/feature/encoder_i2c.cpp>, -<src/feature/fanmux.cpp>, -<src/feature/filwidth.cpp> -<src/gcode/feature/filwidth>, -<src/feature/fwretract.cpp> -<src/gcode/feature/fwretract>, -<src/feature/host_actions.cpp>, -<src/feature/hotend_idle.cpp>, -<src/feature/joystick.cpp>, -<src/feature/leds/blinkm.cpp>, -<src/feature/leds/leds.cpp>, -<src/feature/leds/pca9533.cpp>, -<src/feature/leds/pca9632.cpp>, -<src/feature/leds/printer_event_leds.cpp>, -<src/feature/leds/tempstat.cpp>, -<src/feature/max7219.cpp>, -<src/feature/mixing.cpp>, -<src/feature/mmu2> -<src/gcode/feature/prusa_MMU2>, -<src/feature/password> -<src/gcode/feature/password>, -<src/feature/pause.cpp>, -<src/feature/power.cpp>, -<src/feature/power_monitor.cpp> -<src/gcode/feature/power_monitor>, -<src/feature/powerloss.cpp> -<src/gcode/feature/powerloss>, -<src/feature/probe_temp_comp.cpp>, -<src/feature/runout.cpp> -<src/gcode/feature/runout>, -<src/feature/snmm.cpp>, -<src/feature/solenoid.cpp>, -<src/feature/spindle_laser.cpp> -<src/gcode/control/M3-M5.cpp>, -<src/feature/tmc_util.cpp> -<src/module/stepper/trinamic.cpp>, -<src/feature/twibus.cpp>, -<src/feature/z_stepper_align.cpp>, -<src/gcode/bedlevel/G26.cpp>, -<src/gcode/bedlevel/G35.cpp>, -<src/gcode/bedlevel/G42.cpp>, -<src/gcode/bedlevel/M420.cpp>, -<src/gcode/calibrate/G33.cpp>, -<src/gcode/calibrate/G34_M422.cpp>, -<src/gcode/calibrate/G76_M871.cpp>, -<src/gcode/calibrate/G425.cpp>, -<src/gcode/calibrate/M12.cpp>, -<src/gcode/calibrate/M48.cpp>, -<src/gcode/calibrate/M100.cpp>, -<src/gcode/calibrate/M425.cpp>, -<src/gcode/calibrate/M666.cpp>, -<src/gcode/calibrate/M852.cpp>, -<src/gcode/config/M43.cpp>, -<src/gcode/config/M217.cpp>, -<src/gcode/config/M218.cpp>, -<src/gcode/config/M221.cpp>, -<src/gcode/config/M281.cpp>, -<src/gcode/config/M302.cpp>, -<src/gcode/config/M305.cpp>, -<src/gcode/config/M540.cpp>, -<src/gcode/control/M7-M9.cpp>, -<src/gcode/control/M211.cpp>, -<src/gcode/control/M605.cpp>, -<src/gcode/feature/advance>, -<src/gcode/feature/camera>, -<src/gcode/feature/i2c>, -<src/gcode/feature/L6470>, -<src/gcode/feature/leds/M150.cpp>, -<src/gcode/feature/leds/M7219.cpp>, -<src/gcode/feature/macro>, -<src/gcode/feature/mixing/M163-M165.cpp>, -<src/gcode/feature/mixing/M166.cpp>, -<src/gcode/feature/pause/G27.cpp>, -<src/gcode/feature/pause/G60.cpp>, -<src/gcode/feature/pause/G61.cpp>, -<src/gcode/feature/pause/M125.cpp>, -<src/gcode/feature/pause/M600.cpp>, -<src/gcode/feature/pause/M603.cpp>, -<src/gcode/feature/pause/M701_M702.cpp>, -<src/gcode/feature/trinamic/M122.cpp>, -<src/gcode/feature/trinamic/M569.cpp>, -<src/gcode/feature/trinamic/M906.cpp>, -<src/gcode/feature/trinamic/M911-M914.cpp>, -<src/gcode/geometry/G17-G19.cpp>, -<src/gcode/geometry/G53-G59.cpp>, -<src/gcode/geometry/M206_M428.cpp>, -<src/gcode/host/M16.cpp>, -<src/gcode/host/M113.cpp>, -<src/gcode/host/M360.cpp>, -<src/gcode/host/M876.cpp>, -<src/gcode/lcd/M0_M1.cpp>, -<src/gcode/lcd/M250.cpp>, -<src/gcode/lcd/M73.cpp>, -<src/gcode/lcd/M995.cpp>, -<src/gcode/motion/G2_G3.cpp>, -<src/gcode/motion/G5.cpp>, -<src/gcode/motion/G80.cpp>, -<src/gcode/motion/M290.cpp>, -<src/gcode/probe/G30.cpp>, -<src/gcode/probe/G31_G32.cpp>, -<src/gcode/probe/G38.cpp>, -<src/gcode/probe/M401_M402.cpp>, -<src/gcode/probe/M851.cpp>, -<src/gcode/probe/M951.cpp>, -<src/gcode/scara>, -<src/gcode/sd>, -<src/gcode/temp/M104_M109.cpp>, -<src/gcode/units/G20_G21.cpp>, -<src/gcode/units/M149.cpp>, -<src/libs/L64XX> -<src/module/stepper/L64xx.cpp>, -<src/libs/hex_print.cpp>, -<src/libs/least_squares_fit.cpp>, -<src/libs/nozzle.cpp> -<src/gcode/feature/clean>, -<src/module/delta.cpp>, -<src/module/planner_bezier.cpp>, -<src/module/printcounter.cpp>, -<src/module/probe.cpp>, -<src/module/scara.cpp> -<src/gcode/calibrate/M665.cpp>, -<src/module/stepper/TMC26X.cpp> +<src/HAL/AVR>; framework: arduino; extra_scripts: pre:buildroot/share/PlatformIO/scripts/common-dependencies.py, pre:buildroot/share/PlatformIO/scripts/common-cxxflags.py, post:buildroot/share/PlatformIO/scripts/common-dependencies-post.py; build_flags: -fmax-errors=5 -g -D__MARLIN_FIRMWARE__ -fmerge-all-constants; lib_deps: ; monitor_speed: 250000; monitor_flags: --quiet, --echo, --eol, LF, --filter, colorize, --filter, time)
--------------------------------------------------------------------------------
In file included from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/pins/pins.h:353:0,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:154:
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/pins/teensy2/pins_PRINTRBOARD_REVF.h:71:4: error: #error "USBCON should be defined by the platform for this board."
   #error "USBCON should be defined by the platform for this board."
    ^
CalledProcessError: Command 'avr-g++ -D__MARLIN_FIRMWARE__ -D__MARLIN_PREBUILD__ -w -dM -E -x c++ buildroot/share/PlatformIO/scripts/common-dependencies.h' returned non-zero exit status 1:
  File "/home/speedster/.local/lib/python2.7/site-packages/platformio/builder/main.py", line 167:
    env.SConscript(item, exports="env")
  File "/home/speedster/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 541:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/speedster/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 250:
    exec _file_ in call_stack[-1].globals
  File "/home/speedster/dev/git/Marlin/buildroot/share/PlatformIO/scripts/common-dependencies.py", line 280:
    apply_features_config()
  File "/home/speedster/dev/git/Marlin/buildroot/share/PlatformIO/scripts/common-dependencies.py", line 112:
    if not env.MarlinFeatureIsEnabled(feature):
  File "/home/speedster/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "/home/speedster/dev/git/Marlin/buildroot/share/PlatformIO/scripts/common-dependencies.py", line 256:
    load_marlin_features()
  File "/home/speedster/dev/git/Marlin/buildroot/share/PlatformIO/scripts/common-dependencies.py", line 244:
    define_list = subprocess.check_output(cmd, shell=True).splitlines()
[... cut ...]
BIGTREE_BTT002            IGNORED
LERDGEX                   IGNORED
LERDGES                   IGNORED
LERDGEK                   IGNORED
rumba32                   IGNORED
esp32                     IGNORED
teensy31                  IGNORED
teensy35                  IGNORED
teensy36                  IGNORED
teensy41                  IGNORED
linux_native              IGNORED
include_tree              IGNORED
==================== 1 failed, 0 succeeded in 00:00:00.285 ====================

[Fri Oct  2 12:52:18 2020]
```

thisiskeithb, 2020-10-02:
Can you update PlatformIO & your libraries and see if it’s still broken? 

Current PIO should only output the environment you’re compiling for in the success/failed section.

ellensp, 2020-10-02:
Under bugfix, with current PIO

> Executing task: platformio run <

Processing at90usb1286_dfu (platform: teensy; board: at90usb1286; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
In file included from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/pins/pins.h:353:0,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:150:
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/pins/teensy2/pins_PRINTRBOARD_REVF.h:71:4: error: #error "USBCON should be defined by the platform for this board."
   #error "USBCON should be defined by the platform for this board."
    ^
CalledProcessError: Command 'avr-g++ -D__MARLIN_FIRMWARE__ -D__MARLIN_PREBUILD__ -w -dM -E -x c++ buildroot/share/PlatformIO/scripts/common-dependencies.h' returned non-zero exit status 1.:
  File "/home/ellensp/.platformio/penv/lib/python3.8/site-packages/platformio/builder/main.py", line 167:
    env.SConscript(item, exports="env")
  File "/home/ellensp/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Script/SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/ellensp/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Script/SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/ellensp/code/Marlin2/Marlin/buildroot/share/PlatformIO/scripts/common-dependencies.py", line 280:
    apply_features_config()
  File "/home/ellensp/code/Marlin2/Marlin/buildroot/share/PlatformIO/scripts/common-dependencies.py", line 112:
    if not env.MarlinFeatureIsEnabled(feature):
  File "/home/ellensp/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "/home/ellensp/code/Marlin2/Marlin/buildroot/share/PlatformIO/scripts/common-dependencies.py", line 256:
    load_marlin_features()
  File "/home/ellensp/code/Marlin2/Marlin/buildroot/share/PlatformIO/scripts/common-dependencies.py", line 244:
    define_list = subprocess.check_output(cmd, shell=True).splitlines()
  File "/usr/lib/python3.8/subprocess.py", line 411:
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.8/subprocess.py", line 512:
    raise CalledProcessError(retcode, process.args,
================================================================= [FAILED] Took 0.57 seconds =================================================================

Environment      Status    Duration
---------------  --------  ------------
at90usb1286_dfu  FAILED    00:00:00.569
============================================================ 1 failed, 0 succeeded in 00:00:00.569 ============================================================
The terminal process "platformio 'run'" terminated with exit code: 1.

fiveangle, 2020-10-02:
had already blown away all libs running latest PIO dev:

```
You're up-to-date!
PlatformIO 5.0.2b2 is currently the newest version available.

```
Configs was as per the commit with the regression.  Regardless, blew all caches/libs away (again), and rebased on latest bf2 head (2e1ba73926eeac126dd345b32f40c5b5d4804462) and no change, other than the now-double-spacing on the platform report:

```
*******************************************************************************
If you like PlatformIO, please:
- follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
- star it on GitHub > https://github.com/platformio/platformio
- try PlatformIO IDE for embedded development > https://platformio.org/platformio-ide
*******************************************************************************

Processing at90usb1286_dfu (platform: teensy; extends: env:at90usb1286_cdc; board: at90usb1286; board_build.f_cpu: 16000000L; src_filter: +<src/*> -<src/config> -<src/HAL> +<src/HAL/shared>, -<src/lcd/HD44780> -<src/lcd/TFTGLCD> -<src/lcd/dwin> -<src/lcd/dogm> -<src/lcd/tft>, -<src/lcd/menu>, -<src/lcd/menu/game/game.cpp> -<src/lcd/menu/game/brickout.cpp> -<src/lcd/menu/game/invaders.cpp>, -<src/lcd/menu/game/maze.cpp> -<src/lcd/menu/game/snake.cpp>, -<src/lcd/menu/menu_backlash.cpp>, -<src/lcd/menu/menu_bed_corners.cpp>, -<src/lcd/menu/menu_bed_leveling.cpp>, -<src/lcd/menu/menu_cancelobject.cpp>, -<src/lcd/menu/menu_delta_calibrate.cpp>, -<src/lcd/menu/menu_filament.cpp>, -<src/lcd/menu/menu_info.cpp>, -<src/lcd/menu/menu_job_recovery.cpp>, -<src/lcd/menu/menu_led.cpp>, -<src/lcd/menu/menu_media.cpp>, -<src/lcd/menu/menu_mmu2.cpp>, -<src/lcd/menu/menu_password.cpp>, -<src/lcd/menu/menu_power_monitor.cpp>, -<src/lcd/menu/menu_spindle_laser.cpp>, -<src/lcd/menu/menu_temperature.cpp>, -<src/lcd/menu/menu_tmc.cpp>, -<src/lcd/menu/menu_touch_screen.cpp>, -<src/lcd/menu/menu_ubl.cpp>, -<src/lcd/extui/lib/mks_ui>, -<src/lcd/extui/lib/dgus> -<src/lcd/extui/dgus_lcd.cpp>, -<src/lcd/extui/example.cpp>, -<src/lcd/extui/malyan_lcd.cpp>, -<src/lcd/extui/lib/ftdi_eve_touch_ui>, -<src/lcd/extui/anycubic_chiron_lcd.cpp>, -<src/lcd/extui/anycubic_i3mega_lcd.cpp> -<src/lcd/extui/lib/anycubic_i3mega>, -<src/lcd/lcdprint.cpp>, -<src/sd/usb_flashdrive>, -<src/feature/backlash.cpp>, -<src/feature/baricuda.cpp> -<src/gcode/feature/baricuda>, -<src/feature/bedlevel/abl> -<src/gcode/bedlevel/abl>, -<src/feature/bedlevel/mbl> -<src/gcode/bedlevel/mbl>, -<src/feature/bedlevel/ubl> -<src/gcode/bedlevel/ubl>, -<src/feature/binary_stream.cpp> -<src/libs/heatshrink>, -<src/feature/bltouch.cpp>, -<src/feature/cancel_object.cpp> -<src/gcode/feature/cancel>, -<src/feature/caselight> -<src/gcode/feature/caselight>, -<src/feature/closedloop.cpp>, -<src/feature/controllerfan.cpp> -<src/gcode/feature/controllerfan>, -<src/feature/dac> -<src/feature/digipot>, -<src/feature/direct_stepping.cpp> -<src/gcode/motion/G6.cpp>, -<src/feature/e_parser.cpp>, -<src/feature/encoder_i2c.cpp>, -<src/feature/fanmux.cpp>, -<src/feature/filwidth.cpp> -<src/gcode/feature/filwidth>, -<src/feature/fwretract.cpp> -<src/gcode/feature/fwretract>, -<src/feature/host_actions.cpp>, -<src/feature/hotend_idle.cpp>, -<src/feature/joystick.cpp>, -<src/feature/leds/blinkm.cpp>, -<src/feature/leds/leds.cpp>, -<src/feature/leds/pca9533.cpp>, -<src/feature/leds/pca9632.cpp>, -<src/feature/leds/printer_event_leds.cpp>, -<src/feature/leds/tempstat.cpp>, -<src/feature/max7219.cpp>, -<src/feature/mixing.cpp>, -<src/feature/mmu2> -<src/gcode/feature/prusa_MMU2>, -<src/feature/password> -<src/gcode/feature/password>, -<src/feature/pause.cpp>, -<src/feature/power.cpp>, -<src/feature/power_monitor.cpp> -<src/gcode/feature/power_monitor>, -<src/feature/powerloss.cpp> -<src/gcode/feature/powerloss>, -<src/feature/probe_temp_comp.cpp>, -<src/feature/runout.cpp> -<src/gcode/feature/runout>, -<src/feature/snmm.cpp>, -<src/feature/solenoid.cpp>, -<src/feature/spindle_laser.cpp> -<src/gcode/control/M3-M5.cpp>, -<src/feature/tmc_util.cpp> -<src/module/stepper/trinamic.cpp>, -<src/feature/twibus.cpp>, -<src/feature/z_stepper_align.cpp>, -<src/gcode/bedlevel/G26.cpp>, -<src/gcode/bedlevel/G35.cpp>, -<src/gcode/bedlevel/G42.cpp>, -<src/gcode/bedlevel/M420.cpp>, -<src/gcode/calibrate/G33.cpp>, -<src/gcode/calibrate/G34_M422.cpp>, -<src/gcode/calibrate/G76_M871.cpp>, -<src/gcode/calibrate/G425.cpp>, -<src/gcode/calibrate/M12.cpp>, -<src/gcode/calibrate/M48.cpp>, -<src/gcode/calibrate/M100.cpp>, -<src/gcode/calibrate/M425.cpp>, -<src/gcode/calibrate/M666.cpp>, -<src/gcode/calibrate/M852.cpp>, -<src/gcode/config/M43.cpp>, -<src/gcode/config/M217.cpp>, -<src/gcode/config/M218.cpp>, -<src/gcode/config/M221.cpp>, -<src/gcode/config/M281.cpp>, -<src/gcode/config/M302.cpp>, -<src/gcode/config/M305.cpp>, -<src/gcode/config/M540.cpp>, -<src/gcode/control/M7-M9.cpp>, -<src/gcode/control/M211.cpp>, -<src/gcode/control/M605.cpp>, -<src/gcode/feature/advance>, -<src/gcode/feature/camera>, -<src/gcode/feature/i2c>, -<src/gcode/feature/L6470>, -<src/gcode/feature/leds/M150.cpp>, -<src/gcode/feature/leds/M7219.cpp>, -<src/gcode/feature/macro>, -<src/gcode/feature/mixing/M163-M165.cpp>, -<src/gcode/feature/mixing/M166.cpp>, -<src/gcode/feature/pause/G27.cpp>, -<src/gcode/feature/pause/G60.cpp>, -<src/gcode/feature/pause/G61.cpp>, -<src/gcode/feature/pause/M125.cpp>, -<src/gcode/feature/pause/M600.cpp>, -<src/gcode/feature/pause/M603.cpp>, -<src/gcode/feature/pause/M701_M702 [... cut ...]
    if not env.MarlinFeatureIsEnabled(feature):
[... 150 lines cut ...]

LERDGES                   IGNORED

LERDGEK                   IGNORED

rumba32                   IGNORED

esp32                     IGNORED

teensy31                  IGNORED

teensy35                  IGNORED

teensy36                  IGNORED

teensy41                  IGNORED

linux_native              IGNORED

include_tree              IGNORED

==================== 1 failed, 0 succeeded in 00:00:35.406 ====================

[10/02/20 15:24:35]
```

fiveangle, 2020-10-02:
Reverting to PIO 5.0.1 from "standard" track and it behaves as yours, so another red herring:

```
You're up-to-date!
PlatformIO 5.0.1 is currently the newest version available.
...

...
Environment      Status    Duration

---------------  --------  ------------

at90usb1286_dfu  FAILED    00:00:24.914

==================== 1 failed, 0 succeeded in 00:00:24.914 ====================

[10/02/20 15:50:10]
```

sjasonsmith, 2020-10-10:
There are a couple different issues impacting this build.

@rhapsodyv I'm running into an issue involving library dependencies. The teensy framework has a library called Teensy_ADC, which is incompatible with this board. For some reason it is selected on Windows  (in VS Code), but not in Linux!

Do you know what could cause these libraries to be selected differently on different platforms? I suspect these are coming from PlatformIO, and not directly from the dependency code you worked on.

Windows
```
Dependency Graph
|-- <Wire> 1.0
|-- <SPI> 1.0
|-- <SD> 1.2.2
|   |-- <SPI> 1.0
|-- <EEPROM> 2.0
|-- <Time> 1.6
|-- <Teensy_ADC> 8.0.38
|-- <Servo> 1.1.2
|-- <SoftwareSerial> 1.0
|-- <PWMServo> 2.1
```

Linux (actually WSL on Windows):
```
Dependency Graph
|-- <Wire> 1.0
|-- <SPI> 1.0
|-- <SD> 1.2.2
|   |-- <SPI> 1.0
|-- <EEPROM> 2.0
|-- <Servo> 1.1.2
|-- <SoftwareSerial> 1.0
|-- <PWMServo> 2.1
```

Both Teensy_ADC and Time are selected only on Windows...

sjasonsmith, 2020-10-11:
I just posted a pull request that allows me to build. I have not personally tested it. It would be great if you could try it out, and comment on the pull request.

fiveangle, 2020-10-12:
```
avr-g++ -o .pio/build/at90usb1286_dfu/src/src/HAL/AVR/eeprom.cpp.o -c -Wno-register -fno-exceptions -felide-constructors -std=gnu++11 -fpermissive -fsingle-precision-constant -fmax-errors=5 -g -fmerge-all-constants -Os -Wall -ffunction-sections -fdata-sections -mmcu=at90usb1286 -DPLATFORMIO=50001 -DTEENSY2PP -D__MARLIN_FIRMWARE__ -DUSB_SERIAL -DARDUINO=10805 -DTEENSYDUINO=153 -DCORE_TEENSY -DF_CPU=16000000L -DLAYOUT_US_ENGLISH -IMarlin -I/home/speedster/.platformio/packages/framework-arduinoteensy/libraries/Servo -I/home/speedster/.platformio/packages/framework-arduinoteensy/libraries/EEPROM -I/home/speedster/.platformio/packages/framework-arduinoteensy/libraries/SoftwareSerial -I/home/speedster/.platformio/packages/framework-arduinoteensy/libraries/PWMServo -I/home/speedster/.platformio/packages/framework-arduinoteensy/libraries/SD -I/home/speedster/.platformio/packages/framework-arduinoteensy/libraries/SD/utility -I.pio/libdeps/at90usb1286_dfu/U8glib-HAL/src -I/home/speedster/.platformio/packages/framework-arduinoteensy/libraries/Wire -I/home/speedster/.platformio/packages/framework-arduinoteensy/libraries/Wire/utility -I/home/speedster/.platformio/packages/framework-arduinoteensy/libraries/SPI -I/home/speedster/.platformio/packages/framework-arduinoteensy/cores/teensy -I/home/speedster/.platformio/packages/framework-arduinoteensy/cores Marlin/src/HAL/AVR/eeprom.cpp
Marlin/src/HAL/AVR/MarlinSerial.cpp:50:33: error: expected nested-name-specifier before 'MarlinSerial'
 template<typename Cfg> typename MarlinSerial<Cfg>::ring_buffer_r MarlinSerial<Cfg>::rx_buffer = { 0, 0, { 0 } };
                                 ^
Marlin/src/HAL/AVR/MarlinSerial.cpp:50:45: error: expected initializer before '<' token
 template<typename Cfg> typename MarlinSerial<Cfg>::ring_buffer_r MarlinSerial<Cfg>::rx_buffer = { 0, 0, { 0 } };
                                             ^
Marlin/src/HAL/AVR/MarlinSerial.cpp:51:33: error: expected nested-name-specifier before 'MarlinSerial'
 template<typename Cfg> typename MarlinSerial<Cfg>::ring_buffer_t MarlinSerial<Cfg>::tx_buffer = { 0 };
                                 ^
Marlin/src/HAL/AVR/MarlinSerial.cpp:51:45: error: expected initializer before '<' token
 template<typename Cfg> typename MarlinSerial<Cfg>::ring_buffer_t MarlinSerial<Cfg>::tx_buffer = { 0 };
                                             ^
Marlin/src/HAL/AVR/MarlinSerial.cpp:52:45: error: expected initializer before '<' token
 template<typename Cfg> bool     MarlinSerial<Cfg>::_written = false;
                                             ^
compilation terminated due to -fmax-errors=5.
*** [.pio/build/at90usb1286_dfu/src/src/HAL/AVR/MarlinSerial.cpp.o] Error 1
========================== [FAILED] Took 3.01 seconds ==========================
```

sjasonsmith, 2020-10-26:
This platform broke again, and I fixed it again:
https://github.com/MarlinFirmware/Marlin/pull/19885

I've added it into the CI builds so hopefully it will not be broken so easily as Marlin evolves around it.

@fiveangle, can you verify whether it is working for you with the current `bugfix-2.0.x` branch? If I don't hear anything in a few days I will presume it is ok and close this.

sjasonsmith, 2020-10-28:
@fiveangle I am also presuming this is now fixed. I’ll close the issue, but please let me know if there are still issues.

closed 2020-10-28 by sjasonsmith

fiveangle, 2020-11-18:
Resolved by #19907

fiveangle, 2020-12-30:
> I've added it into the CI builds so hopefully it will not be broken so easily as Marlin evolves around it.

Thank you @sjasonsmith. I swear at90usb DFU seems to break every couple months and I just don't spend enough time tinkering with this old printer to keep up with the regressions.  I submitted at CircleCI track to test all sample configs a few years back in order to try and stay atop these, but it was too long a compile for scott to deal with :)

Nice that someone built out the CI of the project to parallelize it all to a reasonable cadence.

Thanks again for all the help.  BTW- another feather in the cap of everyone working on Marlin right now, I just compiled the head of BF2 for this config and it's literally the first time in about 3 years it's compiled w/o unexpected warnings (3yrs ago I went through to fix all of them, which was promptly outdated within weeks - haha).  Things are looking great everyone ! 👍 

-=dave