MarlinFirmware/Marlin #19594: Mks Abase Ethernet Support

mondobizarro, 2020-10-02:
Hello, I have an MKS SBASE 1.3 board with Marlin 2.0, the problem that I'm having is that the network port on the board is not working with Marlin, but it work with other firmware lke Smoothieboard (just to discard board problems)  So I'm guessing that it's a configuration issue, the USB port works fine, but the ethernet port on the board is blinking when connect a network cable, any Ideas?

ellensp, 2020-10-03:
Seems to be a LAN8720 Ethernet Transceiver.
I don't think there is any Ethernet support in Marlin as yet.

ellensp, 2020-10-03:
Worse than that this is a mac level interface, the MCU need a TCP/IP stack...    
Soothieware seems to use this https://en.wikipedia.org/wiki/UIP_(micro_IP)