MarlinFirmware/Marlin #19599: [BUG] Sensorless homing is either too sensitive or not sensitive enough

chrstrvs, 2020-10-03:
### Bug Description

Since upgrading from 2.0.5.3 to 2.0.7 sensorless homing is too sensitive on the Y axis on my corexy printer with TMC2209, using the same X and Y_STALL_SENSITIVITY. If I lower the sensitivity from 120 to 110 the Y axis is fine, but the X axis is not sensitive enough and crashes. Any values between 110 and 120 is either too sensitive on Y or not sensitive enough on X.
Going back to 2.0.5.3 fixes the issue.

### My Configurations

Are attached.

### Steps to Reproduce

**Expected behavior:**
I expect both axes to home without any issues as they did with 2.0.5.3

**Actual behavior:**
Either X is not sensitive enough or Y is too sensitive. Sensorless homing does not work.

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5321574/Marlin.zip)

hackebike, 2020-10-03:
I could not find a good value either with TMC 2209 or 2226 for Y, but for X worked fine.
I feel that sensorless homing/probing should be at high priority since it eliminates a lot of complexity off the printer.

boelle, 2020-10-03:
> since it eliminates a lot of complexity off the printer.

but also is less precice (not my words)

but for my mendel90 it works fine without any issues, sensitivity is set to 62 for both and homing feedrate is 750

chrstrvs, 2020-10-03:
> > since it eliminates a lot of complexity off the printer.
> 
> but also is less precice (not my words)
> 
> but for my mendel90 it works fine without any issues, sensitivity is set to 62 for both and homing feedrate is 750

Please note that I have a corexy machine which uses both motors when homing both x and y. It does not use one motor at a time.

helgjohn, 2020-10-03:
The homing current can be adjusted that will in turn give more sensitivity at least it worked for me so... Look at line     
#define X_CURRENT_HOME  500  // (mA) RMS current for sensorless homing here is where you can make some change to make it work . 
 #if AXIS_IS_TMC(X)
    #define X_CURRENT       680        // (mA) RMS current. Multiply by 1.414 for peak current.
    #define X_CURRENT_HOME  500  // (mA) RMS current for sensorless homing
    #define X_MICROSTEPS     16    // 0..256
    #define X_RSENSE          0.11
    #define X_CHAIN_POS      -1    // <=0 : Not chained. 1 : MCU MOSI connected. 2 : Next in chain, ...
  #endif

Portzal, 2020-10-04:
I have found using homing currant and lower homing speed has help me. I have a Cartesian (TMC2209's) and a CoreXY (TMC2226's) both on recent Bugfix 2.0.x
I have also been playing with Improve homing reliability, but I had to 
Reduced the guard period from 400 to 100 on my CR-10S Pro. 
I will be testing it on my CoreXY soon.

rlojek, 2020-10-04:
Ender 3 SKR Minie 1.2 , After update do 2.0.7 sensor less homing has issues with bump sensitivity values.
For X axis lowering value was  ok 70->68. On Y axis total mess. Either is to high - last working was 66, or is to low - 55, makes bed moving, yet crashes it with ugly sound effect.

chrstrvs, 2020-10-07:
I've tried adjusting the current both up and down. It makes no difference at all. I still can't home using 2.0.7.

Scope666, 2020-10-10:
I have a very similar issue where the FIRST time I try to home X, it will move like 1 mm and stall.  Any subsequent attempts will work perfectly.  This started around 2.0.7.  If I lower the sensitivity to prevent the stall on the first move, the motor will "click" and not shut off when it hits the end stop bumper.

Crunch69, 2020-10-12:
I have the exact same issue as chrstrvs  with a corexy I thought something went wrong with my printer and have spent  days scratching my head after upgrading firmware.

Portzal, 2020-10-12:
I have had a play with a few different options that really don't seem to make much of a difference. sensor less homing is still working with the latest bugfix 2.0.x on both my Cartesian and CoreXY. With the CoreXY I have been keeping the values for X and Y the same.

rlojek, 2020-10-13:
Hello!
I had a bit success with defining different , lower values for homing current , like was advised by @helgjohn in  [https://github.com/MarlinFirmware/Marlin/issues/19599#issuecomment-703162604](url) above . Yet still there are issues with homing when the second print in the row is started. Sometimes m501, that setup TMC registers in the process helps.

chrstrvs, 2020-10-24:
I just tried 2.0.7.2 and the issue persists. I am stuck on 2.0.5.3 if I want to use sensorless homing, which I do, but 2.0.5.3 has the issue where the temperature of a passive chamber is not reported.

talldonkey, 2020-10-29:
I'm running into this also.  On 2.0.5.3 on a Ender-3 with SKR 1.3 and TMC2209, X Sens and Y Sens both set to 50 work very nicely, with lost of error margin, 60 worked too.

Now on 2.0.7.2 my X homes and grinds for several seconds against the endstop.  Y seems fine at 50.  I can't find a correct sensitivity level.  64 = doesn't move, 63 = grinds against endstop.

Scope666, 2020-10-29:
I finally solved this issue by uncommenting this:

#define SENSORLESS_BACKOFF_MM  { 2, 2 }     // (mm) Backoff from endstops before sensorless homing

There's something about getting the motor to move a little in the opposite direction first that prevents that first attempt stall that I was having.

My full commit that fixed things for me:
https://github.com/Scope666/Marlin/commit/81d1d9ffb5525d93f3a34137b9725e124617dbbe

chrstrvs, 2020-10-29:
> 
> 
> I finally solved this issue by uncommenting this:
> 
> #define SENSORLESS_BACKOFF_MM { 2, 2 } // (mm) Backoff from endstops before sensorless homing
> 
> There's something about getting the motor to move a little in the opposite direction first that prevents that first attempt stall that I was having.
> 
> My full commit that fixed things for me:
> [Scope666@81d1d9f](https://github.com/Scope666/Marlin/commit/81d1d9ffb5525d93f3a34137b9725e124617dbbe)

I tried that before posting this bug report. Unfortunately it makes no difference on my machine.

Scope666, 2020-10-29:
The only thing I can think of is compare the rest of my sensorless settings to yours, see if there's anything different that might be the cause.

talldonkey, 2020-10-29:
`#define SENSORLESS_BACKOFF_MM { 2, 2 } // (mm) Backoff from endstops before sensorless homing
` 

Unfortunately this did not help for me either.  My settings, including my commit attempting to fix that is here: https://github.com/talldonkey/Marlin-2.0-Ender-3-SKR-1.3/commit/cd94c1cd39937281e9be98ce2ea3b7f7649b20d8  @Scope666 I'll compare your config with mine to see what the magic combo is.

talldonkey, 2020-11-06:
I got my config fixed and my sensorless homing working again on X.  What did it for me was HOMING_FEEDRATE_XY in configuration.h changed to 50*60 during what probably was my mistake during my merge from 2.0.5.3 to 2.0.7.  I lowered it back to 20*60 and my original sensitivity values of 50/50 for X/Y on Cartesian Ender 3 work great!

#define HOMING_FEEDRATE_XY (50*60)  // (was 20*60)

hackebike, 2020-11-06:
I have an issue where sensorless homing on X works, but Y does not. So would I need to increase the Y speed?
Also does it work for Z too?
I am on BTT GTR and TMC2226.

chrstrvs, 2020-11-08:
I decided to try lowering the HOMING_FEEDRATE_XY from 50*60 (3000) to 1000. I had to lower X and Y_STALL_SENSITIVITY way down, but finally the printer homed both X and Y without any issues. I changed the HOMING_FEEDRATE_XY to 2500 and it still kind of worked with X and Y_STALL_SENSITIVITY a bit lower than the usual 120 that I use, but after X is homed and when Y starts moving something makes a clonking noise that I don't like.

I'm still going to stay on 2.0.5.3 and I still consider this a bug that I hope is fixed at some point.

To the developers of Marlin: Keep up the great work! The fact that we get this amazing firmware for free is nothing less than amazing.

terrafirma2021, 2020-11-10:
Yes i also have this issue, i can home fine unless im at X_MIN or Y_MIN then it will grind

chrstrvs, 2020-11-10:
Hah, I finally figured it out!

For some reason (I don't know if it's a bug in the firmware or a issue with my printer) X will note home if Y is at Y min or Y max, and Y will not home if X is at X min or X max.
In 2.0.5.3 I had HOMING_BACKOFF_MM set to 10mm for X and Y, but not in 2.0.7 because I didn't see the use for it since there is a new function called SENSORLESS_BACKOFF_MM which backs off before homing.

I consider this case closed for me.

closed 2020-11-10 by chrstrvs