MarlinFirmware/Marlin #19602: [BUG] Power loss recovery does not continue printing

cguerrero1205, 2020-10-03:
### Bug Description

My Ender 3 pauses after a power outage, heats the bed and nozzle, returns to where it was before the power outage, but the nozzle stays still at that point.

### My Configurations

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5321982/Marlin.zip)

### Steps to Reproduce

1. Start an impression.
2. I cut the power manually.
3. I turn on the machine again.
4. I click on "resume printing" in the " power loss recovery" menu.
5. The nozzle and bed begin to heat
6. After the temperature is reached, the nozzle moves to the point where it was before the power failure, but stays still at that point, the machine does not move anymore and on the LCD it says "printing", but I said before, the nozzle doesn't move anymore.

**Expected behavior:** Print needs to resume where it left off.

**Actual behavior:** It remains in an eternal pause.

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

katherinehackworth, 2020-10-04:
I was about to make an issue for the exact same issue. It seems like it only does a single print move after it resumes, and then hangs indefinitely. I'm on a SKR Mini E3 v1.2 with the BTT UPS 24V for power loss recovery, and it looks like you're on the stock board. Same exact issue here. Still looking around to see if this is an isolated thing or a bug in 2.0.7. I can add my config files if needed too.

cguerrero1205, 2020-10-04:
> I was about to make an issue for the exact same problem. It seems like it only does a single print move after it resumes, and then hangs indefinitely. I'm on a SKR Mini E3 v1.2 with the BTT UPS 24V for power loss recovery, and it looks like you're on the stock board. Same exact issue here. Still looking around to see if this is an isolated thing or a bug in 2.0.7. I can add my config files if needed too.

Hi Katerine.
I've been checking the configuration of the Marlin I have loaded on my printer, but I can't find any problems. I even reloaded it with the sample configuration files for Ender 3, only activating the Power Loss Recovery, but I still have the same problem.

katherinehackworth, 2020-10-04:
[I've added my config files in a Gist here for posterity's sake.](https://gist.github.com/katherinehackworth/1d8e944c5fe0655753ebb9e71b1c10d2)

cguerrero1205, 2020-10-04:
I will install some previous version of Marlin, to see if it solves the problem.

pkfloyd, 2020-10-05:
hi, 
have the tevo/homers tarantula pro, with MKS Lgen v1 and have de same problem.

cguerrero1205, 2020-10-06:
Hello. 
I still can't solve the problem. How are you doing with the problem?

pkfloyd, 2020-10-06:
hi, 
i have change the  powerloss.cpp and powerloss.h from marlin 2.0.6 to 2.0.7 and it work, now i have porwer_loss resume with no pin (always save in card)

cguerrero1205, 2020-10-06:
> hi,
> i have change the powerloss.cpp and powerloss.h from marlin 2.0.6 to 2.0.7 and it work, now i have porwer_loss resume with no pin (always save in card)

Great! I will try to do the same.

cguerrero1205, 2020-10-06:
Then the problem is in the PLR library. I will try to compare those libraries in Marlin 2.0.6 and 2.0.7, to find the error. Hopefully some official Marlin developer will see this thread and can help.

pkfloyd, 2020-10-06:
i have tried to do this, but don´t now anything about coding  :(

pkfloyd, 2020-10-08:
hi, have you find the issue?

ths file change work?

cguerrero1205, 2020-10-08:
> hi, have you find the issue?
> 
> ths file change work?

Hi, I have spent hours reviewing and comparing the powerloss files, but they made many modifications from version 2.0.6 to 2.0.7 and I haven't found the problem.
Likewise, I changed the files in the PLR library, but now the problem is that the Z axis starts from the wrong height.
As I said before: the hope is that some Marlin developer will see this thread, and hope that they will make some solution.
I've also been looking at the bugfix version, but so far they haven't talked about PLR.

pkfloyd, 2020-10-08:
the two files, its only for PLR, why change your Z axis...? strange...

cguerrero1205, 2020-10-08:
> the two files, its only for PLR, why change your Z axis...? strange...

Yes, I think it may be that, while the printer is off, the Z-axis goes down because the motor is not braked and that causes the position in the Z-axis to be lost.

At the moment, I have the printer running with PLR disabled, praying that the power does not cut again, as in recent days (I lost a print that took 20 of 22 hours).

pkfloyd, 2020-10-08:
the test i made yes, work great.

i will test again.

it locks the ender 3 v2 with 2.0.7 have the same issue. lets wait for news.

cguerrero1205, 2020-10-08:
Yes, it's like I told you: between the changes they made from Marlin 2.0.6 to 2.0.7, it was in PLR, they changed a lot of things, I guess something went wrong. The strange thing is that until now we have very few people reporting that problem, I don't know if it is something that doesn't affect everyone at the moment of updating.

pkfloyd, 2020-10-08:
maybe people don't notice. 
after installing 2.0.7 teste printing and runout sensor. i notice PLR bug with some lucky on the beginning of piece.

borsegbr, 2020-10-09:
Same problem with skr 1.4 turbo and btt tft35 v3.0

serial prompt say:

```
echo:Active Extruder: 0
M106 P0 S255
echo:Bed Leveling ON
echo:Fade Height 2.00
echo:Now fresh file: /LASER~1.GCO
open failed, File: LASER~1.GCO.
//action:resume

```

pkfloyd, 2020-10-09:
More people with this issue. Ok, need to wait for correction.

thanks

borsegbr, 2020-10-09:
in addition after abort resume serial say 

```
Deletion failed, File: PLR.
Power-loss file delete failed.
```

and if i try to print from media is void i have to select change media couple of time and file appears
seems that sd is not correct inizialize in resume print procedure

pkfloyd, 2020-10-09:
hi, today a test again and its work fine the PLR, (i have change the powerloss.cpp and powerloss.h from marlin 2.0.6 to 2.0.7 ) and use vscode com compile.
i have mks Lgen 32bits board and use cura 4.7.1.

don't now anything about coding.. think i have luck with the file change.

borsegbr, 2020-10-09:
> hi, today a test again and its work fine the PLR, (i have change the powerloss.cpp and powerloss.h from marlin 2.0.6 to 2.0.7 ) and use vscode com compile.
> i have mks Lgen 32bits board and use cura 4.7.1.
> 
> don't now anything about coding.. think i have luck with the file change.

not for me still not working with 2.0.6

cguerrero1205, 2020-10-10:
Today they made a PLR correction in the bugfix version. It's time to do a quick test to see if it solves our problem, although they don't mention anything about it.

https://github.com/MarlinFirmware/Marlin/pull/19676

pkfloyd, 2020-10-11:
hi, never use a bug fix.
how cant a download and test to?

ellensp, 2020-10-11:
@pkfloyd  go to https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x click on big green CODE icon and download zip 
Then update config file as needed. Compile and upload

borsegbr, 2020-10-11:
> Today they made a PLR correction in the bugfix version. It's time to do a quick test to see if it solves our problem, although they don't mention anything about it.
> 
> #19676

even worst, when i start printing any gcode  lcd show resume from powerloss: 3d printer halted: please reboot

pkfloyd, 2020-10-15:
hi, today i compile the marlin 4.0.7.1 and have the same issue.

PLR don't work.  wend  its turn on, say "resume" but nozzle go to piece and don`t move.

cguerrero1205, 2020-10-15:
I have already stopped testing. I am working without PLR. Play like this until the developers help us and solve the problem. I just hope I don't cut the power and lose some long impression.

pkfloyd, 2020-10-15:
think its to important, for not use, go back to version that work...

borsegbr, 2020-10-15:
> think its to important, for not use, go back to version that work...

what version work for you?

pkfloyd, 2020-10-15:
4.0.6 work fine

cguerrero1205, 2020-10-16:
4.0.6 or 2.0.6?

katherinehackworth, 2020-10-16:
I assumed @pkfloyd meant 2.0.6 because there isn't any marlin 4.0.6.

cguerrero1205, 2020-10-16:
> I assumed @pkfloyd meant 2.0.6 because there isn't any marlin 4.0.6.

I assume the same. Tomorrow I will try version 2.0.6.1

pkfloyd, 2020-10-16:
Hi, sorry my bad

Marlin 2.0.6  (not 4.0.6)  

And don't have installed 2.0.6.1

thinkyhead, 2020-10-16:
Power-loss recovery is always a challenge to keep up with. We ought to get some devs on really digging into the real problems and solutions by testing this feature on their own machines, but it's always a question of free time. I put in some hours last month and made some improvements, but as they say in engineering, in any complex system there are bound to be little cracks…

boelle, 2020-10-16:
i can chime in from what i know about how prusa does it

it starts with the controller, the einsy will as soon it get the signal from the psu stop heating the bed, this is done hardware wise and you can check the schematic to verify 

this alone saves power, and gives it enough power and time to move out of the way and save progress

without this only the invisible man in the sky knows what will happen

but the takeaway is that both psu and controller needs to be up for it, if stopping powering the bed is done in software that cost extra power and it might just be what is needed

ibrahimnadi, 2020-10-19:
Hello, I am having the same problem. Did you solve the problem? What method should I follow?

cguerrero1205, 2020-10-19:
> Hello, I am having the same problem. Did you solve the problem? What method should I follow?

Hi, the only solution we have found so far is to go back to version 2.0.6, this until the developers help us, find and solve the problem.

pkfloyd, 2020-10-20:
hi, Marlin 2.0.7.2 its out, anyone have tried if this issue its corrected? 

thanks

ibrahimnadi, 2020-10-20:
I will try today. I write the result.

borsegbr, 2020-10-20:
> hi, Marlin 2.0.7.2 its out, anyone have tried if this issue its corrected?
> 
> thanks

Same issue on skr1.4 turbo

boelle, 2020-10-20:
> developers help us, find and solve the problem.

they work for free so it can take a long time

ibrahimnadi, 2020-10-20:
I've been trying to run the power loss feature for two days. I did not succeed in any version. It did not work in version 2.0.7.2 either. When I say continue printing, the printer goes to the point where it was and stops.

I am using bigtreetech ups module. not detecting power loss. Z axis does not go up. 

Couldn't compile version 2.0.6, gives error vscode. :(

boelle, 2020-10-20:
> Couldn't compile version 2.0.6, gives error vscode. :(

Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

ibrahimnadi, 2020-10-20:
> Lütfen [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x)nerede durduğunu görmek için test edin . Sorun çözüldüyse, bu sorunu kapatabiliriz. Sorun henüz çözülmediyse, daha fazla araştırma yapmalıyız.

Firstly, thank you.

I also compiled and installed the 2.0 bugfix version. But it does not detect power loss. My problem is that:

I want to use Bigtreetech mini ups 2.0 module. I assembled it according to the instructions. When I turn off the power, it should detect the interruption from the Z MAX pin number 19 and raise the z-axis 5mm.

But when I turn off the power, it continues to print and then the printer turns off.

When I turn the printer back on, there is no continue message.

I've tried several marlin firmware to solve this. But I could not reach a solution.

borsegbr, 2020-10-20:
> I've been trying to run the power loss feature for two days. I did not succeed in any version. It did not work in version 2.0.7.2 either. When I say continue printing, the printer goes to the point [...]
This Is another problem not related, here bug is print doesn't restart after successful power loss catch

boelle, 2020-10-21:
> Firstly, thank you.

Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

ibrahimnadi, 2020-10-21:
My configuration and configuration_adv files marlin 2.0.7.2
[configuration and adv 2072.zip](https://github.com/MarlinFirmware/Marlin/files/5415306/configuration.and.adv.2072.zip)

> Please ZIP up your `Configuration.h` and `Configuration_adv.h` files

katherinehackworth, 2020-10-22:
> > Firstly, thank you.
> 
> Configurations, please
> Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
> requested in the Issue template) and drop them into your next reply.
> We'll check them over and see if anything is amiss.

[Here are mine for 2.0.7](https://gist.github.com/katherinehackworth/1d8e944c5fe0655753ebb9e71b1c10d2)

swanlm, 2020-10-24:
> I've been trying to run the power loss feature for two days. I did not succeed in any version. It did not work in version 2.0.7.2 either. When I say continue printing, the printer goes to the point [...]

Check whether Configuration -> Power Loss Resume is set to on in your display. If you didn't set it to default On in your configuration then you have to set via LCD or via M Command.

katherinehackworth, 2020-10-26:
There appear to be several issues here - some have power loss recovery that doesn't trigger at all on outage, some have PLR that triggers but doesn't prompt for resume on power recovery, and some (ie: myself) have PLR that after prompting for resume, heats back up, moves to the correct spot on the print, and then simply sits ad infinitum.

pkfloyd, 2020-10-30:
hi, so any news about this issue?

hollanbri, 2020-10-30:
I can confirm the main problem exists in the latest bugfix build.
I have Ender 3, with BigTreeTech SKR Mini E3 V2 mother board, BTT 24V UPS, BLTouch and Filament sensor.
I downloaded the current BugFix build and compiled with my configuration settings.
Attached are my Configuration.h Configuration_adv.h and src/pins/stm32f1/pins_BTT_SKR_MINI_E3_common.h (due to BLTouch wired to BUZZER and E0 pins - don't ask)

Upon power off the printer soon (within a second or two) starts the power off sequence with Z raising 5mm and E 10mm - then stops as it should.
Upon power resume I observe these actions:
  prompt to recover,
  head moves home
  the bed heats to temperature
  then head heats to temperature
  purge 20mm, 
  move head to where it was when power stop.
  And nothing else...
  
It just sits there - If I do a pause it will move up and on resume back, but still does not start printing.

By the way - I would like it if the bed and head heated at same time - to save my time.

As for the other 2 problems in this thread - I see Marlin detecting the power off, so double check configuration and pin state.  I see the prompt to recover.  So these features do work, at least in my configuration.
[config.zip](https://github.com/MarlinFirmware/Marlin/files/5468080/config.zip)

cguerrero1205, 2020-10-31:
Hello.

I am now in version 2.0.6.1 and PLR works well. I will stay in that version for the time being. Besides I made some modifications to add the z home after a power loss, since the z position was sometimes lost; I disabled some sub-menus, to save memory and invest it in linear advance, S-curve, filament sensor and other useful options.

At the moment, I have no complaints about version 2.0.6.1

Translated with www.DeepL.com/Translator (free version)

Sergey1560, 2020-11-02:
I tried it on 2.0.7.2 (2.0.x branch) and the problem is still there. I researched a little what was the matter.
Printing does not start because the M23 command cannot open the file. The trace is like this:
*M23()* call *file.open(diveDir, fname, O_READ)*
*SdBaseFile::open(SdBaseFile* dirFile, const char* path, uint8_t oflag)*  call  "return open(parent, dname, oflag);"
SdBaseFile::open(SdBaseFile* dirFile, const uint8_t dname[11], uint8_t oflag) call "p = dirFile->readDirCache();"
And, finally, at readDirCache() call to isDir() fail. The value of type_ at that mement is 0.

I could not figure out how the module SdBaseFile works with the file system, so I could not find the cause of the error. But I found a temporary solution. It is enough to add
```
card.release();
card.mount();
```
at start of PrintJobRecovery::resume() function and everything starts working.

I didn't make PR, I hope someone who knows how the module works will be able to find the cause of the error.

cguerrero1205, 2020-11-02:
> I tried it on 2.0.7.2 (2.0.x branch) and the problem is still there. I researched a little what was the matter.
> Printing does not start because the M23 command cannot open the file. The trace is [...]

Excellent!

Everyone to try that solution and comment on the results. It may not be the final solution, but the important thing is that it is functional and allows the use of PLR in version 2.0.7.
Maybe the developers will see your comment, get an idea of what's going on and get to the bottom of the problem. Because they do understand the part of the code that you don't.
Again, thank you very m a solution.
Excellent!

Everyone to try that solution and comment on the results. It may not be the final solution, but the important thing is that it is functional and allows the use of PLR in version 2.0.7.
Maybe the developers will see your comment, get an idea of what's going on and get to the bottom of the problem. Because they do understand the part of the code that you don't.
Again, thank you very much for your help and the time you spent to find

cguerrero1205, 2020-11-02:
I closed the publication by mistake, so I opened it again.

boelle, 2020-11-02:
@cguerrero1205 was this one not solved?

if there is a fix in here it should be sent in as a PR so things can be fixed faster

Sergey1560, 2020-11-02:
> @cguerrero1205 was this one not solved?
> 
> if there is a fix in here it should be sent in as a PR so things can be fixed faster

There is a workaround, but not a fix.  The PLR function will work, but the real problem has not been resolved.

[37 more replies]

PR #20035 (2020-11-06): Fix #19602
This change fix #19602 

Printing does not start because the M23 command cannot open the file.
One solution is to remount the sd card after the fileExists() function. Or just remove diveDir->close();.
closed 2020-12-13 by boelle