MarlinFirmware/Marlin #19603: [BUG] Robin Nano v1.2 with Marlin Color UI: display mirrored (left/right)

Miraculix200, 2020-10-03:
One of the recent commits since Sep 29 causes the 3.5" display on the Robin Nano 1.2 board to be horizontally mirrored (not rotated), when using Marlin Color UI. But I don't find any changes which could cause this, e.g. in the Robin Nano v1.2 pins file. Maybe one of the developers has an idea?

rhapsodyv, 2020-10-03:
Try changing this: `#define ILI9488_ORIENTATION               ILI9488_MADCTL_MX | ILI9488_MADCTL_MV`
to:
`#define ILI9488_ORIENTATION              ILI9488_MADCTL_MY |  ILI9488_MADCTL_MX | ILI9488_MADCTL_MV`

I will end for all, when #19192 get merged... :-(

thisiskeithb, 2020-10-03:
Please follow the Issue Template when filing a bug report. This includes attaching your configs:

```md
<!--

Have you read Marlin's Code of Conduct? By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/.github/code_of_conduct.md

Do you want to ask a question? Are you looking for support? Please don't post here. Instead please use the Marlin Firmware forum at http://forums.reprap.org/list.php?415 or the Marlin Facebook Group https://www.facebook.com/groups/1049718498464482/ or the Marlin Discord Server https://discord.gg/n5NJ59y.

Before filing an issue be sure to test the 1.1 and/or 2.0 "bugfix" branches to see whether the issue is already addressed.

-->

### Description

<!-- Description of the bug or requested feature -->

### Steps to Reproduce

<!-- If this is a Bug Report, please describe the steps needed to reproduce the issue -->

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.
```

Miraculix200, 2020-10-03:
Ok, I will try. For some reason the display is normal again after restarting the printer

rhapsodyv, 2020-10-03:
> Ok, I will try. For some reason the display is normal again after restarting the printer

Hum. It could indicate a bad init code. I will check that.

Did it only happen once?

Miraculix200, 2020-10-03:
Yes, only once after the recent commits (since Sep 29)

thisiskeithb, 2020-10-03:
@Miraculix200 Please attach your configs.

Miraculix200, 2020-10-03:
https://github.com/Miraculix200/Marlin-SapphirePro-2209-UART/blob/SapphirePro-2209-UART/Marlin/Configuration.h
https://github.com/Miraculix200/Marlin-SapphirePro-2209-UART/blob/SapphirePro-2209-UART/Marlin/Configuration_adv.h
https://github.com/Miraculix200/Marlin-SapphirePro-2209-UART/blob/SapphirePro-2209-UART/Marlin/src/pins/stm32f1/pins_MKS_ROBIN_NANO.h

Miraculix200, 2020-10-03:
I tried restarting the printer 10 times but it didn't happen again. Maybe it only happens when the printer reboots automatically after flashing

rhapsodyv, 2020-10-03:
So it’s look like more a bootloader issue.

closed 2020-10-04 by Miraculix200