MarlinFirmware/Marlin #19610: HOTEND_IDLE_TIMEOUT shuts down hot end in while printing

svdb99, 2020-10-04:
[Marlin 2.6.1]

Hi,
I just enabled HOTEND_IDLE_TIMEOUT with the following default settings

```
#define HOTEND_IDLE_TIMEOUT 
#if ENABLED(HOTEND_IDLE_TIMEOUT)
#define HOTEND_IDLE_TIMEOUT_SEC (5*60)
#define HOTEND_IDLE_MIN_TRIGGER   170
#define HOTEND_IDLE_NOZZLE_TARGET   0
#define HOTEND_IDLE_BED_TARGET      0
#endif
```
(printing PLA at 215C)
The time it takes for the nozzle and bed to heat up plus for ABL to complete is about 3 minutes, so 2 minutes into the print the nozzle shuts down and cools, thus failing the first layers.
Disabling the feature allows me to print normally again.

(Sidewinder X1 with stock TFT + SKR 1.4 turbo + TMC2209)

thisiskeithb, 2020-10-04:
Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

thinkyhead, 2020-10-04:
According to my reading of the code there would have to be 5 minutes elapsed between heating up the nozzle to 170 and starting the print with E motion for a timeout to be generated. As soon as any E motion occurs it cancels the timeout.

thisiskeithb, 2020-10-04:
I use this feature on several builds, but with a 30 minute timeout.

I tried to reproduce OP's issue by dropping `HOTEND_IDLE_TIMEOUT_SEC` down to the default `5*60` and starting a print with an ABL routine.

I preheat the nozzle to 160 before ABL, so I reduced `HOTEND_IDLE_MIN_TRIGGER` to 160 as well to ensure I was at the trigger temperature.

The only way I could get `HOTEND_IDLE_TIMEOUT` to kick in was by upping my ABL grid to a 7x7 (3 probes each point) which takes longer than five minutes. If I reduce the ABL grid down to 3x3 or 5x5, warmup + abl routine can be completed in under 5 minutes and `HOTEND_IDLE_TIMEOUT` is never called/the print will go on just fine without shutting off the heaters.

I've also run several all-day prints on my various printers (running the latest bugfix at the time) with a 30 minute timeout and none of shutoff heaters prematurely.

svdb99, 2020-10-06:
@thisiskeithb The goal of the time out is to prevent overcooking the filament in the nozzle and risking of gunking it up. So 30 mins is too long.

boelle, 2020-10-06:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

thisiskeithb, 2020-10-06:
> So 30 mins is too long.
 
it’s a personal preference which is why it’s customizable. Prusa uses a 30min timeout for example. 

It isn’t an issue for me since it’s mainly to catch when I preheat and forget to start a print. The nozzle is only at 160, not full temp, so I don’t have issues with clogs.

boelle, 2020-10-13:
is this a bug after all?

closed 2020-11-20 by github-actions[bot]