MarlinFirmware/Marlin #19611: Bed leveling bilinear not move X and Y axis

Disttrack, 2020-10-04:
When I send a "G29" the printer goes to do a bed leveling but only move the Z axis, X and Y axis are permanently in home position.
Im trying to compile marlin 2.0.7 for Anycubic Chiron.

thinkyhead, 2020-10-04:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

ellensp, 2020-10-04:
Please follow the Issue Template when filing a bug report. This includes attaching your configs:
(I know you didn't see it, as you posted in wrong place and it was manually moved here, but here it is) 
 
```lisp
<!--

Have you read Marlin's Code of Conduct? By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/.github/code_of_conduct.md

Do you want to ask a question? Are you looking for support? Please don't post here. Instead please use the Marlin Firmware forum at http://forums.reprap.org/list.php?415 or the Marlin Facebook Group https://www.facebook.com/groups/1049718498464482/ or the Marlin Discord Server https://discord.gg/n5NJ59y.

Before filing an issue be sure to test the 1.1 and/or 2.0 "bugfix" branches to see whether the issue is already addressed.

-->

### Description

<!-- Description of the bug or requested feature -->

### Steps to Reproduce

<!-- If this is a Bug Report, please describe the steps needed to reproduce the issue -->

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.
```

Disttrack, 2020-10-04:
> Please follow the Issue Template when filing a bug report. This includes attaching your configs:
> (I know you didn't see it, as you posted in wrong place and it was manually moved here, but here it is)
> 
> ```lisp
> 
> 
> ### Description
> 
> 
> 
> ### Steps to Reproduce
> 
> 
> 
> 1. [First Step]
> 2. [Second Step]
> 3. [and so on...]
> 
> **Expected behavior:** [What you expect to happen]
> 
> **Actual behavior:** [What actually happens]
> 
> #### Additional Information
> 
> * Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
> * Provide pictures or links to videos that clearly demonstrate the issue.
> * See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.
> ```

Sorry, i dont read the code of conduct, for the next time i know it.
I will try with the bugfix branch, but i post my config.
Thanks!
[Config_chiron_2.0.7.zip](https://github.com/MarlinFirmware/Marlin/files/5323178/Config_chiron_2.0.7.zip)

Disttrack, 2020-10-05:
I test the bugfix branch but the issue its the same, when leveling (G29) only moves Z axis, X and Y stay in home position.

sjasonsmith, 2020-10-05:
You have both min and max endstops enabled for your X and Y axis. Please disable the extra endstop connectors and test whether that helps.

```
#define USE_XMIN_PLUG
#define USE_YMIN_PLUG
#define USE_ZMIN_PLUG
#define USE_XMAX_PLUG
//#define USE_YMAX_PLUG
#define USE_ZMAX_PLUG
```

penright, 2020-10-07:
Just in case anyone else makes a dumb mistake as I did.
I did not set AUTO_BED_LEVELING_*
So the G29 would appear to hang. I was using OctoPrint ABL plugin. Once I ran the G28/G29 by hand.
I got this from the terminal. Sorry for any confusion I added.
```
Send: G28
Recv: echo:busy: processing
Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
Recv: echo:busy: processing
...
Recv: echo:busy: processing
Recv: X:107.00 Y:107.00 Z:10.00 E:0.00 Count X:8560 Y:8560 Z:4000
Recv: ok
Send: G29
Recv: Mesh Bed Leveling has no data.
Recv: X:107.00 Y:107.00 Z:10.00 E:0.00 Count X:8560 Y:8560 Z:4000
Recv: ok
Send: M113 S2
Recv: ok
```
This is my current configuration. I commented out MESH_BED_LEVELING and uncommented AUTO_BED_LEVELING_BILINEAR.
```
//#define AUTO_BED_LEVELING_3POINT
//#define AUTO_BED_LEVELING_LINEAR
#define AUTO_BED_LEVELING_BILINEAR
//#define AUTO_BED_LEVELING_UBL
//#define MESH_BED_LEVELING

```

Disttrack, 2020-10-07:
> You have both min and max endstops enabled for your X and Y axis. Please disable the extra endstop connectors and test whether that helps.
> 
> ```
> #define USE_XMIN_PLUG
> #define USE_YMIN_PLUG
> #define USE_ZMIN_PLUG
> #define USE_XMAX_PLUG
> //#define USE_YMAX_PLUG
> #define USE_ZMAX_PLUG
> ```

I tried this and its work! I had it uncommented because I thought it was necessary to use the probe connected to that port, hence the confusion. Thanks!

boelle, 2020-10-13:
@Disttrack so this one is "fixed" ? if so click the green close button below

sjasonsmith, 2020-11-18:
I am assigning this to myself to look into the problems caused when `USE_ZMAX_PLUG` is enabled along with probe homing.
I've encountered several users with this error, so I'd like to do something to prevent it.

sjasonsmith, 2020-11-23:
I spent several hours investigating issues involving USE_ZMAX_PIN. The ineractions with the various endstops gets pretty complicated. I've posted one PR (#20254) to avoid one pitfall I found, but I don't think it was directly applicable to this issue.

I'm going to go ahead and let this close since the reporter's configuration is fixed and now working. If we keep encountering issues involving USE_ZMAX_PIN we will need to pursue it further.

closed 2020-11-23 by sjasonsmith