MarlinFirmware/Marlin #19612: [BUG] in language_de.h translation error

kkberg, 2020-10-04:
### Bug Description

some translation into German for filament change messages on display are wrong and in parts they are opposite of what it should be.

**Expected behavior:** 

..............

    PROGMEM Language_Str MSG_ADVANCED_PAUSE_WAITING        = _UxGT(MSG_1_LINE("Klicke zum Fortsetzen"));
    PROGMEM Language_Str MSG_PAUSE_PRINT_PARKING           = _UxGT(MSG_1_LINE("Parken..."));

    PROGMEM Language_Str MSG_FILAMENT_CHANGE_INSERT        = _UxGT(MSG_1_LINE("Laden und Klicken"));
    PROGMEM Language_Str MSG_FILAMENT_CHANGE_HEAT          = _UxGT(MSG_1_LINE("Klicke zum Heizen"));
    PROGMEM Language_Str MSG_FILAMENT_CHANGE_HEATING       = _UxGT(MSG_1_LINE("Heizen..."));
    PROGMEM Language_Str MSG_FILAMENT_CHANGE_UNLOAD        = _UxGT(MSG_1_LINE("Entladen..."));
    PROGMEM Language_Str MSG_FILAMENT_CHANGE_LOAD          = _UxGT(MSG_1_LINE("Laden..."));
    PROGMEM Language_Str MSG_FILAMENT_CHANGE_PURGE         = _UxGT(MSG_1_LINE("Düse Reinigen..."));
    PROGMEM Language_Str MSG_FILAMENT_CHANGE_CONT_PURGE    = _UxGT(MSG_2_LINE("Klicke zum Beenden", "der Fila-
Entladung"));
**Actual behavior:** [What actually happens]

Would be great if someone could change.

Thank you

Speaka, 2020-10-04:
It was indeed wrong translated. I made a PR with the changes i think are the best german language.
Do you agree with that?

kkberg, 2020-10-04:
Thank you, I agree.

closed 2020-10-04 by thisiskeithb