MarlinFirmware/Marlin #19613: [BUG] Compile warning in bugfix on STM32F1 board

XDA-Bam, 2020-10-04:
### Bug Description

Compiling bugfix 604d10d3 on a BTT Mini SKR E3 v1.2 gives a couple of warnings:
```
C:\Users\Username\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\STM32ADC\src\utility\util_adc.c:10:30: warning: 'adc_result' initialized and declared 'extern'
 extern volatile unsigned int adc_result = 0;

C:\Users\Username\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\FreeRTOS701\utility\port.c: In function 'xPortStartScheduler':
C:\Users\Username\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\FreeRTOS701\utility\port.c:178:2: warning: implicit declaration of function 'systick_attach_callback' [-Wimplicit-function-declaration]
  systick_attach_callback(&xPortSysTickHandler);

At top level:
Compiling .pio\build\STM32F103RC_btt\libfb4\FreeRTOS701\utility\tasks.c.o
Compiling .pio\build\STM32F103RC_btt\libfb4\FreeRTOS701\utility\timers.c.o
C:\Users\Username\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\FreeRTOS701\utility\port.c:285:6: warning: 'prvSetupTimerInterrupt' defined but not used [-Wunused-function]
 void prvSetupTimerInterrupt( void )
```

### My Configurations

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/5323153/Config.zip)

### Steps to Reproduce

1. Pull bugfix-2.0.x @ 604d10d3
2. Apply configs
3. Build in VSCode using Marlin Auto Build

**Expected behavior:** Clean compile

**Actual behavior:** Compile finishes, but multiple warnings occur

#### Additional Information

The systick warning might be related to #19565

ellensp, 2020-10-04:
Take a look at the paths
Eg platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\STM32ADC\src\utility\util_adc.c
Ie this is not part of marlin. this is part of framework-arduinoststm32-maple .

closed 2020-10-04 by ellensp

XDA-Bam, 2020-10-04:
Sorry for the extra work 🙈 `#legallyblind`

bspark2, 2020-10-08:
HI There,  I have had the exact error while compiling as mentioned above by XDA-Bam. I am very much a novice...  Just wondering   what the fix was.. Thanks in advance