MarlinFirmware/Marlin #19617: [BUG] (short description)

Borillion, 2020-10-04:
Attempting to select Fan 2 in UI to adjust fan speed selects Fan 3 for adjustment. I cannot adjust the speed of fan 2 in the UI.

I am using an SKR Pro 1.1 with three fans (2 extruders, 1 part cooling).

I am running 2.0.X 20-09-2020

My Configurations
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5324062/Configuration.zip)

Steps to Reproduce
Have 3 fans connected to printer.
Select Temperature|Fan Speed 2:
Note that Fan Speed 3 is selected.
Expected behavior: [What you expect to happen]

I expect to be able to control the speed of fan 2.

Actual behavior: [What actually happens]

Selecting "Fan Speed 2" takes me to a screen to change to change the speed of fan 3. And indeed, the speed of fan 3 is changed. Selecting "Fan Speed 3" in the UI also changes fan 3.

Additional Information
I can change fan speed 2 only via gcode, sending M106 P1 S140 for example

XDA-Bam, 2020-10-04:
This sounds very much like the bug that was fixed by #18400.

thinkyhead, 2020-10-04:
Yes, on examining the current code, I can't see anything that would cause this kind of inconsistency with the second fan item. (although it looks like those singlenozzle items do need an update).

Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

thinkyhead, 2020-10-04:
Er, on a second look, maybe there is something to fix! Please compare the behavior in the Tune menu to the behavior in the Temperature menu. I expect the Tune menu needs to catch up to the patch from 18400.

Borillion, 2020-10-04:
> Er, on a second look, maybe there is something to fix! Please compare the behavior in the Tune menu to the behavior in the Temperature menu. I expect the Tune menu needs to catch up to the patch from 18400.

This is where I am having the issue in 2.0.x bugfix. In the tune menu!. I should have been more clear this is during a print

Borillion, 2020-10-04:
I am running 2.0.X 20-09-2020, this came from the marlin firmware downloads page at https://marlinfw.org/meta/download/

thinkyhead, 2020-10-04:
Ok, I have pushed a fix to `bugfix-2.0.x` so it should work correctly now.

closed 2020-10-04 by thinkyhead

Borillion, 2020-10-04:
Can I just clone from the latest Marlin bugfix fix or that particular
commit?

Thanks!

commit 0ff3ae3c (2020-10-04): Fix Tune/Fan edit items

- Fixes #19617
- Followup to #18400