MarlinFirmware/Marlin #19618: CoreXY  v. 1.1.8 -> 1.1.9 upgrade change axes

Theoristos, 2020-10-04:
Hello.
I tried software upgrade from Marlin 1.1.8 to latest Marlin 1.1.9 on RAMPS_14_EFB board with CoreXY mechanics and obtain a very weird behavior.
After upgrading with the completely same Configuration.h any command to move along X axis results in moving along Y and vice versa. 
X-Y endstops remains the same, so when trying to X or Y home the result was unstoppable movement via wrong axis.
Besides in the 1.1.8 everything works fine.

Has anyone met such behavior of knows what the reason and how to fix it?

thisiskeithb, 2020-10-04:
Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

---

Please follow the Issue Template when filing a bug report. This includes attaching your configs:

```md
<!--

Have you read Marlin's Code of Conduct? By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/.github/code_of_conduct.md

Do you want to ask a question? Are you looking for support? Please don't post here. Instead please use the Marlin Firmware forum at http://forums.reprap.org/list.php?415 or the Marlin Facebook Group https://www.facebook.com/groups/1049718498464482/ or the Marlin Discord Server https://discord.gg/n5NJ59y.

Before filing an issue be sure to test the 1.1 and/or 2.0 "bugfix" branches to see whether the issue is already addressed.

-->

### Description

<!-- Description of the bug or requested feature -->

### Steps to Reproduce

<!-- If this is a Bug Report, please describe the steps needed to reproduce the issue -->

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.
```

Theoristos, 2020-10-04:
Thank you for the answer.
I'm not yet sure is that a real bug or what.
Configs for 1.1.8 and 1.19 versions are attached below.

I'll try to compile 2.0.x, but due a major changes it takes a time
[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/5324248/Configs.zip)

thisiskeithb, 2020-10-04:
`1.1.x` is no longer being actively worked on, so it's best to move to `2.0.x`.

ellensp, 2020-10-05:
Fairly sure there was a bug a long long time ago that inverted an axis by accident that was later fixed.
Since its a corexy simply invert either axis to swap X and Y movement. Ie Set either INVERT_X_DIR or INVERT_Y_DIR to true.

Theoristos, 2020-10-05:
**ellensp**: thanks

sjasonsmith, 2020-10-10:
It seems there is nothing actionable for this Bug, since we are not going to back to change anything in 1.1.9.

@Theoristos, if you have issues configuring 2.x, here are some resources you can go to for help:

For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org/)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-10-10 by sjasonsmith