MarlinFirmware/Marlin #19623: [BUG] emergency parser not working for STM32F1 (BTT SKR E3 mini v2.0)

coudu, 2020-10-05:
### Bug Description

Emergency parser does not work for my STM32F1 borard (BTT SKR E3 mini v2.0).
Emergency commands are not jumping the queue and are executed "in normal queue order".

### My Configurations

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/5325905/configs.zip)

### Steps to Reproduce

1. compile latest bugfix-2.0.x for BTT SKR E3 mini v2.0 with emergency parser enabled
2. send normal commands to the printer
3. then send an emergency command (M112, M108 or M876)

**Expected behavior:** [What you expect to happen]
Emergency command should jump the queue and be executed asap

**Actual behavior:** [What actually happens]
Emergency command is executed after/at the end of other commands

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

coudu, 2020-10-06:
might be related to #19567 as I see the same behavior when sending M108 after filament load...

boelle, 2020-10-13:
Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

coudu, 2020-10-14:
Was already on latest bugfix, I just retested yesterday with latest commits and results are the same. I'am running the same config on a LPC1769 board and there is no issue with emergency parser.

rhapsodyv, 2020-10-14:
None of the emergency parser commands work? or only M108?

coudu, 2020-10-14:
unfortunately none... M108, M112, M876 to answer an host prompt, etc... everything is treated in queue order as if there were no emergency parser.

rhapsodyv, 2020-10-14:
which env are you using?

coudu, 2020-10-14:
I normally use STM32F103RC_btt_512K but I also tested STM32F103RC_btt_512K_USB and results are the same.

rhapsodyv, 2020-10-14:
STM32F103RC_btt_512K with SERIAL_PORT -1 is not currently handled.

But STM32F103RC_btt_512K_USB must work... Can you try again?

coudu, 2020-10-14:
not currently (LPC board in the printer) but i'll drop the STM32F1 in the printer to retest over the week-end.

sjasonsmith, 2020-11-15:
I'm replying to keep this open. The correct solution is probably to move boards to the STM32 HAL.

coudu, 2020-11-15:
sorry but with lockdown back here in France and me not being locked and hard at work, I didn't have any time to put the board back in the printer to test. This SMT32F1 board will eventually go in a printer at work or a friend's one so I'll be able to test sooner or later...
Thanks!

rhapsodyv, 2020-11-15:
There're 3 types of usb/serial in STM32F1:
 1) Hardware Serial - Support Emergency Parser
    Using `SERIAL_PORT != -1`
 2) Usb Serial - ***DON'T*** Support Emergency Parser
   Using `SERIAL_PORT == -1`
 3) Usb Serial + Usb Composite (to use mass storage) - Support Emergency Parser
   Using `SERIAL_PORT == -1` + `USE_USB_COMPOSITE` is defined (envs with `_USB`)

The reason is that lib maple don't give us a way to intercept usb data when using only `UsbSerial`. To do that, we would need to copy too much code from lib maple, and don't worth it in the current state of STM32F1.

I suspect that this issue is because the user are using (2).

Sebazzz, 2020-12-03:
I use SERIAL_PORT=-1 with USE_USB_COMPOSITE and it is also not working. To make it worse, understandably, EMERGENCY_PARSER commands are skipped during normal processing, so M112 does not work at all (gcode.cpp):

https://github.com/MarlinFirmware/Marlin/blob/3327568142f0294fa2b6e69b8b6567fa345ad205/Marlin/src/gcode/gcode.cpp#L493-L502

rhapsodyv, 2020-12-03:
Very weird. From where are you sending the M112? Computer serial or TFT Serial?

Sebazzz, 2020-12-03:
Host, Octoprint.

rhapsodyv, 2020-12-03:
I was looking at the code, and it seems exactly the same as the original PR, that @thisiskeithb tested: https://github.com/MarlinFirmware/Marlin/pull/19281

Sebazzz, 2020-12-03:
Sorry - my fault - 9a5fae436694c75da9595c7bf1e71142179994ed isn't in the source I'm building

rhapsodyv, 2020-12-03:
It's already merged on bugfix since Sep 6.

thisiskeithb, 2020-12-03:
> Sorry - my fault - [9a5fae4](https://github.com/MarlinFirmware/Marlin/commit/9a5fae436694c75da9595c7bf1e71142179994ed) isn't in the source I'm building

Always test with the latest [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip).

Sebazzz, 2020-12-03:
> 
> 
> > Sorry - my fault - [9a5fae4](https://github.com/MarlinFirmware/Marlin/commit/9a5fae436694c75da9595c7bf1e71142179994ed) isn't in the source I'm building
> 
> Always test with the latest [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip).

Sorry I'm always confused with the branching here (vs develop/release branches). It is not in Marlin 2.0.7.2 which is what I was testing - sorry all.

thisiskeithb, 2020-12-03:
Testing the latest `bugfix-2.0.x` (3eddbc7):
- **Board:** SKR Mini E3 V2 
- **Environment:** `STM32F103RC_btt_USB`
- **Serial Ports:** `#define SERIAL_PORT 2` / `#define SERIAL_PORT_2 -1`:

**Using OctoPrint:** I sent `M109 S210` to start a preheat & wait for hotend temp. Then I sent `M112` and OctoPrint disconnected, but Marlin didn't halt.

**Using a serial-connected BigTreeTech TFT:** I sent `M109 S210` to start a preheat & wait for hotend temp. Then I sent `M112` and Marlin halted.

Unfortunately, there's something wrong with my board since it's causing the Z stepper to heat up hotter than the sun upon boot, so I won't be able to test further until I get a replacement in.

rhapsodyv, 2020-12-03:
Anyone can test if this branch solves the issue?

https://github.com/rhapsodyv/Marlin/tree/fix-ep-usb-composite-stm32f1

Thanks

Sebazzz, 2020-12-22:
>     3\. Usb Serial + Usb Composite (to use mass storage) - Support Emergency Parser
>         Using `SERIAL_PORT == -1` + `USE_USB_COMPOSITE` is defined (envs with `_USB`)

I can confirm what @thisiskeithb says. To BTW retry it your can also use other emergency gcodes like M410 quick stop so Octoprint doesn't try to be smart.

rhapsodyv, 2020-12-22:
@Sebazzz my bad. I just fixed the branch. Can you test again?

Sebazzz, 2020-12-23:
> 
> 
> @Sebazzz my bad. I just fixed the branch. Can you test again?

@rhapsodyv No, sorry, doesn't work. All EMERGENCY_PARSER commands get ignored.

Is this bug a regression btw or has it never worked?

rhapsodyv, 2020-12-23:
> 
> Is this bug a regression btw or has it never worked?

It's hard to say. I don't have a F1 with UsbSerial/CDC to test. When I did the development, @thisiskeithb tested it. But as we have some many serial options, could be a wrong test. But the code changed a bit over the months too...

Can you confirm that `my_rx_callback` is called?

Sebazzz, 2021-01-03:
> It's hard to say. I don't have a F1 with UsbSerial/CDC to test. When I did the development, @thisiskeithb tested it. But as we have some many serial options, could be a wrong test. But the code [...]

No, it doesn't appear to be. I've put a `while(true)` in `my_rx_callback` and the printer keeps responding just fine.

FirewinX, 2021-01-17:
i have the same Problem on a Ender 5 with a BIGTREETECH SKR MINI E3 V2.0 with `Config Creality/Ender-5/BigTreeTech SKR Mini E3 2.0` on build `STM32F103RC_btt_512K` unload per btt tft35 is possible, but on octoprint shows permanently "paused for user" and cannot resume the print.
- no feedback about filament changes on the tft35
- after unloading, printing is simply continued without filament
- loading afterwards is not possible

Sebazzz, 2021-01-17:
@thisiskeithb  @rhapsodyv Any pointers where I should look and try to debug it?

FirewinX, 2021-01-17:
what should the line with `while (true)` look like in msc_sd.cpp @Sebazzz

rhapsodyv, 2021-01-17:
Author here. Sadly I don't have such boards to test. If anyone that can test it the code for me, reach on discord, I can fix the issue. Same username.

FirewinX, 2021-01-17:
i have the board for testing available

Sebazzz, 2021-01-18:
@rhapsodyv I contacted you on discord, my handle is the same

FirewinX, 2021-01-18:
i write you on discord on private message @rhapsodyv

arjanmels, 2021-02-06:
I also had problems on my SKR Mini-E3 v1.2 board. I submitted following fix: https://github.com/MarlinFirmware/Marlin/pull/21011, maybe this helps for v2.0 as well?

Sebazzz, 2021-02-08:
Confirmed @arjanmels - your fix works. Thank you.

thisiskeithb, 2021-02-08:
Closing since https://github.com/MarlinFirmware/Marlin/pull/21011 was merged.

closed 2021-02-08 by thisiskeithb