MarlinFirmware/Marlin #19627: Pause before deploy/stow not working with new UI style

alexeyzel, 2020-10-05:
### Bug Description
When I enable pause before deploy/stow feature by uncommenting #define PAUSE_BEFORE_DEPLOY_STOW - i can't confirm bed leveling sensor deployment on the screen, there is just no way to do it - when i click on deployment message - nothing happens.

### My Configurations
I have Flsun Q5 with MKS Robin Nano 1.2 board with 320x240 Graphical TFT screen. Marlin 2.0.7 release.
**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[alexeyzel_config.zip](https://github.com/MarlinFirmware/Marlin/files/5329328/alexeyzel_config.zip)

### Steps to Reproduce

1. Uncomment #define PAUSE_BEFORE_DEPLOY_STOW and enable UBL, compile firmware and upload it to printer
2. Turn the printer on and send "G28" command
3. Send "G29 P1" command

**Expected behavior:** [What you expect to happen]

Clickable message should appear at the center of the screen "Deploy Z-Probe", after manual probe deployment i should be able to confirm probe deployment by taping on the screen and bed leveling process should continue.

**Actual behavior:** [What actually happens]

"Deploy Z-Probe" message appears in the status bar and it is not clickable, there is no way to confirm probe deployment.

[image]

rhapsodyv, 2020-10-06:
I will take a look

Foxies-CSTL, 2020-10-06:
I confirm the same issues with FSMC and Color UI on QQS-Pro. The trick to validate is to press twice: once on an icon (bed, prefs, temp or fan) and again on the validation of the function. You have to do the same trick with the "BabyStepping" function to activate it during the first line of a print.

rhapsodyv, 2020-10-11:
This a bit tricker to solve. When marlin is used with a encoder, a "click" could happened anywhere. So, there's a few marlin operations that waits for those click without a specific screen for that.

I will do some tests and add a PR for testing soon.

rhapsodyv, 2020-10-11:
Can you test this PR? #19694 

Thanks.

boelle, 2020-10-13:
@Foxies-CSTL @alexeyzel did the pr from @rhapsodyv help?

Foxies-CSTL, 2020-10-14:
Hi ,
@boelle, I tried it and I replied on PR #19694 . I do it  with the last update bugfix (TFT Refactoring) and today I have a same issue. Ok for G33 but with the UBL (Build_Cold_Mesh) after have to valid the deploy, my screen freeze and I must do make reset. No problem with the screen to mode "TFT_Classic_UI" only mode "TFT_Color_UI".

rhapsodyv, 2020-10-14:
I will take a look in the UBL thing.

rhapsodyv, 2020-10-15:
This issue can be closed. I asked @Foxies-CSTL to open an issue specific for his problem, with his configs.

closed 2020-10-15 by thisiskeithb

alexeyzel, 2020-10-21:
Sorry for delay with testing. Tested in bugfix - works OK now. I just clicked the screen anywhere to confirm probe deployment.

PR #19694 (2020-10-12): Fix wait_for_user in the Color UI
### Description

Some marlin operations just wait for a encoder click, without any specific screen.

This PR makes the color ui handle it.

### Benefits

Fix #19627 

### Related Issues

#19627