MarlinFirmware/Marlin #19629: Opt-in MarlinFirmware / Marlin repo to participate in Hacktoberfest?

Gamester17, 2020-10-05:
### Description

Any plans to allow this repo to participate in Hacktoberfest? If so then you need to opt-in by classifying this repository with the hacktoberfest topic, see

https://hacktoberfest.digitalocean.com/hacktoberfest-update

### Feature Workflow

1. [First Action]
2. [Second Action]
3. [and so on...]

#### Additional Information

* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

thisiskeithb, 2020-10-05:
I think we’re doing OK without the spam 🙂

Gamester17, 2020-10-05:
Reduced spam is the reason why they updated how they label now, see https://hacktoberfest.digitalocean.com/hacktoberfest-update

ellensp, 2020-10-06:
This is just more work the for the admin. waste of time. Not a bug report, not even a FR.closing.

closed 2020-10-06 by ellensp