MarlinFirmware/Marlin #19630: [BUG][Robin Nano V1.2][FLSUN Q5][LVGL] Display upside down and icons scrambled

danielr1996, 2020-10-05:
### Bug Description

Display is rotated 180° and the icons are scrambled:
[image]
[image]

### My Configurations
The [configuration used](https://github.com/danielr1996/MarlinConfigurations/blob/bugfix-2.0.x/config/examples/delta/FLSUN/Q5-nano_v1/Configuration.h)  is the official from the Marlin/Configurations repo with a little [bugfix](https://github.com/MarlinFirmware/Configurations/pull/252)

I tried it with the 2.0.x and bugfix-2.0.x branches but the results are the same. 
The Mainboard is a MKS Robin Nano v1.2 on FLSUN Q5 with a display also from MKS (I don't no which exactly, the displays pcb only says Makerbase)

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5330118/Configuration.zip)

### Steps to Reproduce

1. Build the image with `platformio run`
2. Upload `.pio\builds\mks_robin_nano35\Robin_nano35.bin` to `<SD-CARD>\Robin_nano.bin` (my board didn't recognize `...nano35.bin`but only `...nano.bin` so I renamed it.
3. Upload `.pio\builds\mks_robin_nano35\assets` to `<SD-CARD>\assets`
4. Insert sd card into printer and turn on.

**Expected behavior:** 
I expected the display to be oriented correctly and the icons to be visible correctly

**Actual behavior:** 
The display is rotated 180° and the icons are scrambled (See images earlier)

rhapsodyv, 2020-10-05:
Is it a 3.5 display? LVGL only works in a 3.5 TFT.

danielr1996, 2020-10-06:
Nope it's a 2,5" Display. Then this might be an issue in the MarlinFirmware/Configurations repository because the config there for my printer has LVGL enabled so I thought it should work. 

Why are only 3.5" displays supported, are other sizes just not implemented yet or is it just not possible. If so we should state that in the configuration so it's clear to everyone why it doesn't work when they enabled it for an unsupported display.

rhapsodyv, 2020-10-06:
Your machine didn't shipped with Marlin LVGL.
It shipped with old MKS UI, that is similar to new LVGL.

Right now, Marlin just support lvgl por 480x320 screens - 3.5".

Mks is actively developing and soon we may have the lvgl for smaller screens, but right now it doesn't support.

It you think it is an issue in the marlin configurations samples, you open an issue in that repository pointing the problem.

danielr1996, 2020-10-06:
> Your machine didn't shipped with Marlin LVGL.
> It shipped with old MKS UI, that is similar to new LVGL.
>
> Right now, Marlin just support lvgl por 480x320 screens - 3.5".
>
> Mks is actively developing [...]

Thanks for you're response. I'll close this issue and open a new one in the configurations repo suggesting to change back to the old ui until support is there.

closed 2020-10-06 by danielr1996