MarlinFirmware/Marlin #19631: [FR] Multi Material Automatic Nozzle Offset

c3D-Dan, 2020-10-06:
Craftbot Idex seems to have the ability to automatically calculate nozzle offsets.  Can this be achieved in Marlin?

To the best of my knowledge, it appears each nozzle has an endstop signal wire attached it.  

Each hotend then touches the inside of a small circle cutout, at the front of the bed.   
The difference in X/Y position of each is then calculated precisely, avoiding the troublesome manual calibration process.  

To see an example of how it is done, please see here :
https://www.youtube.com/watch?v=NDvp8JoAuuo

thisiskeithb, 2020-10-06:
Right now, it’s a manual process using `PROBE_OFFSET_WIZARD`.

See https://github.com/MarlinFirmware/Marlin/pull/18866

c3D-Dan, 2020-10-06:
> Right now, it’s a manual process using `PROBE_OFFSET_WIZARD`.
> 
> See #18866

Calculating the Z offset isnt the problem, that's always pretty easy.  Its the XY offsets im talking about.  See time 2:30 https://www.youtube.com/watch?v=NDvp8JoAuuo

DrumClock, 2021-01-04:
Hi @c3D-Dan 
here it is better to see   ...   https://youtu.be/0-mW-u42-vg?t=100

DrumClock, 2021-01-08:
HI @c3D-Dan .... maybe it's this setting ...

[image]

c3D-Dan, 2021-01-08:
Why haven't I noticed that before?  That feature looks like it might just be what we're looking for. I'll give it a go and report back.

DrumClock, 2021-02-07:
Hi @c3D-Dan 
How does it work ?
Is that what you were looking for?