MarlinFirmware/Marlin #19634: M420 Z fade height not working as expected

DanJunior78, 2020-10-06:
Hello,

I´ve been using my Bltouch without any issues. I was testing to limit the software correction: M420 Z0.8 . Since them i was always crashing my hotend in the bed. Removing it everything was fine.
Did someone had the same issue.
Running Marlin 2.0.5.4

thinkyhead, 2020-10-06:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

boelle, 2020-10-06:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

DanJunior78, 2020-10-06:
Hello, it´s not my repository. I used the one from Github user "Knutwurst".
I did the change via M420 gcode. With M420 Z0.8 it was calculating wrong the heigth and it scratched badly my heatbed. Deactivating it with M420 Z0 and all was good again. Attached are the two requested files.
[Marlin_Knutwurst_2.0.5.4.zip](https://github.com/MarlinFirmware/Marlin/files/5333384/Marlin_Knutwurst_2.0.5.4.zip)
"

closed 2020-11-14 by github-actions[bot]