MarlinFirmware/Marlin #19635: Z-raise on powerloss not working

Thijsvanzwam, 2020-10-06:
When a powerloss occurs, there is no z-raise at all.

Connected a BTT mini UPS 12V to the PWRDET port on my SKR 1.4
Configured Marlin but no luck.
Switched on the power feature on the TFT but nothing happens.

deactivated #define PLR_ENABLED_DEFAULT  but no difference

I do not realy need the print to continue after a power failure, just want to raise the Z axis to prevent the nozzle from melting PLA and become one with the print.

My config:
BTT SKR 1.4 running Marlin 2.0.6.1
BTT TFT 3,5 V3.0 with latest firmware
BTT Mini UPS 12V connected to the PWRDET (P1_00)

[config.zip](https://github.com/MarlinFirmware/Configurations/files/5217363/config.zip)

Don't know if it's a bug or that I'm missing something...

thinkyhead, 2020-10-06:
Be sure to add `M413 S1` to the start of your G-code just to be extra sure the feature is enabled.

Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

boelle, 2020-10-13:
@Thijsvanzwam did you try M413 S1 ?

closed 2020-11-20 by github-actions[bot]

github051, 2020-12-15:
[image]
 You can change this: #define POWER_LOSS_ZRAISE 2 ( Power loss.CPP) file  (Refer attached image)

github051, 2020-12-15:
you can modify the Z raise value how much value you need to lift the nozzle.