MarlinFirmware/Marlin #19637: G35 failed at point 0 in Bed tramming assistant

emaayan, 2020-10-06:
when attempting to do G35, it fails with the error 
READ: X:102.00 Y:135.00 Z:0.00 E:0.00 Count X:8160 Y:10800 Z:0
READ: G35 failed at point 0 (Front-Left) X0.00 Y50.00

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5333128/Marlin.zip)

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

1.  Type G35 

**Expected behavior:** nozzle starts with left corner 
**Actual behavior:** nozzle is lowered to CENTER of the bed and fails 

#### Additional Information

i'm using marlin 2.0.7,i suspect it might be due to the fact that i'm using z end stop for homing and a Probe for leveling (which various SKR boards allow by dedicate a special connector for bltouch)

emaayan, 2020-10-06:
i should also add that when nozzle goes down it doesn't even deploy the probe

qwewer0, 2020-10-06:
The furthest point for Front-Left your printer can measure is X=10 Y=87, and that is why it cannot make it to the first point at x=0 Y=50.

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org/)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: [https://discord.gg/n5NJ59y](https://discord.gg/n5NJ59y)

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

boelle, 2020-10-06:
@emaayan are you asking a question here or reporting a bug?

thisiskeithb, 2020-10-06:
Closing as this is a configuration issue and not a bug.

Please use the links posted above for further configuration troubleshooting.

closed 2020-10-06 by thisiskeithb

emaayan, 2020-10-06:
> @emaayan are you asking a question here or reporting a bug?

i'm reporting what i think is a bug, the documentation says 
// Define positions for probing points, use the hotend as reference not the sensor.

so all my positions were based on the nozzle being as closest possible to the screws (not the nozzle.
if the points i've defined are beyond the limits shouldn't marlin fail to compoile? i recall it failed when i defined bed size which was too little for the probe to reach

Talha909, 2020-10-14:
Any one find solution.