MarlinFirmware/Marlin #19638: "Move screen for Color UI (#19386)[BUG] cause black Screen

andreibobirica, 2020-10-06:
### Bug Description

This commit:
Move screen for Color UI (# 19386)
causes black screen on printer on power up.

### My Configurations
Configuration with this Commit:
https://github.com/andreibobirica/Marlin/tree/5454befdbbd238943f24e7da20821e3f47a03187
COnfiguration without this commit that work:
https://github.com/andreibobirica/Marlin/tree/b96b6b19dbc5c8953a4072146ba04f13416a4e1b

### Steps to Reproduce

The Commit
Move screen for Color UI (# 19386)
modifies some parts of the code concerning the COLOR UI with TFT screen.
I use the Color UI (this problem only happens on the color UI), and I have a 320 x 240 ili9341 screen.
This commit adds the functionality of the move screen but only for 420 x 320 screens, as well as not being compatible with 320 x240 screens makes them totally unusable as the screen remains permanently black when switched on.
it is as if the Backlight pin was not activated.
I solved the problem by reverting this Commit.

#### Additional Information

For the LK4 printer (Alfawise U30) there are several screen versions, in particular I was able to interact with version 1.2 and 1.9, only the orientation of the touchscreen should vary, but in truth it is not only this.
Apparently this problem only happens on version 1.9
All the time ago there was a similar problem on version 1.2.
The problem was the same, the screen remained black when turned on.
However it has been fixed with the latest bugfix patches.
Now it has come back this time for screen version 1.9.

rhapsodyv, 2020-10-11:
I didn't understand how to reproduce the problem. The screen is inaccessible by a 320x240 screen. How could it hang something?

I just tested bugfix with color ui in a TFT28 320x240 and everything seems ok.

andreibobirica, 2020-10-11:
I am sorry, Let me explain better.
I regularly update my branch with the latest BUGFIX news.
I don't "merge COMMITs" every commit, but occasionally.
I noticed that updating from commit: 4d65ab5a0b483886143ae2cd937ed95ea496cbb0
until: 0c7035d8a3725d799cb286516a1bba40c578d202
There were problems with the firmware.

The problem I noticed is this:
After loading the firmware with the FLASH procedure from SD the menu appears, the printer can be used and everything works perfectly.
If the printer was turned off and then on again, the screen would remain black.
In practice it is as if the screen lights up only due to the firmware FLASHING procedure, and not thanks to the firmware.

This problem I have noticed happens only and only with the V1.9 version of the ili9341 screen of the LK4 (U30).
This problem does not happen with version V1.2.
In theory, only the year of production and the orientation of the touchscreen chenages between the two versions, but I am strongly convinced that it is not only this.

To understand how to get the firmware to work with the latest BUGFIX updates on my machine that has V1.9 I went looking for the problem and found that the COMMIT:
5b56d6698a2a9b78a5dd40446bcb606cbfff010d
CAUSE IT.
[image]

I have not had time to analyze what and which code negatively affects the screen, but I notice that some instructions do not operate only on the MOVE SCREEN.
I tried "reverting" individual changes. But the problem persists.
The only solution I found was to totally exclude this commit.
There is something in it that negatively affects the screen unfortunately, and it does on my specific version of the screen.
(for example on 1.2 it doesn't)

I think the problem may be in one of these places:
[image]

rhapsodyv, 2020-10-11:
Did you try the last bugfix? I have no problem. Just turned on/off many times here and everything is ok.

The only thing I can think that this commit can cause in a 320x240 screen is that it add two icons and may increase the flash size.... but, I'm almost sure that the compiler remove unused references.

My only guess now is that your flash usage is too high...

rhapsodyv, 2020-10-11:
The touch changes will not cause any trouble, because it always true for that screen.

rhapsodyv, 2020-10-11:
I can't compile marlin with supplied configs. First, it users a custom env. I tried to copy your custom env. Then I got I error saying the board don't support multi extruder. Commented out that line. Then I got error about undefined pins.

Sorry. But I can't help further if you don't provide a set of working configurations.

thisiskeithb, 2020-10-11:
Closing since this is a custom fork of Marlin issue that doesn't use the standard PIO/Marlin build environment.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-10-11 by thisiskeithb