MarlinFirmware/Marlin #19639: [BUG] Issue with M122 reporting only the first TMC2209 Slave driver settings for each axis

LRTNZ, 2020-10-06:
### Bug Description

Printer Board: BTT SKR V1.3
Drivers: TMC2209, with custom adaptor boards to have 2x drivers per axis, one on address 1, the other address 3
Marlin Version: Bugfix 2.0 branch, built from commit: b31e358270608afd02be895d4e3ca43ae927d31e

When using the M122 Command, only the settings for the first addressed slave driver are printed out.
Settings:
[image]
Output:
[image]

I can switch the addresses around, and it will always be only the first defined slave address that is printed out, the second one is never included.
### My Configurations

[configurations.zip](https://github.com/MarlinFirmware/Marlin/files/5334143/configurations.zip)

### Steps to Reproduce

1. Setup a driver board with a serial line per axis.
2. Using a breadboard, connect two TMC2209 drivers per axis, setting the addresses via the MS1/2 pins.
3. Flash Marlin with the provided configs
4. Once flashed and booted, run M122, and the resulting output should be just one of the drivers for each axis has it's settings printed out. Specifically the first driver that has it's address defined in the slave addresses.

**Expected behaviour:** [What you expect to happen]

I would expect for each of the slave addressed drivers to have an entry in the M122 debug output, so that I can see the settings for each of the drivers. This would allow for users to see the individual settings of each driver chip of an axis,
allowing for better tuning of printer designs that utilise multiple motors per axis. 

**Actual behaviour:** [What actually happens]

Only the first driver address entry under the slave driver configuration for each axis has it's settings printed out.

sjasonsmith, 2020-10-06:
It is working exactly as intended. Your Configuration.h is configured for one driver per axis. Marlin has no idea you want to configure more than one driver per axis.

I don’t know whether it is possible to define two drivers with the same EN/DIR/STEP pins. You could try it and see what happens.

It version key works correctly if you have a board with enough actual driver slots, and configure each driver separately.

LRTNZ, 2020-10-06:
> It is working exactly as intended. Your Configuration.h is configured for one driver per axis. Marlin has no idea you want to configure more than one driver per axis.
> 
I defined the X2 motor to be a TMC 2209 driver as well, unless I am mistaken in what I uploaded. Either way, I have tested it with and without the X2 driver configured, and it does not seem to change anything. How would you suggest I tell it I have more than one driver per axis?
> I don’t know whether it is possible to define two drivers with the same EN/DIR/STEP pins. You could try it and see what happens.
> 
To my understanding, this is the whole point of being able to set slave drivers. Well, that is certainly the reason I am getting from the PR by @thinkyhead that added in TMC2209 support. Also based on the documentation from Trinamic, sharing EN, STEP, and DIR pins is the intended way to use drivers that are slaves together.
> It version key works correctly if you have a board with enough actual driver slots, and configure each driver separately. 
>
? The whole point of what I am doing is to use the slave driver feature that is inbuilt to Marlin, to be able to have multiple TMC drivers per axis connection, without having to consume extra driver slots. Otherwise I would need a driver board with AT LEAST 7 driver slots, so that X, Y, and Z, can have 2 motors, each with their own driver module, and still have one slot for an extruder motor.

sjasonsmith, 2020-10-06:
> The whole point of what I am doing is to use the slave driver feature that is inbuilt to Marlin, to be able to have multiple TMC drivers per axis connection, without having to consume extra driver slots.

I may be wrong, but I have never heard of this feature in Marlin. The slave addresses doesn't make one driver follow the other, it is simply a unique address for each driver on a serial line. They don't have to be on the same access.

I see that you did defined an X2 driver type, I missed that when I looked earlier.
To actually enable dual drivers you need to enable `X_DUAL_STEPPER_DRIVERS`. This is going to default to using your E1 driver for X2, so you will have to modify your pins file to set your X and X2 pins to match. I don't know whether Marlin will let you do that, or if you will encounter issues.

I do think it is an interesting idea to allow for "Follower" drivers which require UART or SPI configuration but do not need extra EN/DIR/STEP pins. This isn't an issue for "traditional" drivers because there is no communication interface, you just attach multiple drivers to the same pins.

Atom-Timing, 2020-10-06:
Thanks Liam (AKA LRTNZ) for posting and thanks Jason for your response - much appreciated!
I thought I would chip in with some visual context regarding what Liam and I are practically trying to do.
I have designed a large 3D printer with a unique belt arrangement which I dubbed a Double H-bot. The main advantages are zero racking, shortened driven belt length and smaller motors (Nema 17 in this case). 
[image]
It does, however require four motors for the x/y-axis... Rather than run them in a compromised series or parallel arrangement I run external TB6600 drivers, however, I and keen to use a standard printer board such as the SKR V1.3 which supports UART. So here is our dual driver shield solution;
[image]

Here is the printer with noisy external TB6600 drivers;
[image]

[image]

Atom-Timing, 2020-10-06:
One other thing to mention is that Liam and I are located in Auckland, New Zealand so if we seem un-responsive it might be because we are down under and dead to the world...

ellensp, 2020-10-07:
sharing slave address is not a valid option. Yes they might get configured the same but they will also both respond over each others communication. 
Having two drivers per axis sharing step/direction with unique salve address is also not an option in marlin.
Only way I can see to do this currently would be to setup  

#define X_DRIVER_TYPE  TMC2209
#define Y_DRIVER_TYPE  TMC2209
#define Z_DRIVER_TYPE  TMC2209
#define X2_DRIVER_TYPE TMC2209
#define Y2_DRIVER_TYPE TMC2209
#define Z2_DRIVER_TYPE TMC2209

Set unique slave addresses for all  (your going to need a second uarts lines as only 4 steppers per uart)

Then enable dual drivers
#define X_DUAL_STEPPER_DRIVERS
#define Y_DUAL_STEPPER_DRIVERS
#define NUM_Z_STEPPER_DRIVERS 2

Finally modify the skr 1.3 pins.h so the  X2,Y2,Z2 pins are set to the same as X,Y and Z or unused pins.

And  Christchurch says hello :)

LRTNZ, 2020-10-07:
> sharing slave address is not a valid option. Yes they might get configured the same but they will also both respond over each others communication.
>
Each of the axis connections on the motherboard has it's own dedicated serial line, and I have configured the two stepper drivers for each x/y/z to have their own address. I know this works, as based on the first slave address I set, the driver with that address on each axis will respond. So they are not talking over the top of each other, nor are multiple drivers per serial connection sharing the same address. 
> Having two drivers per axis sharing step/direction with unique salve address is also not an option in marlin.
>
Oh bother, I wonder if this could be added to Marlin? As that would mean the full power of using slave addresses can be realised, for the boards that have a dedicated serial connection per axis.
> Only way I can see to do this currently would be to setup
>
> #define X_DRIVER_TYPE TMC2209
> #define Y_DRIVER_TYPE TMC2209
> #define Z_DRIVER_TYPE TMC2209
> #define X2_DRIVER_TYPE TMC2209
> #define [...]
Each axis has it's own dedicated UART line, so that is not an issue. Theoretically, there could be 4 stepper
drivers per axis, each with their own address.
> 
> Then enable dual drivers
> #define X_DUAL_STEPPER_DRIVERS
> #define Y_DUAL_STEPPER_DRIVERS
> #define NUM_Z_STEPPER_DRIVERS 2
> 
> Finally modify the skr 1.3 pins.h so the X2,Y2,Z2 pins are set to unused pins. So they are just toggling pointlessly.
> Perhaps -1 could be used for these pins and remove any tests that complain about this.
> 
Thanks, will take a look at this! I will have to see if I can set the UART connection pins for the dual drivers to be the same
as the ones connecting to the drivers for the x/y/z slots, so that the drivers can be then talked to correctly.
> And Christchurch says hello :)
>
Wonderful!

LRTNZ, 2020-10-07:
> I may be wrong, but I have never heard of this feature in Marlin. The slave addresses doesn't make one driver follow the other, it is simply a unique address for each driver on a serial line. They don't have to be on the same axis.
> 
I know that the slave addresses do not make one driver follow the other. That is why each of the axis has a riser board that duplicates the en/step/dir pins between two drivers so that they operate in sync. My main issue/query is how to set marlin to be able to talk over UART to those two drivers independently. On the SKR V1.3 board that is being used, there is a dedicated serial line to each one of the axis, so running out of slave driver addresses by having 2x drivers for X/Y/Z should not be an issue.
> I see that you did define an X2 driver type, I missed that when I looked earlier.
>
No worries!
> To actually enable dual drivers you need to enable `X_DUAL_STEPPER_DRIVERS`. This is going to default to using your E1 driver for X2, so you will have to modify your pins file to set your X and X2 pins to match. I don't know whether Marlin will let you do that, or if you will encounter issues.
> 
To my understanding I can set the pins that are not needed to be -1, which should get past any issues with error checking.
> I do think it is an interesting idea to allow for "Follower" drivers which require UART or SPI configuration but do not need extra EN/DIR/STEP pins. This isn't an issue for "traditional" drivers because there is no communication interface, you just attach multiple drivers to the same pins.
>
Exactly! This is what will allow this printer to be able to work, as it is using the technology that is already there with these drivers, in terms of talking to them with UART etc. It just gets around the need to dive deep into the pins file, and do a bunch of modifications there that may interfere with other functionality of the board, such as being able to run dual extrusion. As if to marlin the extra extruder slots are being taken up by the dual drivers for each axis, that could make it think there are no slots left to use to drive the extruder motors.

Atom-Timing, 2020-10-07:
Thanks for your neighbourly contribution @ellensp. I do mechanical - am a bit pathetic when it comes to electronics and code, however, am following along as best as I can.

Atom-Timing, 2020-10-07:
FYI
Here is the schematic of the SKR 1.3 board showing the five stepper driver socket serial connections to the CPU; https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.3/hardware/SKR-V1.3-SCH.pdf

A detail showing the stepper driver socket schematic;
[image]

Here is the schematic of the Fysetc TMC2209 V2.1 stepper driver module;
[image]

Here is a table of the TMC2209 signal descriptions;
[image]

sjasonsmith, 2020-10-07:
I don’t think there is any confusion regarding the board wiring or driver capabilities. @ellensp and I just don’t know whether Marlin will let you specify two drivers on an axis without unique EN/DIR/STEP pins for each.

LRTNZ, 2020-10-07:
> 
> 
> I don’t think there is any confusion regarding the board wiring or driver capabilities. @ellensp and I just don’t know whether Marlin will let you specify two drivers on an axis without unique EN/DIR/STEP pins for each.
>
Hmmm. Is there a way that Marlin could be changed to allow the user to specify multiple slave addresses per axis, that can be treated like the same driver most of the time, but can be individually addressed when required, such as debug outputs, configuring driver settings, etc? Or would that be a massive deviation from the core structure of Marlin, and what it defines as a "driver" traditionally?