MarlinFirmware/Marlin #19641: Printer rebooting when printing from SD using Auto Bed Leveling

Camarda, 2020-10-06:
### Bug Description

I have a custom RepRap Prusa i3 printer with Ramps 1.4 and I recently installed a BLTouch to use Auto Bed Leveling.
The leveling part itself works great, but I'm having a strange behavior.
If I manually select the leveling function through the LCD and then start a print from the SD, everything works. But if I include the G29 command after G28 in the GCodes, the printer will do heating and leveling procedures and then reboot just when the hotend reaches the position for printing the first line. I was using Marlin 2.0.6.1 and upgraded to 2.0.7, with same results. I tried adding the G29 command in the pre print code in Cura, and also manually added, giving the same results. Size and complexity of models don't seem to interfere in the results.

### My Configurations

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5334417/Configuration.zip)

### Steps to Reproduce

1. Include G29 command in any GCode
2. Start print from SD Card
3. Wait for heating and leveling routines to end

Expected behavior: The print should continue after the Auto Bed Leveling.

Actual behavior: Firmware reboots and print is stoped.

#### Additional Information

Printer is doing the bed leveling while heating the bed to save time (I'm aware that this reduces precision, but for my setup the effect seems negligible). Still doing the leveling before or after full heating causes no difference in the output.

ellensp, 2020-10-07:
Ramps notoriously have power issues on the 5v line. The bltouch might just be a the straw that broke the camels back.   
Also please provide actual gcode example.   (can cut file just after the crash point)

Camarda, 2020-10-08:
Any way to test this possible 5v problem? Just changing boards? And below is an example gcode:

```
;FLAVOR:Marlin
M73 P0
M117 Time Left 0h46m59s
;TIME:2819
;Filament used: 1.75809m
;Layer height: 0.2
;MINX:71.05
;MINY:69.05
;MINZ:0.3
;MAXX:126.95
;MAXY:124.95
;MAXZ:43.9
;POSTPROCESSED
;Generated with Cura_SteamEngine 4.7.1
M82 ;absolute extrusion mode
M140 S115.0 ; start preheating the bed WITHOUT wait to what is set in Cura

G28 ; home
G29 ; auto level Bed

M190 S115.0 ; start heating the bed to what is set in Cura and WAIT
M109 S230 ﻿T0 ; start heating hotend to what is set in Cura and WAIT

G1 Z5.0 F6000 ; move the extruder up 5mm

G92 E0 ; prime the extruder
G1 F200 E3
G92 E0
G92 E0
G92 E0
G1 F2100 E-4.5
;LAYER_COUNT:219
;LAYER:0
M107
G0 F3600 X74.058 Y85.844 Z0.3
;TYPE:SKIRT
G1 F2100 E0
G1 F1200 X74.7 Y85.371 E0.02894
```

boelle, 2020-10-13:
> Ramps notoriously have power issues on the 5v line.

> Any way to test this possible 5v problem?

The 5V problem comes from the controller board paired with the ramps, mega2560 is kind of junk when it comes to its 5v regulator. re-arm is far better build

closed 2020-11-20 by github-actions[bot]