MarlinFirmware/Marlin #19643: [BUG] Compiling VSCode with PIO

koltipelto, 2020-10-06:
I was compiling the Marlin 2.0.7 using the VSCode with PIO and this error occurred.

[image]
[image]

thisiskeithb, 2020-10-06:
Please follow the Issue Template when filing a bug report. This includes attaching your configs:

```md
<!--

Have you read Marlin's Code of Conduct? By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/.github/code_of_conduct.md

Do you want to ask a question? Are you looking for support? Please don't post here. Instead please use the Marlin Firmware forum at http://forums.reprap.org/list.php?415 or the Marlin Facebook Group https://www.facebook.com/groups/1049718498464482/ or the Marlin Discord Server https://discord.gg/n5NJ59y.

Before filing an issue be sure to test the 1.1 and/or 2.0 "bugfix" branches to see whether the issue is already addressed.

-->

### Description

<!-- Description of the bug or requested feature -->

### Steps to Reproduce

<!-- If this is a Bug Report, please describe the steps needed to reproduce the issue -->

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.
```

ellensp, 2020-10-07:
No configs. clearly its a default_envs = STM32F103RET6_creality so either a CREALITY_V4 or CREALITY_V427 controller.
Updated serial_port -1, no other settings are known...
Compiles fine. 
Closing until further information is provided

closed 2020-10-07 by ellensp