MarlinFirmware/Marlin #19644: Ender 3 v2, extruder moves when choosing to move Z

pixelicous, 2020-10-06:
### Bug Description

Ender 3 V2 upgraded to marlin 2.0.7, default config from github for the printer.
When stopping a print, it homes to 0,0,0 but then goes back directly to where it stopped.Hitting anything in its path.
It says "parking" when doing so.
From that point, I usually move the Z axis, to handle the failed print.
After upgraded to marlin it does move Z axis, but also triggers the extruder, which retracts back very fast the whole filament (i gave it the value 60, it shoot out everything).

I have to "auto home" after a stopped print to be able to move the Z axis safely.
I didn't have such issue on offical firmware

### My Configurations

https://github.com/MarlinFirmware/Configurations/tree/release-2.0.7
I didn't change anything in the config files, just changed platformio to `default_envs = STM32F103RET6_creality`

### Steps to Reproduce

1. Start print
2. Stop print
3. Under prepare>move, change Z axis to any positive number

**Expected behavior:** [What you expect to happen]

Stop print, take nozzle back "home", stay there.
Allow to move Z up

**Actual behavior:** [What actually happens]

Nozzle doesnt stay at home, extruder is inititated together with moving Z axis

signetica, 2020-10-07:
Try the latest bugfix branch, I think we just fixed it.

pixelicous, 2020-10-07:
> Try the latest bugfix branch, I think we just fixed it.
> 

I saw the bug fix 2.0 branch, but I I was afraid to use it.. Maybe it would be more harmful then good. If it fixes my issue do people actually stay on that branch or revert when new stable surfaces? Rather new user that's why asking..

Thanks in advance

Will try your recommendation in a few hours

signetica, 2020-10-07:
I had a similar problem resulting from stopping prints, and so I looked at the code enough to figure out how it worked and where the problem was.  I'd say that fix is safe.  You have a lot of choices depending upon your interests:

Stick with the Creality supplied firmware;
Update to the bugfix-2.0.x nightly from time to time;
Download one of the bugfix-2.0.7 nightly versions, last night or later, and stick with it if it works;
Add just the one fix to the 2.0.7 release and stick with that;
and so on.

If you are happy with the release of the firmware there's no need to update too often, although nice features are being introduced.  After 2.0.8 is released this particular bugfix will be in it.

As to the reliability or safety of the bugfix branch, who knows.  It's probably fine but I'd say unless you like testing or need a bug fixed, you might prefer to move to the next release when it comes out.  I'm using the bugfix branch now; I figure it can deal with problems without much trouble.  But if I were running a production shop I'd just add the one fix I needed, and only upgrade after testing and when needed.

pixelicous, 2020-10-08:
@signetica Thanks for the elaborated response, it is as i suspected.
I will try bug fix first and will close this issue till then.

closed 2020-10-08 by pixelicous

sjasonsmith, 2020-12-10:
#19642