MarlinFirmware/Marlin #19646: Laser odd/dangerous behavior in Menu- 2 part

buzkill420, 2020-10-07:
### Bug Description

A) Going to Laser Control setting power (anything other then 255) and enabling Laser, laser output is turned on, and laser fires at correct setting but then turns off by going back to main menu. But Letting info screen time-out happen, laser stays on and going back to menu then turns it off.

B) Setting laser to 255, laser stays on in menu, but if you then change laser power lower, when you go back to menu, Laser goes back to full power.

### My Configurations

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5338174/Marlin.zip)

### Steps to Reproduce

A1. Go to Laser Control and set power to anything other then 255.
A2. Enable Laser.
A3. (optional)  Let info screen-time-out do it's magic.
A4.  Go back to the main menu. Laser will TURN OFF.

B1. Go to Laser Control and set power to 255.
B2. A2. Enable Laser.
B3. Go back to the main menu. Laser will STAY ON.
B4. Go to Laser Control and set power to below 255.
B5. Go back to the main menu. Laser will go back to FULL POWER (255).

**Expected behavior:** [What you expect to happen]
 
Laser to turn on and off at set power when i enable it and STAY that way until told otherwise or unsafe (something errors and kills laser. ie. arm switch disabled)

**Actual behavior:** [What actually happens]

Laser will turn off or on depending on situation and presents a potentially unsafe situation where someone has set power to 0 and takes off safety glasses before screen timeout has taken effect.

#### Additional Information

Would love to help with making a version of marlin specific for lasers / helping with marlins laser features and bug testing

shitcreek, 2020-10-08:
Thanks for the report, but I was unable to reproduce this issue. I am using a slightly older version of Marlin but there hasn't been changes to the laser codes I don't think. I'll try the latest later and do some more digging.

buzkill420, 2020-10-08:
Did you swap pins at all?  Maybe that's what's happening...

buzkill420, 2020-10-08:
> Thanks for the report, but I was unable to reproduce this issue. I am using a slightly older version of Marlin but there hasn't been changes to the laser codes I don't think. I'll try the latest later and do some more digging.

Did you swap pins at all? Maybe that's what's happening...

buzkill420, 2020-10-09:
> Thanks for the report, but I was unable to reproduce this issue. I am using a slightly older version of Marlin but there hasn't been changes to the laser codes I don't think. I'll try the latest later and do some more digging.

And do you think you could please upload your configs and link your build please!  Id like to see what ihave different

buzkill420, 2020-11-11:
@shitcreek  any chance you could upload your configs for me to check.  I'm still getting this issue no matter what i do.  Multiple boards and versions and settings....

shitcreek, 2020-11-12:
[laser confg.zip](https://github.com/MarlinFirmware/Marlin/files/5533254/laser.confg.zip)

@buzkill420 After upgrading, I did noticed (perhaps it was the same before but I was just not paying close enough attention) that when you allow the LCD to time out while you are inside the laser power submenu (which while in it turns off the laser until you set a new power), the laser will stay off even though it will still show that it's on in the status screen. This is because it's expecting a click to enable the laser again. I have been busy trying to fix another laser issue/bug and haven't gotten to diving deeper into this one.

descipher, 2020-11-27:
Issue duplicated in 2.0.7.2:
Config Specifics:

Configuration.h
// Choose the name from boards.h that matches your setup
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_RAMPS_14_EFB
#endif  

Configuration_adv.h:
#define LASER_FEATURE
#if EITHER(SPINDLE_FEATURE, LASER_FEATURE)
  #define SPINDLE_LASER_ACTIVE_STATE    LOW    // Set to "HIGH" if the on/off function is active HIGH
  #define SPINDLE_LASER_PWM             true   // Set to "true" if your controller supports setting the speed/power
  #define SPINDLE_LASER_PWM_INVERT      false  // Set to "true" if the speed/power goes up when you want it to go slower
  #define SPINDLE_LASER_FREQUENCY       2500   // (Hz) Spindle/laser frequency (only on supported HALs: AVR and LPC)

Observation:
On power up servo pin D4 is HIGH - Correct
When entering the "Laser Control" menu for the first time the state is "Laser On" - Bug 1 not the correct state as it is not active,
moving to the "Laser On" menu action item and clicking the encoder button changes the servo D4 output to LOW, however the menu display remains at "Laser On" Bug 2 action is delayed and still inverted. When the encoder is reversed and the "Laser On" action menu item is exited the text value is set to "Laser Off" and the servo D4 pin is LOW so the laser is actually in the active state (Very dangerous!) Also during the Laser action menu item exit the "Laser Power:" menu item is updated from 0 to the stored preset value of 204 and the PWM is now appears to be driving servo D6 (subject to scope verification) however the "Laser Off" state remains on the menu. Bug 3

Perhaps this is a good time to separate the combined CNC/LASER menu/display functions. I assert this due to other factors. 
I am actively building a RepRap K40 style CO2 Laser frame and the RAMPS/Marlin combo is a good fit. The ability to control active water chillers, water pumps, thermal checks and an optional Z axis lift is important. RAMPS hosts this capability as well Marlin is platform friendly providing better compute for the target build. With the additional elements such as water temp, pump control, thermals which are inverted. e.g. cooling instead of heating. It is difficult to bond these functions to a Spindle Object and come out with a seamless non intrusive K40 like option on Marlin.

closed 2021-01-04 by github-actions[bot]