MarlinFirmware/Marlin #19648: [BUG] Tramming assistance boundry check and documentation

emaayan, 2020-10-07:
### Bug Description

this is related to issue  #19637  

those 2 defects in one 
when defining the tramming points the documenation says:
// Define positions for probing points, use the hotend as reference not the sensor.
which made me think i should use the coordinates where the NOZZLE is located i.e move the nozzle to where the screws are and use those  coordinates.

it failed with  failed at point 0 in Bed tramming assistant which didn't give much to go on 

i tried looking at the code and if i got it right, 
``
the issue occurs here:
```
 if (probe_relative) {                                     // The given position is in terms of the probe
    if (!can_reach(npos)) {
      **if (DEBUGGING(LEVELING)) DEBUG_ECHOLNPGM("Position Not Reachable");**
      return NAN;
    }
    npos -= offset_xy;                                      // Get the nozzle position
  }

```
so why not do this:
```
if (probe_relative) {                                     // The given position is in terms of the probe
    if (!can_reach(npos)) {
      if (DEBUGGING(LEVELING)) DEBUG_ECHOLNPGM("Position Not Reachable");
      SERIAL_ERROR_MSG("Position Not Reachable for probe, defined beyond Nozzle To Probe offsets");
      return NAN;
    }
    npos -= offset_xy;                                      // Get the nozzle position
  `}`

```
so the out put would be something like this:
```
READ: Error:Position Not Reachable for probe, defined beyond Nozzle To Probe offsets
READ: G35 failed at point 0 (Front-Left) X0.00 Y50.00
READ: G35 aborted.
```

other then that, i don't understand how come this can't be checked in compile time as i recall, that when i tried to defined probing arears  which were beyond the bed, it failed, shouldn't this be the case?

**Expected behavior:**
1. Clearer documentation
2. Runtime error indicating the reason for failure 
3. compiler error 

**Actual behavior:** [What actually happens]
1. unclear error message 

#### Additional Information
using Marlin 2.0.7

boelle, 2020-10-07:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

emaayan, 2020-10-07:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5339657/Marlin.zip)

thisiskeithb, 2020-10-07:
> // Define positions for probing points, use the hotend as reference not the sensor.
which made me think i should use the coordinates where the NOZZLE is located i.e move the nozzle to where the screws are and use those coordinates.

This is still true and how it works on the printers I run with that feature.

> Runtime error indicating the reason for failure

Having a compile-time sanity check would be ideal.

qwewer0, 2020-10-07:
@thisiskeithb I had the compile-time sanity check idea too, but @thinkyhead said that it isn't ideal, as the Probe Offset isn't fixed, and can be changed (M851).

thisiskeithb, 2020-10-07:
There could be both.

emaayan, 2020-10-08:
well initially my coordinates were  TRAMMING_POINT_XY { {  0, 50 }, { 205,  50 }, { 205, 260 }, { 0, 260 } }
[image]

because i have a 220,280 build plate, and a 205,280 print area, the screws are inwards, so the nozzle would be around 0,50, but then i the probing would fail, as my probe y  offset is actually 87 mm behind the nozzle.

qwewer0, 2020-10-08:
https://github.com/MarlinFirmware/Marlin/issues/18926#issuecomment-705594370

boelle, 2020-10-13:
> #18926

you closed that one, why if its relavant?

qwewer0, 2020-10-13:
I was ok with @sjasonsmith's answer, so I closed it.

closed 2020-11-20 by github-actions[bot]