MarlinFirmware/Marlin #19650: [FR]/[BUG] PROBE_OFFSET_WIZARD should move the nozzle to the bed center

qwewer0, 2020-10-07:
### Description

PROBE_OFFSET_WIZARD should move the nozzle to the center of the bed after it homed, but as of now, e817773 it only homes then gives the wizard options.

### Feature Workflow

This is what should happen:

1. Select Z Probe Wizard
2. Printer homes XYZ
3. Nozzle moves to the center of the bed
4. Use the wizard

#### Additional Information

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5340404/Configuration.zip)

thisiskeithb, 2020-10-07:
It will of `Z_SAFE_HOMING` is enabled (and the defaults are used), but I can see where this is problematic for users that home with a Z endstop and the probe is off the bed.

bill-orange, 2020-10-07:
@qwewer0 observation would also apply to my setups.  

Is this new feature intended to to be useful with most probes or just the BL Touch?

qwewer0, 2020-10-07:
@thisiskeithb My idea was a two in one for the solution. For a setup that uses Z endstop to home, it obviously would need to move to the center of the bed, and for a setup that uses a probe for homing, while it is homing at the center of the bed, but that is relative to probe, but for both setups after the homing is done, the nozzle should move to the center of the bed relative to the nozzle and not the probe, in order to have the most accurate outcome.

@bill-orange I think this is for every bed leveling probe.

thisiskeithb, 2020-10-07:
> Is this new feature intended to to be useful with most probes or just the BL Touch?

It works with multiple probe types. I've only tested it with a BLTouch and fix mounted probe though.

thisiskeithb, 2020-10-10:
@sjasonsmith Brought up another good point when using this feature.

It’d make sense to move the nozzle over to the probe point (based on X/Y probe offsets) for a more accurate Z probe offset, but I’m not sure how’d that factor into configs where users are homing with a Z endstop/not using `Z_SAFE_HOMING`.

Maybe @mwinters-stuff will have some ideas.

qwewer0, 2020-10-10:
> It’d make sense to move the nozzle over to the probe point (based on X/Y probe offsets) for a more accurate Z probe offset, but I’m not sure how’d that factor into configs where users are homing with a Z endstop/not using Z_SAFE_HOMING.

@thisiskeithb That is exactly what this [FR]/[BUG] is opened for.
The best way to have this, is if Z_SAFE_HOMING isn't enabled, then PROBE_OFFSET_WIZARD should move the nozzle to the center of the bed, and if Z_SAFE_HOMING is enabled, then PROBE_OFFSET_WIZARD should move the nozzle to the Z_SAFE_HOMING_X_POINT and Z_SAFE_HOMING_Y_POINT, which is X_CENTER/Y_CENTER by default.

sjasonsmith, 2020-10-10:
@qwewer0 I'm not sure if you completely understood my concern.

When probing, the _probe_ should be in the center of the bed.

When measuring the offset, the _nozzle_ should be in the center of the bed.

There should be an X/Y move between the probe and measure steps, so that they are referring to the same X/Y coordinate.

qwewer0, 2020-10-10:
> @qwewer0 I'm not sure if you completely understood my concern.
>
> When probing, the _probe_ should be in the center of the bed.
>
> When measuring the offset, the _nozzle_ should be in the center of the [...]

Am I not writing in english? That is exactly why I opened this issue.

qwewer0, 2020-10-10:
If the measured point by the probe is different from the measured point for the nozzle, then we could get wrong z offset, if the bed isn't absolutely flat.

mwinters-stuff, 2020-10-11:
I have always used safe z homing, that was a default config on my printer. I have recently moved house and my printer is still packed away, I wont have time for a few weeks to look at this.

I think we would have to add an option, where the safe way is to provide an X/Y coordinate for where the nozzle should be when this feature is being used (after the homing step). This could be used even if safe-homing is turned on.

qwewer0, 2020-10-11:
@mwinters-stuff `Z_SAFE_HOMING_X_POINT` and `Z_SAFE_HOMING_Y_POINT` are already in marlin for probe homing (`Z_SAFE_HOMING` enabled), and for z-endstop homing (`Z_SAFE_HOMING` disabled), the safest would be to just use `X_CENTER` and `Y_CENTER` as default.

I tried the `PROBE_OFFSET_WIZARD` with a `G0` "move to center" injected after the if (all_axes_homed()), and it worked fine.
So I think, we only need to add something like this (barbarian code) in menu_probe_offset.cpp at line 135:

```
    if (all_axes_homed()) {
      if ENABLED(Z_SAFE_HOMING) "Move nozzle to Z_SAFE_HOMING_X_POINT and Z_SAFE_HOMING_Y_POINT"
      else "Move nozzle to X_CENTER and Y_CENTER"

      ui.goto_screen(probe_offset_wizard_menu);
      ui.defer_status_screen();
    }
```

> I think we would have to add an option, where the safe way is to provide an X/Y coordinate for where the nozzle should be when this feature is being used (after the homing step). This could be used even if safe-homing is turned on.

The main problem is that the homed position and the examined position for the nozzle can be at a different Z height, so if there would be an option to change the examined position for the nozzle, that would only add more problems e.g. "bug report" from users that don't understand this.
And for z-endstop homing (`Z_SAFE_HOMING` disabled), it would be the best to just stay consistent and have examined position for the nozzle at the center of the bed.

alan-creater3d, 2020-10-15:
my understanding would be to probe the center of the bed (x,y) to probe activation then move nozzle to same (x,y) position and measure the offset(my custom machine homes z max, so z offset is +300mm at start of z movement in the wizard a lot of manual movement at 1mm a time)

Loreagle, 2020-10-15:
I agree with this request: during the wizard, exactly after homing, the nozzle should move according to the probe offset in X and Y so it will be in the same position where the probe measured the distance with the bed. After this mod the feature will be perfect!

qwewer0, 2020-10-17:
Updated the Description, and the Feature Workflow to better reflect my idea for ideal `PROBE_OFFSET_WIZARD` operation.

swissnorp, 2020-11-16:
In my Opinion this feature is not that flawless and I like your approach @qwewer(). I did some work on this but my code kept crashing my printer xD... I finaly gave up because I am no coder myself.

_Flaws of the actual Feature_
- `SET_SOFT_ENDSTOP_LOOSE(true)` must be called after `G28`. (G28 resets it to false) **[BUG]**
- The Nozzle does not move to the same position where the probing was done.
Therefor it is not possible to use this feature without `Z_SAFE_HOMING` (Homing by regular Z-Endstop instead of Probe) **[Weaknes]**
- In my Opinion `PROBE_OFFSET_START` is obsolete and `NOZZLE_TO_PROBE_OFFSET.Z` should be used instead.

My suggestion is as followed.. may you could pick it up for the feature request?!

**CONFIGURATION**
- It should be possible to uncomment `PROBE_OFFSET_START` . If uncommented, `NOZZLE_TO_PROBE_OFFSET.Z` would be used instead!
`PROBE_OFFSET_START` should also be allowed to be **Zero**.
- It should be possible to define a `PROBE_OFFSET_POS { X , Y }` to define where the probing should be done (after probing the nozzle would move to the same position). The default values would be `{ X_CENTER , Y_CENTER}`. This would be better for a big bed.

**ACCESS**
- Of course the Feature needs a Probe. This must be checked in SanityCheck or wherever.
- I think it's okay if the Feature is reachable by the Advanced Configuration/Settings. But other opinions are welcome.

**SUGGESTED BEHAVIOR**
1. Wizard is called by Menu
2. (Store `probe.offset.z` for backup and overwrite it with `PROBE_OFFSET_START`)
3. Do `G28 O` (or `G28 Z` if homed by probe)
4. Store Bed-Leveling-State and deactivate it
5. Deactivate Software-Endstops
6. Probe Z at `PROBE_OFFSET_POS { X , Y }` : `popxy_height = probe.probe_at_point()`
7. The probed height should be used value for calculation: `calculated_z_offset = probe.offset.z + current_position.z + popxy_height`
8. The Nozzle moves to the Probing point `PROBE_OFFSET_POS { X , Y }` and the Menu for adjustment opens as it is now. Nothing needs to be changed
9. The Done and the Cancel Action are good as they are. But after accepting with "Done" there should be a `G28 Z` if homing is done by probe!

Good Night and Bye , norp

qwewer0, 2020-11-27:
Closing, since solved.

closed 2020-11-27 by qwewer0